micropython/micropython #2343: Wipy: Make SSL/TSL encryption version configurable

hoihu, 2016-08-22:
see
http://forum.micropython.org/viewtopic.php?f=11&t=2066&sid=513f4049239becc0dc23c3d9caf38ecb

at the moment support for TLS V1 is hard coded in the wipy, although the CC3200 could support TLS V1.2 (which is mandatory for amazon aws services and others).

Could that be made configurable?

wjbuys, 2016-10-01:
It looks like [TLS 1.2 is now supported by the latest upstream version of axTLS](https://github.com/esp8266/Arduino/issues/2537). This would be really helpful for using cheap ESP8266 modules with AWS IoT (which requires TLS 1.2)

danicampora, 2016-10-01:
Somehow I missed this... Will fix now...

danicampora, 2016-10-01:
Solved in https://github.com/micropython/micropython/commit/a0d97fe40872b46872657bedc0e47cfd704c59aa

closed 2016-10-01 by danicampora

hoihu, 2016-10-02:
excellent thanks!