micropython/micropython #2347: sys.stdout.write changes \n bytes to \r\n

ryannathans, 2016-08-24:
I'm attempting to write binary directly out but every time a \n is printed, a \r is automatically inserted before it which corrupts the stream.

Is this a bug or should I be printing raw data out a different way?

stinos, 2016-08-24:
Please state platform, toolset, ... and maybe some code exhibiting the behaviour?

pfalcon, 2016-08-24:
Also important to read: [How to Report Bugs Effectively](http://www.chiark.greenend.org.uk/~sgtatham/bugs.html).

ryannathans, 2016-08-24:
I'm currently operating on the world's worst Internet connection. It works about one minute every hour in this rain. This bug is occurring on the stmhal port.
Reproduce by sys.stdout.write(b'\n'). It prints \r\n when I expected this to be a raw write.

https://github.com/micropython/micropython/issues/2347#issuecomment-242017192

pfalcon, 2016-08-24:
You need to use UART object (from machine or pyb module). sys.stdout is a stream in text mode and can do arbitrary transformations to binary data.

dhylands, 2016-08-24:
Or USB_VCP if writing to the usb-port.

You can also use sys.stdout.buffer to get a binary mode stream (rshell uses
that for doing the file transfers).

closed 2016-08-28 by pfalcon