micropython/micropython #2349: Disabling REPL

crosvera, 2016-08-24:
Hey there people!

I have an esp8266 board (nodeMCU, lolin v3) and I want to use the UART GPIOs to read from a CO2 sensor, but REPL is using it. Is there any way to stop REPL, so I can release UART GPIOs for my own software?

Thanks in advance!
Best Regards.

deshipu, 2016-08-24:
Well, REPL only runs when you program is not running, so that's one way -- just write your program and let it run from ``main.py``. Of course that makes it impossible to try things out with WebREPL and so on, so it's not perfect.

It would be nice to be able to disable the REPL on the serial connection, but still keep WebREPL working.

crosvera, 2016-08-24:
Thanks @deshipu,
BTW, what files should I see to figure out how REPL works?

Best regards.

deshipu, 2016-08-24:
I guess that would be https://github.com/micropython/micropython/blob/master/esp8266/esp_mphal.c

roger-, 2017-05-06:
I'm interested in using the UART and webrepl too, any updates on this?

peterhinch, 2017-05-06:
There are threads in the forum on this topic, e.g. [this one](https://forum.micropython.org/viewtopic.php?f=16&t=2078&hilit=UART).

craftyguy, 2017-05-19:
Hmm, I read through that thread, and this ticket, and it seems that:

- If I don't care about REPL or WebREPL, I can use UART as-is to 'talk' to other devices. 

- If I want WebRepl, but need to disable REPL, there's a PR somewhere to implement this but it's still WIP. 

- If I want UART and webrepl too, then basically use a NPN transistor to manually switch between UART and REPL.

Is that a fair summary of the situation?

peterhinch, 2017-05-19:
The transistor mentioned in the forum post serves as an 'or' gate. Its purpose is to force Txd (Tx out) high (inactive) in order to suppress the initial 75Kbps data stream from the ESP8266. It has no connection with switching UART and REPL, nor with nonblocking reads.

craftyguy, 2017-05-19:
Ah, I see, thank you for the clarification.

torwag, 2017-07-07:
There is a partial solution coming with the SDK not yet implemented in MP.

https://forum.micropython.org/viewtopic.php?f=16&t=3533

dpgeorge, 2018-05-21:
It's now possible to disable the REPL on UART(0) on the esp8266, see afd0701bf7a9dcb50c5ab46b0ae88b303fec6ed3 and PR #3784.

closed 2018-05-21 by dpgeorge