micropython/micropython #2350: Fail to build with i686-5.3.0-release-posix-dwarf-rt_v4-rev0

buaabyl, 2016-08-24:
The installer of Visual Studio 2013 com is too big...
Most of time I use mingw32 without MSYS or Cygwin.

The default windows/Makefile can't create micropython.exe
So I write a script to build it.

This is script:

```python
#!/bin/python
# -*- coding:utf-8 -*-

from __future__ import print_function

import os
import stat
import glob
import subprocess

CSRCS_ = [
    r"../py/*.c",
    r"../extmod/machine_mem.c",
    r"../extmod/machine_pinbase.c",
    r"../extmod/modubinascii.c",
    r"../extmod/moductypes.c",
    r"../extmod/moduhashlib.c",
    r"../extmod/moduheapq.c",
    r"../extmod/modujson.c",
    r"../extmod/modurandom.c",
    r"../extmod/modure.c",
    r"../extmod/moduzlib.c",
    r"../lib/mp-readline/*.c",
    r"../unix/file.c",
    r"../unix/gccollect.c",
    r"../unix/input.c",
    r"../unix/main.c",
    r"../unix/modos.c",
    r"../unix/modtime.c",
    r"../unix/modmachine.c",
    r"../windows/*.c",
    r"../windows/msvc/*.c",
]

CSRCS = []
for pattern in CSRCS_:
    CSRCS.extend(glob.glob(pattern))

def file_put_contents(fn, d):
    f = open(fn, 'w')
    f.write(d)
    f.close()

def is_update(src, dst):
    if not os.path.isfile(dst):
        return True
    elif os.stat(src)[stat.ST_MTIME] > os.stat(dst)[stat.ST_MTIME]:
        return True
    return False

def gcc_preprocess(cfiles):
    CFLAGS = r"-I . -I .. -I ../windows -E -Wp,-C,-dD,-dI -DNO_QSTR -DN_X64 -DN_X86 -DN_THUMB -DN_ARM"

    l = []
    for infn in CSRCS:
        infn = os.path.abspath(infn)
        outfn = 'tmp\\' + infn[len(ROOT)+1:].replace('\\', '_') + '.pp'
        l.append((infn, outfn))

        if is_update(infn, outfn):
            print(' %s' % cfn)
            res = subprocess.check_output("gcc " + CFLAGS + " " + infn)
            file_put_contents(outfn, res)

    return l

def analysis_preprocessed(c_pp_pairs):
    CFLAGS = r"-I . -I .. -I ../windows/msvc -E"

    for cfn, ppfn in c_pp_pairs:
        outfn = os.path.abspath(cfn)
        for m, r in [("/", "__"), ("\\", "____"), (":", "@"), ("..", "@@")]:
            outfn = outfn.replace(m, r)
        outfn = "genhdr/qstr/" + outfn + ".qstr"

        if is_update(ppfn, outfn):
            print(' %s' % ppfn)
            subprocess.check_call("python ../py/makeqstrdefs.py " + \
                    "split %s genhdr/qstr genhdr/qstrdefscollected.h" % ppfn)

    print('create genhdr/qstrdefscollected.h ...')
    subprocess.check_call("python ../py/makeqstrdefs.py " + \
            "cat none genhdr/qstr genhdr/qstrdefscollected.h")

    res = subprocess.check_output("gcc " + CFLAGS + " ../py/qstrdefs.h ../unix/qstrdefsport.h")
    file_put_contents("genhdr/qstrdefspreprocessed.h", res)

    res = subprocess.check_output("python ../py/makeqstrdata.py " + \
            "genhdr/qstrdefspreprocessed.h genhdr/qstrdefscollected.h")
    file_put_contents("genhdr/qstrdefs.generated.h", res)

def gcc_compile(CSRCS):
    CFLAGS = r"-I . -I .. -Os -Wpointer-arith -Werror -ansi -std=gnu99 " + \
             r"-DUNIX -D__USE_MINGW_ANSI_STDIO=1 -DMICROPY_USE_READLINE=1 -c"
    for cfn in CSRCS:
        outfn = cfn
        for m, r in [("/", "__"), ("\\", "____"), (":", "@"), ("..", "@@")]:
            outfn = outfn.replace(m, r)
        outfn = 'tmp/' + outfn + '.o'

        if is_update(cfn, outfn):
            print(' %s' % cfn)
            subprocess.check_call('gcc ' + CFLAGS + ' '+ cfn + ' -o ' + outfn)

    subprocess.check_call('gcc -c ../py/nlrx86.S -o tmp/@@__py__nlrx86.o')
    subprocess.check_call('gcc tmp/*.o -lm -o micropython')

if __name__ == '__main__':
    if not os.path.isdir('genhdr'):
        os.mkdir('genhdr')

    if not os.path.isdir('genhdr/qstr'):
        os.mkdir('genhdr/qstr')

    if not os.path.isdir('tmp'):
        os.mkdir('tmp')

    subprocess.check_call("python ../py/makeversionhdr.py genhdr/mpversion.h")

    PWD = os.getcwd()
    ROOT = os.path.dirname(PWD)

    print('preprocess...')
    PP_FILES = gcc_preprocess(CSRCS)

    print('analysis...')
    analysis_preprocessed(PP_FILES)

    print("compiling...")
    gcc_compile(CSRCS)
```

stinos, 2016-08-24:
Looks nifty but you're just duplicating a lot of existing functionality.

> The default windows/Makefile can't create micropython.exe

What is the problem exactly? I think you're better of trying to fix it. Or at least create a proper issue describing the problem.

buaabyl, 2016-08-25:
The error message is:

```sh
python ../py/makeversionhdr.py build/genhdr/mpversion.h
"GEN build/genhdr/qstr.i.last"
if [ "../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/lexer.c ../py/lexerstr.c ../py/lexerunix.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmx64.c ../py/asmx86.c ../py/asmthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py/emitglue.c ../py/runtime.c ../py/runtime_utils.c ../py/nativeglue.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty.c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ../py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrunicode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../py/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport.c ../py/builtinevex.c ../py/modarray.c ../py/modbuiltins.c ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmath.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ../py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint.c ../py/frozenmod.c ../py/../extmod/moductypes.c ../py/../extmod/modujson.c ../py/../extmod/modure.c ../py/../extmod/moduzlib.c ../py/../extmod/moduheapq.c ../py/../extmod/moduhashlib.c ../py/../extmod/modubinascii.c ../py/../extmod/virtpin.c ../py/../extmod/machine_mem.c ../py/../extmod/machine_pinbase.c ../py/../extmod/machine_pulse.c ../py/../extmod/machine_i2c.c ../py/../extmod/modussl_axtls.c ../py/../extmod/modurandom.c ../py/../extmod/modwebsocket.c ../py/../extmod/modwebrepl.c ../py/../extmod/modframebuf.c ../py/../extmod/fsusermount.c ../py/../extmod/vfs_fat.c ../py/../extmod/vfs_fat_ffconf.c ../py/../extmod/vfs_fat_diskio.c ../py/../extmod/vfs_fat_file.c ../py/../extmod/vfs_fat_lexer.c ../py/../extmod/vfs_fat_misc.c ../py/../extmod/moduos_dupterm.c ../py/../lib/embed/abort_.c ../py/emitnative.c ../unix/main.c ../unix/file.c ../unix/input.c ../unix/modos.c ../unix/modmachine.c ../unix/modtime.c ../unix/gccollect.c windows_mphal.c realpath.c init.c sleep.c ../lib/mp-readline/readline.c" = "" ]; then \
    echo "QSTR Looks like -B used, trying to emulate"; \
    gcc -E -DNO_QSTR -DN_X64 -DN_X86 -DN_THUMB -DN_ARM -Ibuild/tmp -I../lib -I../lib/netutils -I. -I.. -Ibuild -Wall -Wpointer-arith -Werror -ansi -std=gnu99 -DUNIX -D__USE_MINGW_ANSI_STDIO=1 -DMICROPY_USE_READLINE=1 -Os  ../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/lexer.c ../py/lexerstr.c ../py/lexerunix.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmx64.c ../py/asmx86.c ../py/asmthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py/emitglue.c ../py/runtime.c ../py/runtime_utils.c ../py/nativeglue.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty.c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ../py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrunicode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../py/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport.c ../py/builtinevex.c ../py/modarray.c ../py/modbuiltins.c ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmath.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ../py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint.c ../py/frozenmod.c ../py/../extmod/moductypes.c ../py/../extmod/modujson.c ../py/../extmod/modure.c ../py/../extmod/moduzlib.c ../py/../extmod/moduheapq.c ../py/../extmod/moduhashlib.c ../py/../extmod/modubinascii.c ../py/../extmod/virtpin.c ../py/../extmod/machine_mem.c ../py/../extmod/machine_pinbase.c ../py/../extmod/machine_pulse.c ../py/../extmod/machine_i2c.c ../py/../extmod/modussl_axtls.c ../py/../extmod/modurandom.c ../py/../extmod/modwebsocket.c ../py/../extmod/modwebrepl.c ../py/../extmod/modframebuf.c ../py/../extmod/fsusermount.c ../py/../extmod/vfs_fat.c ../py/../extmod/vfs_fat_ffconf.c ../py/../extmod/vfs_fat_diskio.c ../py/../extmod/vfs_fat_file.c ../py/../extmod/vfs_fat_lexer.c ../py/../extmod/vfs_fat_misc.c ../py/../extmod/moduos_dupterm.c ../py/../lib/embed/abort_.c ../py/emitnative.c ../unix/main.c ../unix/file.c ../unix/input.c ../unix/modos.c ../unix/mo [... cut ...]
w64-mingw32-static/include' CXXFLAGS='-O2 -pipe -I/c/mingw540/i686-540-posix-dwarf-rt_v5-rev0/mingw32/opt/include -I/c/m
ingw540/prerequisites/i686-zlib-static/include -I/c/mingw540/prerequisites/i686-w64-mingw32-static/include' CPPFLAGS= LD
FLAGS='-pipe -L/c/mingw540/i686-540-posix-dwarf-rt_v5-rev0/mingw32/opt/lib -L/c/mingw540/prerequisites/i686-zlib-static/
lib -L/c/mingw540/prerequisites/i686-w64-mingw32-static/lib -Wl,--large-address-aware'
Thread model: posix
gcc version 5.4.0 (i686-posix-dwarf-rev0, Built by MinGW-W64 project)
```

dhylands, 2016-08-25:
Yeah - I think that's a command line length problem.

I measured from the if to the fi and got 8397 characters. I think Windows has a limit of 8K (8192)

buaabyl, 2016-08-25:
Compiled under MSYS with bash, everything ok......

buaabyl, 2016-08-25:
https://support.microsoft.com/en-us/kb/830473
> Even though the Win32 limitation for environment variables is 32,767 characters, Command Prompt ignores any environment variables that are inherited from the parent process and are longer than its own limitations of either 2047 or 8191 characters

stinos, 2016-08-25:
> I measured from the if to the fi and got 8397 characters

I have no clue how make interacts with the command line, but does that matter? The if and fi get interpreted by the makefile so I would guess it just prints to stdout, there shouldn't be a limit on that? And gcc itself is invoked as a process not through cmd? Just guessing here..

Anyway if this is the problem it should be fixable by splitting up some stuff maybe? Like storing the condition for the if in a variable?

dhylands, 2016-08-25:
Everything from the if to the fi is considered as a single line by make (which is why all the backslashes are required), and it launches the shell to process it. So the if is a bash if.

You can check $(MAKEFLAGS) to see if it contains a capital B rather than checking $?
This will cut down the size by a third. And that check can be done using a make if statement which means that each of the gcc lines can be a single command. So that gets the length of each gcc line down to roughly a third of the current line.

buaabyl, 2016-08-26:
Summary:
1. make create a bat file when need to check something.
2. windows bat have limit 8K line length.
3. this generated bat file is bash syntax.

This means that the Makefile need MSYS, So I come back to use my python script.

when "make -d", it show

```
Creating temporary batch file Z:\Temp\make4572-2.bat
CreateProcess(Z:\Temp\make4572-2.bat,Z:\Temp\make4572-2.bat,...)
```

dig in to make source...

make/job.c
```c
static char **
construct_command_argv_internal (char *line, char **restp, const char *shell,
                                 const char *shellflags, const char *ifs,
                                 int flags, char **batch_filename UNUSED)
{

            /* Generate a file name for the temporary batch file.  */
            sprintf (fbuf, "make%d", id);
            *batch_filename = create_batch_file (fbuf, 0, &temp_fd);
            DB (DB_JOBS, (_("Creating temporary batch file %s\n"),
                          *batch_filename));

            /* Create a FILE object for the batch file, and write to it the
               commands to be executed.  Put the batch file in TEXT mode.  */
            _setmode (temp_fd, _O_TEXT);
            batch = _fdopen (temp_fd, "wt");
            fputs ("@echo off\n", batch);
            DB (DB_JOBS, (_("Batch file contents:\n\t@echo off\n")));
...
          DB (DB_JOBS, (_("Cleaning up temp batch file %s\n"),
                        c->sh_batch_file));

          errno = 0;
          rm_status = remove (c->sh_batch_file);
          if (rm_status)
            DB (DB_JOBS, (_("Cleaning up temp batch file %s failed (%d)\n"),
                          c->sh_batch_file, errno));
```

make/w32/subproc/subproc.c
```c
                DB (DB_JOBS, ("CreateProcess(%s,%s,...)\n",
                        exec_path ? exec_path : "NULL",
                        command_line ? command_line : "NULL"));
                if (CreateProcess(
                        exec_path,
                        command_line,
```

make full log
```sh
Live child 0215FFB8 (build/genhdr/qstr.i.last) PID 34578088 
"GEN build/genhdr/qstr.i.last"
Reaping winning child 0215FFB8 PID 34578088 
Cleaning up temp batch file Z:\Temp\make4572-1.bat
Creating temporary batch file Z:\Temp\make4572-2.bat
Batch file contents:
	@echo off
	if [ "../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/lexer.c ../py/lexerstr.c ../py/lexerunix.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmx64.c ../py/asmx86.c ../py/asmthumb.c ../py/emitinlinethumb.c ../py/asmarm.c ../py/formatfloat.c ../py/parsenumbase.c ../py/parsenum.c ../py/emitglue.c ../py/runtime.c ../py/runtime_utils.c ../py/nativeglue.c ../py/stackctrl.c ../py/argcheck.c ../py/warning.c ../py/map.c ../py/obj.c ../py/objarray.c ../py/objattrtuple.c ../py/objbool.c ../py/objboundmeth.c ../py/objcell.c ../py/objclosure.c ../py/objcomplex.c ../py/objdict.c ../py/objenumerate.c ../py/objexcept.c ../py/objfilter.c ../py/objfloat.c ../py/objfun.c ../py/objgenerator.c ../py/objgetitemiter.c ../py/objint.c ../py/objint_longlong.c ../py/objint_mpz.c ../py/objlist.c ../py/objmap.c ../py/objmodule.c ../py/objobject.c ../py/objpolyiter.c ../py/objproperty.c ../py/objnone.c ../py/objnamedtuple.c ../py/objrange.c ../py/objreversed.c ../py/objset.c ../py/objsingleton.c ../py/objslice.c ../py/objstr.c ../py/objstrunicode.c ../py/objstringio.c ../py/objtuple.c ../py/objtype.c ../py/objzip.c ../py/opmethods.c ../py/sequence.c ../py/stream.c ../py/binary.c ../py/builtinimport.c ../py/builtinevex.c ../py/modarray.c ../py/modbuiltins.c ../py/modcollections.c ../py/modgc.c ../py/modio.c ../py/modmath.c ../py/modcmath.c ../py/modmicropython.c ../py/modstruct.c ../py/modsys.c ../py/moduerrno.c ../py/modthread.c ../py/vm.c ../py/bc.c ../py/showbc.c ../py/repl.c ../py/smallint.c ../py/frozenmod.c ../py/../extmod/moductypes.c ../py/../extmod/modujson.c ../py/../extmod/modure.c ../py/../extmod/moduzlib.c ../py/../extmod/moduheapq.c ../py/../extmod/moduhashlib.c ../py/../extmod/modubinascii.c ../py/../extmod/virtpin.c ../py/../extmod/machine_mem.c ../py/../extmod/machine_pinbase.c ../py/../extmod/machine_pulse.c ../py/../extmod/machine_i2c.c ../py/../extmod/modussl_axtls.c ../py/../extmod/modurandom.c ../py/../extmod/modwebsocket.c ../py/../extmod/modwebrepl.c ../py/../extmod/modframebuf.c ../py/../extmod/fsusermount.c ../py/../extmod/vfs_fat.c ../py/../extmod/vfs_fat_ffconf.c ../py/../extmod/vfs_fat_diskio.c ../py/../extmod/vfs_fat_file.c ../py/../extmod/vfs_fat_lexer.c ../py/../extmod/vfs_fat_misc.c ../py/../extmod/moduos_dupterm.c ../py/../lib/embed/abort_.c ../py/emitnative.c ../unix/main.c ../unix/file.c ../unix/input.c ../unix/modos.c ../unix/modmachine.c ../unix/modtime.c ../unix/gccollect.c windows_mphal.c realpath.c init.c sleep.c ../lib/mp-readline/readline.c" = "" ]; then     echo "QSTR Looks like -B used, trying to emulate";     gcc -E -DNO_QSTR -DN_X64 -DN_X86 -DN_THUMB -DN_ARM -Ibuild/tmp -I../lib -I../lib/netutils -I. -I.. -Ibuild -Wall -Wpointer-arith -Werror -ansi -std=gnu99 -DUNIX -D__USE_MINGW_ANSI_STDIO=1 -DMICROPY_USE_READLINE=1 -Os  ../py/mpstate.c ../py/malloc.c ../py/gc.c ../py/qstr.c ../py/vstr.c ../py/mpprint.c ../py/unicode.c ../py/mpz.c ../py/lexer.c ../py/lexerstr.c ../py/lexerunix.c ../py/parse.c ../py/scope.c ../py/compile.c ../py/emitcommon.c ../py/emitbc.c ../py/asmx64.c ../py/asmx86.c ../py/asmthumb.c ../py/emitinlinethumb.c ../py/asmarm.c [... cut ...] /modmachine.c ../unix/modtime.c ../unix/gccollect.c windows_mphal.c realpath.c init.c sleep.c ../lib/mp-readline/readline.c >build/genhdr/qstr.i.last; fi
CreateProcess(Z:\Temp\make4572-2.bat,Z:\Temp\make4572-2.bat,...)
Live child 0215FFB8 (build/genhdr/qstr.i.last) PID 34578088 
Reaping losing child 0215FFB8 PID 34578088 
Cleaning up temp batch file Z:\Temp\make4572-2.bat
../py/mkrules.mk:73: recipe for target 'build/genhdr/qstr.i.last' failed
Removing child 0215FFB8 PID 34578088 from chain.
```

closed 2016-08-26 by buaabyl

buaabyl, 2016-08-26:
I will try to write a CMake script later, maybe that is better for cross platform.

dhylands, 2016-08-26:
This patch will reduce the command line length to 1/3 of what it was before:
```
diff --git a/py/mkrules.mk b/py/mkrules.mk
index 3cf0e30..428e07d 100644
--- a/py/mkrules.mk
+++ b/py/mkrules.mk
@@ -71,12 +71,12 @@ $(OBJ): | $(HEADER_BUILD)/qstrdefs.generated.h $(HEADER_BUILD)/mpversion.h
 
 $(HEADER_BUILD)/qstr.i.last: $(SRC_QSTR) | $(HEADER_BUILD)/mpversion.h
        $(ECHO) "GEN $@"
-       $(Q)if [ "$?" = "" ]; then \
-           echo "QSTR Looks like -B used, trying to emulate"; \
-           $(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $^ >$(HEADER_BUILD)/qstr.i.last; \
-       else \
-           $(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $? >$(HEADER_BUILD)/qstr.i.last; \
-       fi
+ifeq ($(findstring B,$(MAKEFLAGS)),B)
+       echo "QSTR Looks like -B used, trying to emulate"
+       $(Q)$(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $^ >$(HEADER_BUILD)/qstr.i.last
+else
+       $(Q)$(CPP) $(QSTR_GEN_EXTRA_CFLAGS) $(CFLAGS) $? >$(HEADER_BUILD)/qstr.i.last
+endif
 
 $(HEADER_BUILD)/qstr.split: $(HEADER_BUILD)/qstr.i.last
        $(ECHO) "GEN $@"
```