micropython/micropython #2351: unix: Port won't build on current OSX

blmorris, 2016-08-24:
Been away from MicroPython development for a little while, and ran into a problem compiling the unix port on OSX after updating to the current master. OS version is 10.11.6, and clang version info here:
```
$ clang -v
Apple LLVM version 7.0.2 (clang-700.1.81)
Target: x86_64-apple-darwin15.6.0
```

The build process fails with one error that gets triggered several times building different files; the error is due to the redefinition of the 'memmove' macro in 
`lib/berkeley-db-1.xx/PORT/include/compat.h`:

```
In file included from ../lib/berkeley-db-1.xx/mpool/mpool.c:46:
In file included from ../lib/berkeley-db-1.xx/PORT/include/db.h:46:
../lib/berkeley-db-1.xx/PORT/include/compat.h:132:9: error: 'memmove' macro redefined [-Werror,-Wmacro-redefined]
#define memmove(a, b, n)        bcopy(b, a, n)
        ^
/usr/include/secure/_string.h:70:9: note: previous definition is here
#define memmove(dest, src, len)                                 \
        ^
1 error generated.
make: *** [build/genhdr/qstr.i.last] Error 1
make: *** Deleting file `build/genhdr/qstr.i.last'
```

I can work around the error by commenting out line 132 of 
`lib/berkeley-db-1.xx/PORT/include/compat.h`:
```
#if defined(BSD) || defined(BSD4_3)
#define	strchr(a, b)		index(a, b)
#define	strrchr(a, b)		rindex(a, b)
#define	memcmp(a, b, n)		bcmp(a, b, n)
/*#define	memmove(a, b, n)	bcopy(b, a, n)*/
#endif
```

The resulting binary passes the test suite.  (In particular, `extmod/btree1.py` passes.)

I'm guessing that this isn't the correct fix. Also, given that `lib/berkeley-db-1.xx` is a submodule, it might not even be the right place to report, but I'm not sure the what the right way is to deal with submodule build issues.

pfalcon, 2016-08-24:
Thanks for the report, this definitely needs to be looked into.

pfalcon, 2016-08-24:
Actually, this is dup of https://github.com/micropython/micropython/issues/2289, sorry.

closed 2016-08-24 by pfalcon

blmorris, 2016-08-24:
Yes, didn't look back far enough and missed that report. Looks like you are on it, and I have some catching up to do. Thanks!