micropython/micropython #2352: esp8266 read vcc

ssendev, 2016-08-24:
The esp8266 can be configured in a way that makes reading VCC (battery) voltage with the ADC possible without needing an external voltage divider or other stuff.

[here](https://github.com/esp8266/Arduino/issues/225#issuecomment-222520821) is an example Arduino sketch that prints VCC.

I tried to follow the code but `ADC_MODE(ADC_VCC);` leads to [here](https://github.com/esp8266/Arduino/blob/5dd6acc4c20e2dfa199ae3f21b4e2bd9a8e66bb8/cores/esp8266/core_esp8266_phy.c#L258) and the only phy stuff in micropython i could find is [this](https://github.com/micropython/micropython/blob/822e9ca8f3d2f63c7477bb7fb859b5d9ad948a3b/esp8266/modnetwork.c#L432) which might be related since there are [reports](https://forum.micropython.org/viewtopic.php?t=1906#p10778) that reading VCC needs RF enabled.

The other ting needed is `system_get_vdd33` which is somewhere in the espressif sdk which i didn't see in the open sdk but maybe i didn't look in the right submodule.

espressif [docs](https://github.com/esp8266/Arduino/issues/338#issuecomment-113099775)

Here are nodemcu [docs](http://nodemcu.readthedocs.io/en/dev/en/modules/adc/#adcreadvdd33) and [code](https://github.com/nodemcu/nodemcu-firmware/blob/4911d2db8bafedd3fcc214241a8f7b2e340c8e2a/app/modules/adc.c)

So it should be possible to implement an `esp.getVCC()` or similar function.

dpgeorge, 2016-08-25:
The VCC is available on channel 1 of the ADC.  You can access it this way:
```
import machine
vcc = machine.ADC(1)
vcc.read()
```
The read() call is calling system_get_vdd33(), which is what you want.  But on my board it always returns 65535 (with RF enabled or disabled).

deshipu, 2016-08-25:
If I remember correctly, there is a compile-time option that needs to be set in the firmware for the `get_vdd33` to work (and then the normal ADC doesn't work). That's why the Arduino has it as a macro, not a function call, and why you have to put it on top of your sketch. (https://github.com/esp8266/Arduino/blob/master/doc/reference.md#analog-input)

I'm not sure why it's done that way, but I suspect that the ADC initialization at startup is different for those two cases. I have no idea if it's possible to re-do the initialization to switch that on runtime, but it would be great. See discussion at https://github.com/esp8266/Arduino/issues/225

deshipu, 2016-08-25:
More details http://bbs.espressif.com/viewtopic.php?t=827

ssendev, 2016-08-25:
that's why nodemu has http://nodemcu.readthedocs.io/en/dev/en/modules/adc/#adcforce_init_mode

ssendev, 2016-08-25:
I found the missing piece `esp_init_data_default.bin` on https://nodemcu.readthedocs.io/en/dev/en/flash/#upgrading-from-sdk-09x-firmware

ssendev, 2016-08-25:
It's alive

```python
import esp
import machine
sector_size = 0x1000
flash_size = 0x100000 # device dependent
init_sector = int(flash_size / sector_size - 4)
data = bytearray(esp.flash_read(init_sector * sector_size, sector_size))
data[107] = 255
esp.flash_erase(init_sector)
esp.flash_write(init_sector * sector_size, data)
machine.restart()
```

```python
from machine import ADC
ADC(1).read()
```

Is there already a way to detect the flash size? otherwise this is good to go.

dpgeorge, 2016-08-26:
> is there a way to add this to esp or ADC in python or does it have to be written in c?

It sounds like you have solved the problem purely in Python.  What do you mean by "is there a way to add this", what do you want to add?

ssendev, 2016-08-26:
add `esp.adc_mode(ADC_VCC)` to micopython

dpgeorge, 2016-08-26:
> add esp.adc_mode(ADC_VCC) to micopython

Since this function works fine written in Python it would be best to keep the implementation in Python.  Then since the esp module is pure C the adc_mode function would need to go in a separate module (eg espconfig).  @pfalcon thoughts?

pfalcon, 2016-08-26:
Suggested further direction for this ticket: try readvdd33() as mentioned in one of the links above, and share the findings (and likely, a patch).

ssendev, 2016-08-26:
The code i posted works. The question is where to add the function

askvictor, 2016-10-01:
I'm finding the reading from this function is off; on two different devices I'm getting about 200mV difference, but in different directions on either device (one reads 200mV over, the other 200mV under). Is there a way to calibrate the voltage internally? Otherwise, it shouldn't be too hard to write a calibration function, but will mean a bit of messing around.

eduvik, 2016-10-03:
@ssendev - it would be nice to have the function check the value of this in flash, and only write if it needs to be changed (this would extend the life of the flash memory)

diginfo, 2017-03-05:
apologies for picking up this issue again after so many months, and thanks for the info for enabling ADC(1).

I have read the espressif documentation and it mentions that the adc pin should be left '_**dangled**_' which I presume means floating ??

When I do this I get a value of 3547 on one specific ESP-12, however I have verified that vcc pin is pretty much exactly 3.3v, so even using 1/1024 per bit calculation 3.464 volts which is pretty inaccurate and I guess will be different device to device.

So is it necessary to individually calibrate the conversion factor, which in this case = 0.93 or am I misunderstanding / miscalculating the results ??

If a manual calibration is required, does this value have to be managed manually or can a calibration factor be applied to the device's registers ?

DocSneider, 2019-07-26:
> apologies for picking up this issue again after so many months, and thanks for the info for enabling ADC(1).
>
> I have read the espressif documentation and it mentions that the adc pin should be left [...]

Same situation here, I would also be interested in how to get the calibration done :-)

jonnor, 2024-07-27:
There seems to be a plain Python way of doing this functionality, which is provided further up in the thread. Proposing to close this issue.

robert-hh, 2024-07-27:
> which is provided further up in the thread.

closed 2024-07-27 by robert-hh