micropython/micropython #2355: stmhal: problem with sockets swept by GC

blazewicz, 2016-08-25:
Let me start with an example:

## Code

### Pyboard
```Python
import socket
import gc

addr = ('192.168.1.100', 8000)

def run():
    s = socket.socket()  # create new object with finaliser
    s.connect(addr)   # bind the socket object with actual physical socket
    s.send(b'foo') 
    gc.collect()  # simulate memory consuming work which triggers GC
    s.send(b'bar')  # ENOTCONN
    s.close()  # close the socket after job is done
    # `s` goes out of scope, ready to be swept by GC
```

### Host
```Python
import socket

addr = ('0.0.0.0', 8000)
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
s.bind(addr)
s.listen(1)

while True:
    con, src = s.accept()
    print('[connection from %s:%d]' % src)
    while True:
        msg = con.recv(1024)
        if msg == b'':
            print('[connection closed by peer]')
            break
        print(msg)
```

## Output

### Pyboard
```
>>> run()
>>> run()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 6, in run
OSError: [Errno 107] ENOTCONN
```

### Host
```
[connection from 192.168.1.101:49157]
b'foo'
b'bar'
[connection closed by peer]
[connection from 192.168.1.101:49158]
b'foo'
[connection closed by peer]
```

## Explanation

In example above on first call to `run()`, variable `s` is bound to Wiznets socket number 0, after the work is done and `s.close()` is called, this socket gets marked as free to use. The local variable `s` goes out of scope.

On second call to `run()`, Wiznets socket number 0 is the first free to use and is again bound to new local viarable `s`. Now when GC triggers, it finds the old `s` and calls its finaliser: `s.__del__()` which is identical to `s.close()` (`socket_close` in _modusocket.c_).

The problem is that both old and new `s` are bound to the same Wiznets socket, therefore GC unintentionally closes the socket that is actually in use and we get unexpected ENOTCONN when we try to use it.

This problem basically makes it impossible to use sockets as local variables as we may never know when and which socket will be unexpectedly closed by GC.

## Workaround / Solution
The workaround might be to always use sockets as global variables, this way GC will never sweep them.

```Python
s = socket.socket()  # will never go out of scope
def run():
    s.connect(addr)
    s.send(b'foo')
    gc.collect()
    s.send(b'bar')
    s.close() 
```

But then, why do we even need a finaliser?

We could also explicitly trigger GC just before new socket is created, but it doesn't sound right to me.

Or apply this little patch:
```Diff
diff --git a/stmhal/modusocket.c b/stmhal/modusocket.c
index fbd1196..c0e30ce 100644
--- a/stmhal/modusocket.c
+++ b/stmhal/modusocket.c
@@ -83,6 +83,7 @@ STATIC mp_obj_t socket_close(mp_obj_t self_in) {
     mod_network_socket_obj_t *self = self_in;
     if (self->nic != MP_OBJ_NULL) {
         self->nic_type->close(self);
+        self->nic = MP_OBJ_NULL;
     }
     return mp_const_none;
 }
```

Then GC will close only a socket for which .close() wasn't explicitly called before it got out of scope.

Here comes another problem - use after close() (which currently is possible altough uPy and mainstream Python docs says it shouldn't), but that might be a topic for another issue.

Reference:
[https://docs.python.org/3/library/socket.html#socket.socket.close]

dhylands, 2016-08-25:
I think that your patch is the right solution. The python object (mod_network_socket_obj_t) needs to know whether it's closed the underlying socket yet or not.

The finalizer is to cover the case where s goes out of scope and you didn't close it. Then when s is gc'd the finalizer will be called, which will in turn close the socket (otherwise s would get collected and we'd then leak a socket since the underlying socket would never get closed).

blazewicz, 2016-08-25:
@dhylands I understand why we have the finaliser here, I just think it doesn't work well in this situation

Such behavior (blindly close sockets that went out of scope) makes sense for *nix, where sockets are software abstracts, not here where we have limited pool (8 for w5500) of hardware sockets.

Taking it even further, I think there should be some explicit way of marking socket (the software layer) as closed so it can't be used again and should not be closed by GC.

@dpgeorge what do you think about that?

I'm willing to prepare a PR if we agree on something.

dhylands, 2016-08-25:
> Such behavior (blindly close sockets that went out of scope) makes sense for *nix, where sockets are software abstracts, not here where we have limited pool (8 for w5500) of hardware sockets.

Which seems like all the more reason to close them when they go out of scope. Otherwise you'll very quickly run out of sockets. If you don't want the socket to be closed, don't let s go out of scope (i.e. by making s global).

> Taking it even further, I think there should be some explicit way of marking socket (the software layer) as closed so it can't be used again and should not be closed by GC.

It seems to me that setting self->nic to MP_OBJ_NULL (which is what your patch does) is the correct way of doing this.

blazewicz, 2016-08-25:
> Which seems like all the more reason to close them when they go out of scope. Otherwise you'll very quickly run out of sockets. If you don't want the socket to be closed, don't let s go out of scope (i.e. by making s global).

But its lame solution if it causes fresh sockets to be closed with no apparent reason, don't you think?

> It seems to me that setting self->nic to MP_OBJ_NULL (which is what your patch does) is the correct way of doing this.

Not exactly, of course it will prevent GC from closing this socket, but it won't prevent from further use as we'd expect it to.

For example:
```Python
...
s.close()
s.connect(addr)
```

will throw OSError 97 [EAFNOSUPPORT], not beacause `self->nic == MP_OBJ_NULL` but because `self->u_param.domain != MOD_NETWORK_AF_INET` (`wiznet5k_socket_socket` in _modnwwiznet5k.c_)

This comes from fact, that socket is bound to NIC on connection attempt, not on socket creation.

Thats why my patch requires some further work, take it as a concept or a hacky-fix :)

blazewicz, 2016-08-25:
Of course we should always remember what the docs say:

> Sockets are automatically closed when they are garbage-collected, **but it is recommended to close() them explicitly, or to use a with statement around them.**

So running out of sockets is always an easy-fix in Python code, topic of this issue is rather implementation problem.

pfalcon, 2016-08-25:
@blazewicz, you seem to overcomplicate the issue. "Fresh sockets aren't closed with no apparent reason". Sockets which can't be referenced are garbage-collected. This is especially important for embedded environments with scarce resources, where otherwise you'll run out of resources and will need to reboot after few attempts. Beyond that, it should be possible to close an already closed socket, and it should be no-op. If your patch does *that*, it's good.

dpgeorge, 2016-08-26:
The proposed patch is fine.  It allows you to call close again on the socket (and will be a no-op on subsequent calls).  You can't make any other calls on the socket after it's closed, but that's not something you should be doing anyway.

The alternative is to apply a similar patch to wiznet5k_socket_close to invalidate the socket number after it is closed (eg `socket->u_param.fileno = _WIZCHIP_SOCK_NUM_`).  But such a patch then needs to be applied to all NIC types, whereas the proposed patch above needs only to be done once, then all NICs work in the same way.

PR #2357 (2016-08-26): stmhal/modusocket: set self->nic to MP_OBJ_NULL after socket close
This patch makes second and next calls to <socket>.close() a no-op.
It prevents GC from closing the underlying resource after user
already used <socket>.close() explicitly.

fixes #2355
commit fc73c9b4 (2016-08-26): stmhal/modusocket: set self->nic to MP_OBJ_NULL after socket close

This patch makes second and next calls to <socket>.close() a no-op.
It prevents GC from closing the underlying resource after user
already used <socket>.close() explicitly.

fixes #2355
closed 2016-08-26 by dpgeorge