micropython/micropython #2361: Unix: Use *printf() function provided by the OS

dxxb, 2016-08-26:
I am loading a shared library using FFI which uses vsnprintf(), micropython defines it's own vsnprintf() which ultimately calls mp_vprintf() and the dynamic linker decides to use micropython's vsnprintf() which has undefined behaviour for unsupported format specifiers.

So my options are:
- Make micropython use libc's *printf() functions. How do I do that? I tried removing lib/utils/printf.c from LIB_SRC_C and it seems to work on but it probably has side effects I am unaware of.
- Hide *printf() symbols using visibility options when linking micropython (I tried, no joy)
- Implement format specifiers in mp_vprintf() each time I stumble upon an unsupported one I need.

I figured the best option is to make micropython use libc's implementation because it doesn't require changes to mp_vprintf(). Any advice please? Thank you.

pfalcon, 2016-08-26:
> I am loading a shared library using FFI which uses vsnprintf(), micropython defines it's own vsnprintf() which ultimately calls mp_vprintf() and the dynamic linker decides to use micropython's vsnprintf() which has undefined behaviour for unsupported format specifiers.

Which platform is that? That's pretty weird, that might work that way on a normal ELF system if MicroPython exported dynamic symbols, but it doesn't. The most obvious solution would be still to look how instruct dynamic linker to not do that.

> I figured the best option is to make micropython use libc's implementation because it doesn't require changes to mp_vprintf(). Any advice please?

When we switched to own printf() implementation, one of the requirements was to still support libc's. I don't remember anything of that now though, so commit messages and github tickets are your best source of information (I'd need to do the same to answer your question, so please try yourself first).

dhylands, 2016-08-26:
I also don't see anyplace in the code that MicroPython actually uses vsnprintf.

And in fact, I was able to #if 0 out snprintf and vsnprintf and the unix version still builds.

dpgeorge, 2016-08-27:
> It looks like stmhal still references snprintf in a couple places

cc3200 uses snprintf quite a bit.

> There are a few more places that some of the stmhal code calls printf, but with a bit of cleanup we can probably eliminate lib/utils/printf.c entirely.

printf.c is needed for the printf function, which is useful for debugging on stmhal.  It's also generally useful for ports that don't have (or don't want to use) the C library version.

dxxb, 2016-08-29:
@pfalcon 
> Which platform is that?

```
# file unix/micropython
unix/micropython: ELF 32-bit LSB executable, ARM, version 1 (SYSV), dynamically linked (uses shared libs)
```

> That's pretty weird, that might work that way on a normal ELF system if MicroPython exported dynamic symbols, but it doesn't.

Actually it does:
```
# arm-none-linux-gnueabi-objdump -T unix/micropython | grep printf
00045690 g    DF .text  00000030  Base        printf
0004567c g    DF .text  00000014  Base        vprintf
00045628 g    DF .text  00000024  Base        snprintf
000455cc g    DF .text  0000005c  Base        vsnprintf
```

and it does so also here:
```
# file unix/micropython
unix/micropython: ELF 32-bit LSB  executable, Intel 80386, version 1 (SYSV), dynamically linked (uses shared libs)
# objdump -T unix/micropython | grep printf
0807c02d g    DF .text  00000018  Base        vprintf
0807c0e7 g    DF .text  0000001a  Base        snprintf
0807c014 g    DF .text  00000019  Base        printf
0807c09a g    DF .text  0000004d  Base        vsnprintf
```

> The most obvious solution would be still to look how instruct dynamic linker to not do that.

It looks like stdio.h ends up being included on unix builds even if micropython uses its own *printf() implementation but prototypes in stdio.h declare *printf() as extern so that's why -fvisibility=hidden didn't work when I tried it.

> Be aware that headers from outside your project, in particular system headers and headers from any other library you use, may not be expecting to be compiled with visibility other than the default. You may need to explicitly say #pragma GCC visibility push(default) before including any such headers.

> extern declarations are not affected by -fvisibility, so a lot of code can be recompiled with -fvisibility=hidden with no modifications. However, this means that calls to extern functions with no explicit visibility use the PLT, so it is more effective to use __attribute ((visibility)) and/or #pragma GCC visibility to tell the compiler which extern declarations should be treated as hidden.

(see https://gcc.gnu.org/onlinedocs/gcc/Code-Gen-Options.html)

So I am left with the impression that using micropython's *printf() implementation in the unix port needs work and that the sensible thing would be to use libc's implementation by default. (my 2c)

To solve my issue I decided I want to link with libc's implementation which I think should be available as an option (I do not care if it's the default for unix or not). As far as I can tell to do so it suffices to remove utils/printf.c from LIB_SRC_C in the Makefile. Would a pull request implementing this, i.e. adding utils/printf.c to LIB_SRC_C only if (say) MICROPY_USE_LIBC_PRINTF equals 1, be accepted?

Thank you

dxxb, 2016-08-31:
@pfalcon, ping :)

pfalcon, 2016-08-31:
> Would a pull request implementing this, i.e. adding utils/printf.c to LIB_SRC_C only if (say) MICROPY_USE_LIBC_PRINTF equals 1, be accepted?

Yes, if it's done cleanly, and doesn't affect size of the minimal builds (or decreases it).

dxxb, 2016-09-01:
See https://github.com/micropython/micropython/pull/2377

dxxb, 2016-09-05:
Something similar was implemented by e2ac8bb

closed 2016-09-05 by dxxb