micropython/micropython #2362: pixel() in extmod/modframebuf.c returning unexpected result

mcauser, 2016-08-26:
`framebuf.pixel()` is returning results different to what is being set.

Given a simple 16x16 buffer.

```
>>> import framebuf
>>> width = 16
>>> height = 16
>>> pages = height // 8
>>> buffer = bytearray(pages * width)
>>> framebuf = framebuf.FrameBuffer1(buffer, width, height)
```

Set 3 pixels to black:

```
>>> framebuf.pixel(0,0,0)
>>> framebuf.pixel(0,1,1)
>>> framebuf.pixel(0,2,1)
>>> framebuf.pixel(0,3,1)
```

Then retrieve them:

```
>>> framebuf.pixel(0,0)
14
>>> framebuf.pixel(0,1)
7
>>> framebuf.pixel(0,2)
3
>>> framebuf.pixel(0,3)
1
```

Expected to see:

```
>>> framebuf.pixel(0,0)
0
>>> framebuf.pixel(0,1)
1
>>> framebuf.pixel(0,2)
1
>>> framebuf.pixel(0,3)
1
```

I can get the expected result if I AND them with 1:

```
>>> framebuf.pixel(0,0) & 1
0
>>> framebuf.pixel(0,1) & 1
1
>>> framebuf.pixel(0,2) & 1
1
>>> framebuf.pixel(0,3) & 1
1
```

Am I wrong in my expectation that `col` in `pixel(0,1,col)` and `col = pixel(0,1)` should match?
Is it useful for pixel() to return its adjacent bits?

dpgeorge, 2016-08-27:
Thanks for the report, it's a bug :)  Fixed in 531217a06b54d612ac49c341a358b3fca1d59985.

closed 2016-08-27 by dpgeorge

mcauser, 2016-08-29:
Using the framebuf to power Conway's GoL on a 5110 lcd:
https://github.com/mcauser/MicroPython-ESP8266-Nokia-5110-Conways-Game-of-Life