micropython/micropython #2370: ESP8266: onewire, read DS18S20 sensor with DS18B20 class?

ast0815, 2016-08-28:
Hello,

is it possible to read out DS18S20 sensors with the DS18B20 class (note the different letter in the middle)?
I'm trying to read a DS18S20 with the DS18B20 class, but it does not work and I wonder whether that is expected behaviour or something is wrong.
When I call `scan()` from the bare `OneWire` object, it lists one device as expected.
But if I call `scan()` from the `DS18B20` object, I get an empty list.
Also trying to use the returned rom of the `OneWire` object with the `DS18B20` object's `read_temp` gives me wrong readings, even after calling `convert_temp()`.

I'm using a NodeMcu V3 board with the latest firmware blob from the website. The sensor is powered with 3.3 V and I connected the data line to V+ with a 5 kOhm resistor. I tried multiple IO pins, but the behaviour is the same.

Trebastian, 2016-08-28:
The __init__ function of the module
```
def __init__(self, pin):
        self.ow = OneWire(pin)
        # Scan the 1-wire devices, but only keep those which have the
        # correct # first byte in their rom for a DS18x20 device.
        self.roms = [rom for rom in self.ow.scan() if rom[0] == 0x10 or rom[0] == 0x28]
```

only works if the first byte is 0x10 or 0x28. What is the output of onewire.scan() for your sensor?

dpgeorge, 2016-08-29:
I have updated the 1-wire driver to be able to read from the DS18S20 devices (as well as the DS18B20); see 59a9509703c6e525b3419ffea1b69f3b6992e006.  Please try the latest version of the code.

ast0815, 2016-08-30:
Just tested it. Works like a charm!

Thanks for the very fast fix!

closed 2016-08-30 by ast0815