micropython/micropython #2372: esp8266: can't seem to see nearest wifi network?

mheld, 2016-08-29:
### Overview

I can’t for the life of me seem to see my personal wifi network on micropython (while the arduino example wifiscan sketch seems to always have it).

### The Setup
- Using an esp8266 module (elecrow IoT board)
- The name of my network is weft-2.4 (it’s broadcasting on a cruddy xfinity AP on 802.11g) and the following are the steps I use to scan

### Fresh install of micropython:
esptool.py --port /dev/tty.SLAB_USBtoUART erase_flash
esptool.py --port /dev/tty.SLAB_USBtoUART --baud 115200 write_flash --flash_size=32m 0 Downloads/esp8266-20160825-v1.8.3-49-ga589fa3.bin

### Now in console:
(please excuse the formatting -- trying to make it as legible as possible but backticks are everywhere!)
MicroPython v1.8.3-49-ga589fa3 on 2016-08-25; ESP module with ESP8266
Type "help()" for more information.
`>>> import time`
`>>> import network`
`>>> wlan = network.WLAN(network.STA_IF)`
`>>> wlan.active(True)`
`mode : sta(5c:cf:7f:80:1a:01) + softAP(5e:cf:7f:80:1a:01)`
`#6 ets_task(4020e33c, 28, 3fff7ec0, 10)`
`add if0`
`>>> wlan.isconnected()`
`False`
`>>> wlan.scan()`
`scandone`
`[(b'Zoomc36a', b'\x00\x1c{\xf3\x9b\x14', 1, -75, 4, 0), (b'FQ1601', b'\x00q\xc2\xe0\xc1\xf8', 1, -92, 4, 0), (b'xfinitywifi', b' %dd"\xa2', 1, -93, 0, 0), (b'xfinitywifi', b'\x00q\xc2\xe0\xc1\xfa', 1, -90, 0, 0), (b'cooper', b' %dd"\xa0', 1, -92, 4, 0), (b'NETGEAR22', b'\xe4\xf4\xc6\x14\xe0K', 3, -81, 3, 0), (b'Biesheuvel', b'\x14\xcc p{`', 4, -91, 3, 0), (b'JCD123', b'\x8c\x04\xff\xee\xa1\xaf', 1, -91, 4, 0), (b'Vince2', b'\x00%\x9c\xff\x08\xe3', 6, -62, 1, 0), (b'Spinaltap-2.4', b'`\x02\x92\xae\xf48', 6, -64, 4, 0), (b'xfinitywifi', b'`\x02\x92\xae\xf4:', 6, -64, 0, 0), (b'Mountain_Time', b'\xbc\xca\xb5&\xbc\x90', 6, -71, 3, 0), (b'xfinitywifi', b'\xb6\xca\xb5&\xbc\x90', 6, -72, 0, 0), (b'xfinitywifi', b'\x8e\x04\xff\xee\xa1\xa1', 1, -91, 0, 0), (b'FQ1603', b'\xd8\x97\xba\xcb,\x08', 11, -54, 4, 0), (b'xfinitywifi', b'\xd8\x97\xba\xcb,\n', 11, -55, 0, 0), (b'Boston1705', b'`\x02\x92\xf3\x86x', 11, -88, 4, 0), (b'GingerPanda', b'<z\x8a\x04\xd0z', 11, -73, 4, 0), (b'xfinitywifi', b'Nz\x8a\x04\xd0z', 11, -75, 0, 0)]`
`>>> len(_)`
`19`
`>>> time.sleep(5)`
`>>> wlan.scan()`
`scandone`
`[(b'xfinitywifi', b'\xc0|\xd1\x84\xda\x8a', 1, -51, 0, 0), (b'FQ1601', b'\x00q\xc2\xe0\xc1\xf8', 1, -91, 4, 0), (b'xfinitywifi', b'\x00q\xc2\xe0\xc1\xfa', 1, -88, 0, 0), (b'xfinitywifi', b'\x8e\x04\xff\xee\xa1\xa1', 1, -96, 0, 0), (b'NETGEAR22', b'\xe4\xf4\xc6\x14\xe0K', 3, -79, 3, 0), (b'Biesheuvel', b'\x14\xcc p{`', 4, -89, 3, 0), (b'Vince2', b'\x00%\x9c\xff\x08\xe3', 6, -63, 1, 0), (b'Spinaltap-2.4', b'`\x02\x92\xae\xf48', 6, -67, 4, 0), (b'xfinitywifi', b'`\x02\x92\xae\xf4:', 6, -67, 0, 0), (b'Mountain_Time', b'\xbc\xca\xb5&\xbc\x90', 6, -77, 3, 0), (b'xfinitywifi', b'\xb6\xca\xb5&\xbc\x90', 6, -76, 0, 0), (b'FQ1603', b'\xd8\x97\xba\xcb,\x08', 11, -55, 4, 0), (b'xfinitywifi', b'\xd8\x97\xba\xcb,\n', 11, -54, 0, 0), (b'GingerPanda', b'<z\x8a\x04\xd0z', 11, -75, 4, 0), (b'xfinitywifi', b'Nz\x8a\x04\xd0z', 11, -78, 0, 0)]`
`>>> len(_)`
`15`

### Trying the WifiScan arduino example sketch shows me these results in the serial monitor:

tail 8
chksum 0x2d
csum 0x2d
v09f0c112
scan start
scan done
23 networks found
1: Zoomc36a (-74)*
2: weft-2.4 (-49)*
3: xfinitywifi (-52) 
4: xfinitywifi (-92) 
5: JCD123 (-93)*
6: FQ1601 (-93)*
7: NETGEAR22 (-82)*
8: xfinitywifi (-92) 
9: Biesheuvel (-89)*
10: xfinitywifi (-92) 
11: Mountain_Time (-71)*
12: Spinaltap-2.4 (-65)*
13: xfinitywifi (-65) 
14: Vince2 (-62)*
15: xfinitywifi (-76) 
16: xfinitywifi (-93) 
17: FQ1603 (-49)*
18: xfinitywifi (-48) 
19: GingerPanda (-77)*
20: xfinitywifi (-91) 
21: xfinitywifi (-75) 
22: NETGEAR06 (-92)*
23: cooper (-92)*

scan start
scan done
21 networks found
1: Zoomc36a (-75)*
2: weft-2.4 (-49)*
3: xfinitywifi (-54) 
4: xfinitywifi (-95) 
5: xfinitywifi (-92) 
6: JCD123 (-95)*
7: xfinitywifi (-94) 
8: FQ1601 (-91)*
9: NETGEAR22 (-80)*
10: Biesheuvel (-88)*
11: Spinaltap-2.4 (-64)*
12: Vince2 (-63)*
13: xfinitywifi (-65) 
14: Mountain_Time (-76)*
15: xfinitywifi (-76) 
16: cooper (-93)*
17: FQ1603 (-52)*
18: GingerPanda (-79)*
19: xfinitywifi (-78) 
20: xfinitywifi (-52) 
21: xfinitywifi (-92) 

scan start
scan done
19 networks found
1: weft-2.4 (-48)*
2: xfinitywifi (-49) 
3: Zoomc36a (-73)*
4: FQ1601 (-90)*
5: xfinitywifi (-90) 
6: JCD123 (-93)*
7: xfinitywifi (-91) 
8: cooper (-91)*
9: NETGEAR22 (-80)*
10: Biesheuvel (-89)*
11: Vince2 (-61)*
12: Spinaltap-2.4 (-65)*
13: xfinitywifi (-65) 
14: xfinitywifi (-77) 
15: xfinitywifi (-93) 
16: FQ1603 (-50)*
17: xfinitywifi (-51) 
18: GingerPanda (-75)*
19: xfinitywifi (-79) 

scan start
scan done
18 networks found
1: Zoomc36a (-73)*
2: weft-2.4 (-50)*
3: xfinitywifi (-50) 
4: xfinitywifi (-94) 
5: xfinitywifi (-94) 
6: FQ1601 (-91)*
7: xfinitywifi (-96) 
8: cooper (-92)*
9: NETGEAR22 (-81)*
10: Mountain_Time (-72)*
11: Vince2 (-63)*
12: xfinitywifi (-67) 
13: xfinitywifi (-74) 
14: Spinaltap-2.4 (-67)*
15: FQ1603 (-51)*
16: xfinitywifi (-51) 
17: xfinitywifi (-76) 
18: GingerPanda (-77)*

<br />
<br />
Maybe it’s a linked list issue?

deshipu, 2016-08-29:
By the way, you can put your code in triple backticks

````
```
like this
```
````

to make a whole block of text display as code.

profra, 2016-08-29:
I have the same problem with AP at work... scan doesn't see it...
BUT if I explicitly do cmd "wlan.connect('mySSID'. 'myPSW')" I am connected... try it...

mheld, 2016-08-29:
Thanks. I can definitely connect to my network, just can't see it in a scan.

profra, 2016-08-29:
Glad to hear it... the problem is serious .. I can't write reliable and smart boot.py ... which connects only to trusted APs :-(

dpgeorge, 2016-09-04:
@mheld which version of the Arduino ESP8266 code did you use?  Do you know which ESP SDK you used with it?

mheld, 2016-09-05:
@dpgeorge I used the bin from micropython.org -- esp8266-20160825-v1.8.3-49-ga589fa3.bin, which looks like it comes from this commit ->https://github.com/micropython/micropython/tree/a589fa3e0bdb65f9aa9ad30f720bfc93788689e9

Not really sure which versions of the SDK or esp code it's using

dpgeorge, 2016-09-06:
@mheld I mean, which version of the *Arduino* firmware did you use.  Then I can check how that behaves.

mheld, 2016-09-06:
@dpgeorge 
I'm using:
- arduino v1.6.11
- arduino core for esp8266 v2.3.0 (https://github.com/esp8266/Arduino/releases/tag/2.3.0, though I'm using the one installed via the board manager)
- and the following code:

```
/*
 *  This sketch demonstrates how to scan WiFi networks. 
 *  The API is almost the same as with the WiFi Shield library, 
 *  the most obvious difference being the different file you need to include:
 */
#include "ESP8266WiFi.h"

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(10);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(5000);
}
```

dpgeorge, 2016-09-06:
Thanks @mheld for the info, I was able to reproduce the issue.  I looked at the Arduino implementation and they handle the scan data differently to us.  Apparently the pointer that is passed to the scan callback function is a pointer to the first network, whereas our code was assuming that it was a different structure and skipping the first 2 entries.  It should now be fixed in b4be5a8f3499ca19bdc9ac3ef94625529141b0ff.

@pfalcon you may want to review this.

closed 2016-09-06 by dpgeorge

mheld, 2016-09-06:
Gracias @dpgeorge!

profra, 2016-09-06:
@dpgeorge BIG THANK also from me... I can use my "clever_boot" now :-)

pfalcon, 2016-09-06:
@dpgeorge : Looks good, thanks.