micropython/micropython #2374: Make frozen modules path double slash on OSX

mcauser, 2016-08-29:
I have not been able to compile frozen bytecode for a long time now, reverting to flashing precompiled binaries and WebREPL sending files. I have been trying to compile the latest MicroPython ESP8266 from source on OS X Yosemite.

**Today, I discovered a hack which made it start working.**

Usually I see: `could not find module '_boot.py'` and am unable to `import webrepl` or seemingly anything in `modules`.

Flashing with verify always returns OK and `esp.check_fw()` always True.
`import os` and `os.listdir()` would always throw `OSError: [Errno 19] ENODEV`

I noticed when I run `make clean && make axtls && make` the modules had double slashes in the path.

```
CC ../lib/fatfs/option/ccsbcs.c
CC ../drivers/dht/dht.c
MPY modules//_boot.py
MPY modules//flashbdev.py
MPY modules//webrepl.py
MPY modules//webrepl_setup.py
Creating build/frozen_mpy.c
CC build/frozen_mpy.c
```

**None of the other paths had double slashes.**

So I searched the code for places where the filenames had leading slashes and found:
`micropython/esp8266/build/frozen_mpy.c`

At the end, there are the lines:

```
const char mp_frozen_mpy_names[] = {
"/_boot.py\0"
"/ds18x20.py\0"
"/flashbdev.py\0"
"/onewire.py\0"
"/webrepl.py\0"
"/webrepl_setup.py\0"
"\0"};
```

I tried removing the leading forward slashes from each line and flashed again.

```
esptool.py -p /dev/tty.wchusbserial1420 erase_flash
make PORT=/dev/tty.wchusbserial1420 deploy
```

To my surprise, **it worked!**
I saw **Performing initial setup** in the first few lines - something I haven't seen in a long time.
I am now able to import frozen modules and the filesystem exists.

```
import webrepl
import ds18x20
import os
os.listdir()
['boot.py']
```

So, my question is, is this Mac OS X specific? Do other os's treat the double slash as a single slash?
Do you see double slashes `MPY modules//_boot.py` when you `make -C unix`?

Related forum thread: http://forum.micropython.org/viewtopic.php?f=16&t=2144&start=10

Gist of my make output: https://gist.github.com/mcauser/b837daee2ae54feb248f8fc51a28f666

robert-hh, 2016-08-29:
No: on my Linux machine there are no leading slashes in frozen_mpy.c
But I have not running OS X device for cross check (only broken ones and an outdated VM).

dpgeorge, 2016-08-30:
@mcauser : What is the output of `find modules/ -type f -name '*.py' on your Mac OS X?  That could be the command that's inserting 2 slashes (on Linux that command has only 1 slash).

robert-hh, 2016-08-30:
In my OS X VM, that inserts 2 slashes. Using `'find modules -type f -name '*.py' (w/o the slash), it returns a single slash, both on OS X and Linux.

mcauser, 2016-08-30:
@dpgeorge On OS X Yosemite

:-1: Find with trailing slash:
```
Mikes-MBP:esp8266 mike$ find modules/ -type f -name '*.py'
modules//_boot.py
modules//ds18x20.py
modules//flashbdev.py
modules//onewire.py
modules//webrepl.py
modules//webrepl_setup.py
```

👍 Find without trailing slash:
```
Mikes-MBP:esp8266 mike$ find modules -type f -name '*.py'
modules/_boot.py
modules/ds18x20.py
modules/flashbdev.py
modules/onewire.py
modules/webrepl.py
modules/webrepl_setup.py
```
👍 Find with * wildcard:
```
Mikes-MBP:esp8266 mike$ find modules/* -type f -name '*.py'
modules/_boot.py
modules/ds18x20.py
modules/flashbdev.py
modules/onewire.py
modules/webrepl.py
modules/webrepl_setup.py
```

:+1: On my Debian VM all of these return single slashes:
```
find modules -type f -name '*.py'
find modules/ -type f -name '*.py'
find modules/* -type f -name '*.py'
```

dhylands, 2016-08-30:
You definitely don't want the variations with the star. bash will try to expand that before calling find. The variation without the trailing slash seems right to me.

commit 9c04ef2a (2016-08-31): unix,stmhal,esp8266: When find'ing frozen files don't use extra slash.

This extra forward slash for the starting-point directory is unnecessary
and leads to additional slashes on Max OS X which mean that the frozen
files cannot be imported.

Fixes #2374.
closed 2016-08-31 by dpgeorge