micropython/micropython #2376: ESP8266 1.8.3-80 i2c.scan shows all available ports

Cediddi, 2016-08-30:
Hi. While I was trying to connect a 24LC02B external 2k EEPROM to my wemos 1D mini board (essentially a esp8288 with usb serial converter) I had problem to scan i2c ports.

This is what I tried.

    >>> from machine import Pin, I2C
    >>> i2c = I2C(scl=Pin(5), sda=Pin(4))
    >>> i2c.scan()
    [8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119]

I connected eeprom's scl sda to 5 and 4 respectively and WP and Address pins are grounded (expecting a 80 address from the eeprom)

When I ignore this problem and try to writeto_mem it think it succeeded but readfrom_mem only returns 0x00.

Downgrading also doesn't work, issue is persistent with 1.8.2-104.

mcauser, 2016-08-30:
Did you include pull-up resistors between D1 (SCL) -> 3V3 and D2 (SDA) -> 3V3?

Cediddi, 2016-08-30:
As far as I can understand, the board doesn't have pull-up resistors. I also did not included any resistor.

http://www.wemos.cc/Products/images/d1_mini.pdf

dpgeorge, 2016-08-30:
You need to have pull-up resistors on both SDA and SCL.  If you are getting a list of all addresses from i2c.scan() then that probably means SDA is constantly low.

Cediddi, 2016-08-30:
Thank you Damien :)

closed 2016-08-30 by Cediddi