micropython/micropython #2378: Dictionaries in frozen code behave wrongly when accessed with non-frzoen string keys

gratefulfrog, 2016-09-01:
Dear all,

I am sorry for the hard to understand subject of this bug, but the bug itself is hard to understand.

My FROEZN code reads a csv file from the pyboard sd-card, and stores the data as dictionaries. The first line of the csv file contains string headings which are used as keys. In particular, 2 of the headings are a bit longer than the the others: HorizontalSelector and VerticalSelector . The code also stores the headings as strings in a variable (self.header in the Presets class)

Later, when I want to access the elements of the dictionaries, I use the strings to look up values via the variable, the lookup may fail with a value error.

The failure seems to depend on the source of the strings.

If the strings are in RAM loaded code, i.e. from a .py file on the SD-card, then there is no failure.

But if the strings are from FROZEN code, then there is a failure.

I've attached a set of files which can consistently reproduce the bug.  The usage is a bit complex. It is documented in the file bug.py, but I explain it here.

There are several steps involved:
1. Make frozen bytecode and load it on the pyboard: take all the .py files in the root directory, build frozen byte code and deploy to the pyboard.
2. copy the Data directory to a micro-sd car that will be used for the tests.
3. copy the file from the SD directory (config_.py) to the root of the sd-card. Be sure to copy the right file.
4. put the sd-card into the slot on the pyboard.
At this point, the test can be run:
5. connect the pyboard in pyb.usb_mode('CDC+MSC') mode (put that code in the boot.py).
6. evaluate the following expressions:
```
 >>> import bug
[... messages ...]
>>> sd=bug.getP()  # this creates an instance using only strings from the sd-card as keys to the dictionary
[... messages ...]
>>> frozen=bug.getP(False)  # this creates an instance using strings from the frozen bytecode as keys to the dictionary
```
Now, you can execute the lookups and see the results:

```
>>> bug.test(sd)  # see what happens when all strings come from sd-card..
# there should be no error
>>> bug.test(frozen)  # see what happens when strings from frozen are keys, but other come from sd-card..
[... messages ...]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "bug.py", line 33, in test
KeyError: <value>

``` 
I realize that this is a pretty complex sequence and believe me that I am sorry to not be able to find a simpler example.

I really hope this somehow helps micropython devt.!
Cheers,
Bob

[FrozenStringBug.zip](https://github.com/micropython/micropython/files/449580/FrozenStringBug.zip)

peterhinch, 2016-09-01:
This code, when frozen, produces a dictionary which works only for keys shorter than a certain number of characters. Longer keys incorrectly produce a KeyError (i.e. 'HorizontalSelector' and 'VerticalSelector'). The key 'MasterTone', and all others, work. 10 < threshold <= 16. Attempts to simplify such as freezing a module containing only the dictionary fail to reproduce the fault.

dpgeorge, 2016-09-02:
Thanks for the very comprehensive bug report!

The issue was a combination of 3 things:

1. Strings longer than 10 bytes are not interned but instead make proper string objects.
2. When freezing proper string objects the hash is not computed (it is computed for unfrozen code).
3. The runtime was not correctly handling the case of an uncomputed hash in some places.

I fixed (3) in 5f3bda422a18d49fb282a93968b658c568343b7d, and did (2) in b6bdf18debfd75e9d39c2e8452c7b4ade448bcc5.  Both these fixes on their own fixed the bug reported above.

closed 2016-09-02 by dpgeorge

gratefulfrog, 2016-09-02:
Damien, Thank you for fixing this issue! 

I'm glad I was able to "help" (is finding bugs helping? ;-)

Keep up the GREAT WORK!

Cheers,
Bob

dpgeorge, 2016-09-02:
@gratefulfrog my pleasure!  And yes, finding bugs like this (with reproducible code) is very helpful. Especially this one which was a proper bug, very nasty, and only came up in very special circumstances, but could cause big problems if it wasn't fixed.  Thanks!