micropython/micropython #2379: unix port socket.settimeout not implemented

danni, 2016-09-02:
`socket.settimeout()` isn't implemented in the UNIX port, which makes it extremely difficult to test code using socket timeouts on UNIX.

(This is especially fun because the ESP8266 doesn't have `uselect`, I have no platform independent way to regain control.)

pfalcon, 2016-09-02:
This situation is known to maintainers, you're the first one to bring it from community (to help assess its priority). We discussed this situation with @dpgeorge , and I believe he even had timeout code for unix port. My reply at that time was however that providing different ways to achieve the functionality isn't too good an idea, and adding select() (well, poll()) support to modlwip is the right way.

Feel free to explain your usecase, why you need timeouts in unix port, then impact on code size of their implementation, and this feature should be configurable to be able to maintain small code size.

danni, 2016-09-02:
Reasoning came from wanting a platform independent way to sleep waiting for IO and uselect not being available on all platforms (preferred choice... well after asyncio). Adding settimeout to the UNIX port seemed easier than adding select to the ESP port as a first cut, to see if my code works.

Also I was surprised by not having it in the UNIX port when it's on the chips (the docs could be clearer in what functions are in what ports). Is code size such a big deal on the UNIX port?

danni, 2016-09-02:
More specifically socket.io needs to ping the server occasionally https://github.com/danni/uwebsockets

pfalcon, 2016-09-03:
> Also I was surprised by not having it in the UNIX port when it's on the chips (the docs could be clearer in what functions are in what ports).

"chips" are standalone, thing-in-itself systems, so if something is not available for them, it may be complicated to add it. Unix port OTOH is an open-end system, similar to CPython. Large part of its functionality is implemented in Python in standard library: https://github.com/micropython/micropython-lib

> Is code size such a big deal on the UNIX port?

These and related questions are covered in https://github.com/micropython/micropython/wiki/ContributorGuidelines .

danni, 2016-09-03:
I did look at trying to extend the socket object with timeout support in pure Python, but it needed a lot of knowledge from `socket.h` to run on both Linux and OSX. Also I had assumed that there would be some intention of having the ports be reasonably equivalent in terms of methods available.

Somewhat related, but is there a push to unify the docs between the ports, and have an "Availability: Pyboard, ESP8266, ...". The multiple sets of docs for the ports caused me no end of confusion starting out. Especially writing code on the train without the chip handy, and then finding out it wasn't going to run.

danni, 2016-09-03:
I'm happy for you to reject this request/PR if you definitely don't want to grow the size of the UNIX port, because I agree `select` is better. I think I can get a kludey select going on the ESP pretty easily to retain some platform independence. However in this case, maybe settimeout should just be deprecated on all platforms with "Use select"?

pfalcon, 2016-09-03:
> I did look at trying to extend the socket object with timeout support in pure Python, but it needed a lot of knowledge from socket.h to run on both Linux and OSX.

It can be implemented using uselect.poll().

> Also I had assumed that there would be some intention of having the ports be reasonably equivalent in terms of methods available.

Yes, using the more powerful apparatus, not less powerful. uselect.poll can timeout on multiple sockets in parallel, settimeout() only on one.

> Somewhat related, but is there a push to unify the docs between the ports, and have an "Availability: Pyboard, ESP8266, ...".

Yes, even more, not to have "availability". That means actually make everything work across ports. It's very easy to hack some adhoc code, it's hard to design consistent full-featured interface for different hardware, and then actually implement it across various hardware. So, it's ongoing work.

dpgeorge, 2016-09-03:
> The multiple sets of docs for the ports caused me no end of confusion starting out.

This is a recurring problem that users are having.  Maybe we need to have just one set of docs (ie no separate builds for separate ports), and stuff that's completely port specific (quickref and tutorials, but even those can and should have generic versions) goes in individual sections that are accessible from the main index page.  I think this would be easier to maintain, and easier for users to use.

pfalcon, 2016-09-03:
> I'm happy for you to reject this request/PR if you definitely don't want to grow the size of the UNIX port, because I agree select is better. I think I can get a kludey select going on the ESP [...]

uselect.poll should be standard interface, implemented on all ports. Beyond that, it's up to maintainers of particular ports what to have more.

> I think I can get a kludey

While people will go this way - doing kludgy things on individual ports, there will be the situation like you describe. It take a bit more effort and forward-looking thinking to get ahead of that. I'm saying that as someone who contributes to MicroPython almost daily last 2.5 years.

pfalcon, 2016-09-03:
> Maybe we need to have

What we need to do about the docs as a next step is to have proper section ordering in ToC. Look at ToC-based documentation format, like PDF, to see how funny sections are ordered.

deshipu, 2016-09-03:
> Maybe we need to have just one set of docs (ie no separate builds for separate ports), and stuff that's completely port specific (quickref and tutorials, but even those can and should have generic versions) goes in individual sections that are accessible from the main index page.

This would also encourage more compatibility between ports, by making all inconsistencies easy to see, awkward and inconvenient.

g-sam, 2017-08-08:
Plus one for this. Ports should be consistent unless there is a good reason for them not be.

I ran into this because I wanted to have my tests for esp8266 code run on unix.

pfalcon, 2017-11-03:
The issue in the title is now documented, including the suggested alternative at http://docs.micropython.org/en/latest/unix/library/usocket.html?#usocket.socket.settimeout . Given the code size constraints we'd like to maintain, presence of an alternative (more verbose, but also more powerful), and a queue of features we'd like to add, which don't have viable alternatives, I think it's fair to say that timeout support in Unix port won't be implemented on C level. Interested parties are welcome to implement it in https://github.com/micropython/micropython-lib/blob/master/socket/socket.py . 

I'm thus closing this.

closed 2017-11-03 by pfalcon

dpgeorge, 2018-10-17:
socket.settimeout() on the unix port was implemented in 80a25810f9b246094b31ce7d465a0f25842c9be6 and 0c18633ea9bed6a0c1031357c4eacbb016deb41a