micropython/micropython #2380: `uselect` missing from the ESP8266 port

danni, 2016-09-02:
There is no platform independent way to work out when your sockets are ready to be read from/or timeout to run your other events from time to time.

See also #2379.

peterhinch, 2016-09-02:
The [umqtt library](https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py) achieves this (check_msg and wait_msg methods) by issuing
```python
socket.setblocking(False)
```
enabling the socket to be polled.

pfalcon, 2016-09-02:
Implementing poll() support for modlwip is definitely a needed feature, and on long to do, but that's literally first request for it. So, it may take some time to be implemented still, given other tasks in queue. Feel free to try to implement it and contribute, if you need it soon.

Here's my idea how I wanted to implement it: don't implement it in naive polling way, e.g. how it's implemented in stmhal. Instead, add "poll_obj" pointer to each socket object (that's memory size drawback, yes), and that means that each socket can be member of only one poll object (another drawback, but should be rather acceptable). Then, lwip callbacks can just signal updated socket state to poll object, and it can just return ready state to caller. That's rough idea, and there may be many implementation details, caveats, and possibly even blockers. Note that all uselect.poll() features should be supported (like one-shot mode).

Otherwise, I'm going to tackle this when approaching uasyncio porting to esp8266.

deshipu, 2016-09-02:
First request, eh? http://forum.micropython.org/viewtopic.php?f=16&t=2009&p=11268

pfalcon, 2016-09-02:
No ticket == forgotten. Ok, 2nd request.

deshipu, 2016-09-02:
Personally I saw that answer and just assumed you are working on it, so decided to not bother anyone. It will be ready when it's ready. Good to see that now it's actually true.

danni, 2016-09-02:
@pfalcon right I considered doing something like that to get uasyncio working via some kind of ESP specific event loop, then I decided that getting select working made more sense. I might give it a go now I have my own firmware compiling, although I'm not clear what debugging options I have beyond printf, which is somewhat of a turn off.

danni, 2016-09-02:
Also apologies in advance I feel like I'm only just beginning to get my head around how micropython works, r.e. the ports, different functionality in each port, where code for various things is, etc.

dpgeorge, 2016-12-02:
uselect (and select, via weak link) module are now implemented for esp8266, with basic but functional support for select/poll of sockets.

closed 2016-12-02 by dpgeorge