micropython/micropython #2382: Prepare for v1.8.4

dpgeorge, 2016-09-02:
It's been a month since the last release so time for the next one.

pfalcon, 2016-09-02:
I'd like to land streaming support for uzlib module, it needs some work yet. Nothing else big from me, though would be nice to do more about WDT class.

https://github.com/micropython/micropython/issues/2315 is important, but there's no even a plan how to handle it, so for sure doesn't fit.

dpgeorge, 2016-09-03:
> I'd like to land streaming support for uzlib module, it needs some work yet

+1  But we need to discuss the API (whether it's compatible with CPython std lib or not).

> would be nice to do more about WDT class.

Yes.  I have some watch-dog code for stmhal which can also go in now.

BTW, is it too crazy to suggest that the "WDT" class be renamed to "WatchDog"?  Rationale: 1) we already have acronym overload (eg I2C, SPI, ADC, etc) and a more human friendly name would be welcome; 2) a new-comer would understand WatchDog better than WDT without reading the docs; 3) some people are used to WDG and others WDT, so WatchDog is a better common ground.

pfalcon, 2016-09-03:
> +1 But we need to discuss the API (whether it's compatible with CPython std lib or not).

It already went in (major part, there're still things to improve). It's compatible with standard Python's stream protocol (pull-style access). It's not compatible with accidental push-style class in CPython's zlib. Shouldn't be a problem given it's *uzlib* module, not *zlib*. Push-style class supposedly can be implemented in terms of pull-style class. Push-style class itself isn't useful for intended usages of uzlib (like, supporting running upip of no-real-heap systems), as it requires a lot of manual (Python-level) handholding to do useful operations (== memory fragmentation).

pfalcon, 2016-09-03:
> BTW, is it too crazy to suggest that the "WDT" class be renamed to "WatchDog"?

I'm all for changing APIs when there're good reasons for that, and to compensate that attitude, -1 on changing them when there're no good enough reasons ;-). 

> we already have acronym overload (eg I2C, SPI, ADC, etc)

These acronyms, as well as WDT, are standard industry terms for the underlying functionality. Calling them in common words would be rather confusing for people with basic knowledge (and not really helpful for people who don't know basic terminology - at least now they have a chance and incentive to learn).

> some people are used to WDG and others WDT, so WatchDog is a better common ground.

We cam call e.g. Wikipedia for jury: https://en.wikipedia.org/wiki/WDT vs https://en.wikipedia.org/w/index.php?title=WDG&redirect=no

dpgeorge, 2016-09-03:
> It already went in (major part, there're still things to improve).

I see it.  I agree it's clean.  Let's just hope zlib.decompressobj is something that's rarely used (google only returns 5k search results) and not missed by anyone if it's not there.

dpgeorge, 2016-09-03:
Ok, we keep it WDT.  See 76c366df56ad64eae3b021304954954e663bf1df for stmhal implementation.

pfalcon, 2016-09-04:
@dpgeorge : I believe I now have everything what I wanted to push, in the master. I may be more busy/have less Inet connectivity next week, so if you decide to go for the release, please do. I otherwise continue on RE side of things (I started in 10's of August, and would like it to be a month).

dpgeorge, 2016-09-04:
@pfalcon ok I will go for it within the next few days.

dpgeorge, 2016-09-08:
@pfalcon the btree1.py test is failing consistently on esp8266 with a MemoryError.  It seems to be in the db.close() line.  This is not critical to fix for a release but would be nice if it has an easy fix.

dpgeorge, 2016-09-09:
v1.8.4 was tagged at 3611dcc260cef08eaa497cea4e3ca17977848b6c.

closed 2016-09-09 by dpgeorge