micropython/micropython #2385: How to get serial device and volume mounted on Mac OS X El Capitan

BrendanSimon, 2016-09-02:
I've installed MP on an Olimex E407 board and I can use REPL via a Windows7 VirtualBox running on my MacBook (OS X El Capitan).

I'd prefer to use native OS X (instead of Win7) but the serial port and volume do not appear.  i.e. nothing in /dev and nothing in /Volumes.

Any ideas how to get Mac OS X to create the /dev entry and /Volume mount ??

So on linux I presume you need the udev stuff, but what about Mac OS X ??

Thanks, Brendan.

dhylands, 2016-09-03:
On my Mac Mini (running Yosemite), my pyboard shows up as /dev/tty.usbmodem622 (as well as /dev/cu.usbmodem622)

If I open Finder and click on PYBFLASH, then I see /Volumes/PYBFLASH

I didn't have to anything special that I can remember to make either of those work.

BrendanSimon, 2016-09-12:
RESOLVED

I removed the existing FTDI drivers (possibly from an install on a previous version of OS X before I upgraded) and reinstalled the latest for OS X 10.9+.

http://www.ftdichip.com/Support/Documents/AppNotes/AN_134_FTDI_Drivers_Installation_Guide_for_MAC_OSX.pdf

I now get /dev/tty.usbmodemFA132 and /dev/cu.usbmodemFA132 entries, and /Volumes/PYBFLASH.

Cheers, Brendan :)

dpgeorge, 2016-09-13:
Great!

closed 2016-09-13 by dpgeorge