micropython/micropython #2386: PyBoard CAN doesn't work?

kamikaze, 2016-09-03:
When trying following code:
`from pyb import CAN

can1 = CAN(1, CAN.NORMAL)
can2 = CAN(2, CAN.NORMAL)

can1.setfilter(0, CAN.LIST16, 0, (123, 124, 125, 126))
can2.setfilter(0, CAN.LIST16, 0, (123, 124, 125, 126))

can1.send('magic', 123)
can2.recv(0)

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 110] ETIMEDOUT`

and after many send() calls I'm getting

`OSError 16` which is device or resource busy.

I've tried to connect Y3-Y6 and Y4-Y5 (CAN1 - CAN2) directly. I've tried to connect with the following scheme of two transceivers: https://easyeda.com/kamikaze/CAN_bus_shield_for_PyBoard_v1_1-CusOLJ53N

for me it seems that something is wrong with the code, board or micropython... please help me to figure out )

HenrikSolver, 2016-09-03:
Most likely there is nothing wrong with the pyboard or its firmware, so please close this issue and start a thread in the forum.

kamikaze, 2016-09-03:
An issue is here for a reason...

http://forum.micropython.org/viewtopic.php?p=13368#p13368

kamikaze, 2016-09-03:
my fault, did wrong voltage connection. Sorry.

closed 2016-09-03 by kamikaze