micropython/micropython #2388: Unexpected interractions between pyb.Accel and ExtInt on pins X9, and X10

gratefulfrog, 2016-09-03:
I have discovered an unexpected interaction between an instance of  pyb.Accel and ExtInt on pins X9 and X10.
There are 2 cases with different, yet what I believe to be incorrect behavior:

Case 1: Accel is instantiated before the ExtInts: Accel is disabled by the creation of the ExtInts

```
PYB: sync filesystems
PYB: soft reboot
MicroPython v1.8.3-93-gb6bdf18 on 2016-09-02; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>>  from pyb import ExtInt,Pin,Accel
>>> a=Accel()
>>> a.x()  # readings are correct
-18
>>> a.y()
-2
>>> a.z()
-6
>>> pins=('X9','X10')
>>> def callback(line):
...     print(line)
... 
>>> i=[ExtInt(p,ExtInt.IRQ_FALLING,Pin.PULL_UP,callback) for p in pins]
>>> a.x() # readings are now constant and wrong!
4
>>> a.y()
4
>>> a.z()
4
>>> 
```
Case 2: The ExtInts are instatiated before the Accel: interrupts are fired, but the readings are correct!

```
PYB: sync filesystems
PYB: soft reboot
MicroPython v1.8.3-93-gb6bdf18 on 2016-09-02; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> 
>>> from pyb import ExtInt,Pin,Accel
>>> pins=('X9','X10')
>>> def callback(line):
...     print(line)
... 
>>> i=[ExtInt(p,ExtInt.IRQ_FALLING,Pin.PULL_UP,callback) for p in pins]
>>> a=Accel() # instantiation causes interrupts to be fired!
7
6
6
7
6
7
7
6
6
7
6
7
6
7
7
6
7
7
>>> a.x()  # readings cause interrupts to be fired, but the return value is correct
7
6
6
7
6
7
6
7
6
6
7
6
6
7
6
-20
>>> a.y()
7
6
6
7
6
7
6
7
7
6
6
7
6
6
7
6
-2
>>> a.z()
7
6
6
7
6
7
6
7
7
6
6
7
6
6
7
6
-6
>>> 
```

I've attached a more comprehensive testing suite which shows these results more clearly.

I hope this is not simply my own misunderstanding!

Cheers,
Bob
[accelTester.py.zip](https://github.com/micropython/micropython/files/453436/accelTester.py.zip)

dhylands, 2016-09-03:
X9 and X10 corresponds to PB6 and PB7 which are the SDA and SCL pins used for the I2C1 bus which is used to communicate with the Accelerometer.

So yes, using the Accelerometer means that pins X9, X10 are tied up for the I2C bus.

You can see this by checking the pins before and after running pyb.Accel:
```
MicroPython v1.8.3-36-g080cbf7 on 2016-08-18; PYBv1.0 with STM32F405RG
Type "help()" for more information.
>>> pyb.Pin('X9')
Pin(Pin.cpu.B6, mode=Pin.IN)
>>> pyb.Pin('X10')
Pin(Pin.cpu.B7, mode=Pin.IN)
>>> pyb.Accel()
<Accel>
>>> pyb.Pin('X9')
Pin(Pin.cpu.B6, mode=Pin.AF_OD, af=Pin.AF4_I2C1)
>>> pyb.Pin('X10')
Pin(Pin.cpu.B7, mode=Pin.AF_OD, af=Pin.AF4_I2C1)
```
That should probably be mentioned on the pyb.Accel documentation page.

gratefulfrog, 2016-09-03:
I see.

Perhaps this could simply be documented somewhere since it is not really a defect.  Or maybe experts would understand without more documentation?

If you want to close the issue, I'm fine with that.

Sorry for the inconvenience.
Cheers,
Bob

dpgeorge, 2018-12-04:
A note about this was added in 46e59c52afe60d1c345df62158b2e5b6c64e9146

closed 2018-12-04 by dpgeorge