micropython/micropython #2396: Problem talking I2C to Olimex-OLED on OLIMEX-E407 board

BrendanSimon, 2016-09-06:
I have built MP for OLIMEX-E407 board.  I have an OLIMEX-OLED board connected to the UEXT connector, which has an I2C interface.

I ported the Olimex OLED python module for linux to MP, but I can't get the OLED to display anything.

First I fixed the I2C mappings on the OLIMEX-E407 port as follows.
-#define MICROPY_HW_I2C1_SCL (pin_B6)
-#define MICROPY_HW_I2C1_SDA (pin_B7)
+#define MICROPY_HW_I2C1_SCL (pin_B8)
+#define MICROPY_HW_I2C1_SDA (pin_B9)

The scan() function works and I can see device 60 on the bus, which is correct.

I am sending commands 0x80, 0xAE to the device.  The data sheet says the I2C bus should have
[ Device Adress, Control Byte, 0x80, Control Byte, 0xAE ]
where Control Byte = 0xC0,

however looking at a Logic Analyser capture, I can see the Control Byte being written is 0xC2.
i.e. [ DevAddr, 0xC2, 0x80, 0xC2, 0xAE ] instead of [ DevAddr, 0xC0, 0x80, 0xC0, 0xAE ]

Is this a problem?
Is this potentially a reason why the OLED is not working?

Why/how is the Control Byte generated?

I'm not sure if I'm using DMA mode or not.  Is there a way to configure this at run time ??

Thanks, Brendan.

dpgeorge, 2016-09-06:
> First I fixed the I2C mappings on the OLIMEX-E407 port as follows.
> -#define MICROPY_HW_I2C1_SCL (pin_B6)
> -#define MICROPY_HW_I2C1_SDA (pin_B7)
> +#define MICROPY_HW_I2C1_SCL (pin_B8)
> +#define MICROPY_HW_I2C1_SDA (pin_B9)

If those lines are wrong then they should be changed.  @stefanhoelzl can you confirm this mapping is wrong?

> I am sending commands 0x80, 0xAE to the device.

How are you creating the I2C object and how are you sending the commands, please give exact code.

[BTW, this issue is really more suited to the forum.micropython.org]

BrendanSimon, 2016-09-06:
OK.  Thanks.  I didn't realise there was a forum.  I will post on there.

The schematic for the E407 board has the following lines going to the
UEXT connector.

PB8/I2C1_SCL    => pin 5 or UEXT
PB9/I2C1_SDA    => pin 6 or UEXT

Regards,
Brendan.

BrendanSimon, 2016-09-07:
I've partially sorted things out.  The C2/C3 "control bytes" being written weren't actually "control bytes" at all.  The actual "control bytes" are the 0x80 values (and 0x40 values in subsequent data writes) in my data stream.

The problem was me converting a list of integers to a string of bytes using ''.join(data), since i2c.send() takes a buffer interface (at a guess) and not a list.  Since this is Python3 (first time I've used Py3) the result was infact a unicode string, not an ascii string.

I used bytes(data) instead and now I get the OLED to display something -- random snow instead of text, but at least it's doing something :)

dpgeorge, 2016-09-19:
I2C1 mappings fixed in 4874bde10428302eac848b3130e5076e685c7774.

closed 2016-09-19 by dpgeorge