micropython/micropython #2401: Move ports to subdirectory

tannewt, 2016-09-06:
What do you all think about moving ports to a separate subdirectory? I'll do the work if we all agree it'd be worthwhile.

It'd take the top level from

```
ACKNOWLEDGEMENTS
CODECONVENTIONS.md
CONTRIBUTING.md
LICENSE
README.md
bare-arm/
cc3200/
docs/
drivers/
esp8266/
examples/
extmod/
lib/
logo/
minimal/
mpy-cross/
pic16bit/
py/
qemu-arm/
stmhal/
teensy/
tests/
tools/
unix/
windows/
```

To:
```
ACKNOWLEDGEMENTS
CODECONVENTIONS.md
CONTRIBUTING.md
LICENSE
README.md
docs/
drivers/
examples/
extmod/
hardware_ports/
lib/
logo/
mpy-cross/
py/
tests/
tools/
```

Hardware ports would be:
```
bare-arm/
cc3200/
esp8266/
minimal/
pic16bit/
qemu-arm/
stmhal/
teensy/
unix/
windows/
```

stinos, 2016-09-07:
The name _hardware_ports_ isn't really correct though, even a bit confusing, since minimal/unix/windows are not exactly hardware-bound

pfalcon, 2016-09-07:
@tannewt: No revolutions please. If you would like to contribute to MicroPython, but never did before (as https://github.com/micropython/micropython/graphs/contributors shows), we have a list of useful things to do just for such folks: https://github.com/micropython/micropython/issues/1331 . I'd especially like to draw attention to the need to restore test coverage to be back to 90%+, that's where more hands would really help.

roger-, 2016-09-07:
Simply "ports" would be a better name IMHO, and I agree it would be nice to neaten up the directory structure.

tannewt, 2016-09-07:
@pfalcon I'd be happy to help with the testing coverage over the long term.

I'd love your opinion on the ports subdirectory though because I'm adding samd21 support to micropython. My current code adds it as another top level but it'd be cleaner IMO if they were all in a subdirectory. Code is here: https://github.com/adafruit/micropython

pfalcon, 2016-09-07:
@tannewt : You can read MicroPython's author opinion on that here: https://github.com/micropython/micropython/issues/1907 . I personally think that it's far from the top-priority issue in uPy development, and an obvious, existing way "just works". That way being: fork repo, maintain your port, work together with uPy community on resolving general uPy issues, when it's clear to everyone that yourself and your port are there to stay (i.e. being maintained daily, you'll be around to answer users' questions), it can be merged to mainline. It seems that you're on that way already, and that's great.

Problem of "too many ports" can be resolved when there will be indeed too many _well maintained_ ports, and there're a lot of work for everyone to do to get there yet. In the meantime, I'm strongly -N or renaming something, because it complicates usage of "git log" - a tool, which should be daily friend of everyone who contributes to uPy (or any other project for that matter).

tannewt, 2016-09-07:
@pfalcon thanks for the link. That background is really useful. I came into this assuming centralizing is the right way to do it. Having copies in different repos risks making sharing of work (code and documentation) harder. The points about noisy git log and repo size make total sense though.

Has anyone already made a micropython port which separates micropython well that I can use as an example? I was looking at mynewt and they do something similar. Their SAMD21 support is separated into this repo for example: https://github.com/runtimeinc/mynewt_arduino_zero

dpgeorge, 2016-09-08:
@tannewt it would be nice to eventually clean up the top-level directory structure.  But we can already address your (valid) concerns re sharing work without doing a restructure: a lot of the shared code is slowly being moved to the lib/ directory (see lib/utils/pyexec for example), and the common modules are in extmod (eg machine stuff, fatfs stuff).

tannewt, 2016-09-08:
@dpgeorge is there a reason not to do the restructure I suggest? I'm willing to do the work if we can decide how to restructure things. (Splitting into separate repos was another suggestion I saw.)

I didn't mean to get into the resharing discussion here though I'd love to have it elsewhere.

dpgeorge, 2016-09-08:
> is there a reason not to do the restructure I suggest?

Main 2 reasons not to do it are:
1) Messes with the git history.
2) There are lots of forks of this repo and lots of people have branches/rebase on to it with their personal changes.  Everyone will need to update their code to work with our changes.

Of course, the sooner we do such a restructure the better because points 1&2 above (and everything else) just get more entrenched over time.

@tannewt what are your main reasons for wanting to restructure?  The work to do it is insignificant compared with the "fallout".

deshipu, 2016-09-08:
Perhaps a good compromise would be to use the new structure for the new ports (with clear instructions on how to do it), and once that has proved working well, maybe gradually start to move the other ports?

tannewt, 2016-09-08:
I'm not sure why 1 is an issue. Doesn't `git mv` correctly preserve the history? (I'm a relative git noobie. I did perforce for the last sixish years.)

2 is tricky. One option is to submit changes to other forks as well or simply explaining what the change was and how to do it yourself. I come from a monolithic codebase setting where you just warn everyone and make sure big changes have landed before the switch.

The reason I thought of doing it is simply to separate core micropython from port specific micropython. Doing that makes it clearer which directories are for ports and which are micropython. Once thats clearer you can start imposing stronger assumptions of the port directory layout, implementation APIs and file naming.

If we could go further and structure it in a way where core micropython is in a separate repo from port specific stuff that could be even better IMO. How do you decide what parts of micropython get pulled into the microbit version?

@deshipu I'm not sure that provides the clarity benefit though.

dpgeorge, 2016-09-08:
> How do you decide what parts of micropython get pulled into the microbit version?

microbit is a special case because of requirements from the BBC and the yotta build system, and I was just copying needed files into the microbit repo.  But BBC requirements and yotta are now gone (effectively) and it would be nice to bring the microbit port (actually a nRF51 port, with microbit being a board within that port) into the main repo.  That's a very low priority thing though.

tannewt, 2016-09-08:
That would be awesome to reintegrate it. Someone just pushed nrf52 support
too. On my phone now but I can send a link later.

hoihu, 2016-09-09:
> Of course, the sooner we do such a restructure the better because points 1&2 above (and everything else) just get more entrenched over time.

I think so too - restructuring may cause some  "fallout" but the sooner the better IMO. 

The number of ports (and users wanting to port..) seem to be consistently increasing, so thats great! If the structure of the repository is hindering devs to do the ports that would be a pitty.

deshipu, 2016-09-09:
@tannewt what I proposed at least prevents this entrenching and buys us some time

tannewt, 2016-09-09:
@deshipu Thats true! If we introduce it now, then new ports at least can go there.

What external ports are you referring to?

What is your preference for the new layout? Seemed like a few people earlier preferred the directory name `ports`.

deshipu, 2016-09-09:
I think my favorite is the idea of keeping each port in a separate repository, and having the micropython core as a subrepository there. It also lets the authors review and merge their pull request without having commit rights in the main repository, which could greatly spread the reviewing work.

deshipu, 2016-09-09:
I went ahead and split off the esp8266 port just to see what it would take: https://github.com/deshipu/micropython-port-esp8266

(it would probably still have its own readme and docs and maybe even tests)

deshipu, 2016-09-09:
It could also have the history of all the deleted files pruned, then it would be really fast to check out.

toolmacher, 2016-09-12:
I like both idea's, restructure and separate repo's. Better now than later when it is still complicated.
something like this:
/
arch/
--cc3200
--pyboard
--teensy/
--esp8266/
--microbit/
--minimal/
common/
--extmod/
--logo/
--tests/
docs/
drivers/
hal/
--st/ (stmhal)
py/
lib/
os/
--unix
--windows
tools/

and all boards in folder "arch" as separate repo.

deshipu, 2016-09-12:
I have to admit I didn't think about doing it that way -- putting the ports as submodules of the main repository. My example puts the main repository as a submodule of each port, and after some consideration, I think this is actually better, because:

* You don't need any permission or pull request or anything to start working on a new port. You just clone an "empty port" repository that would be provided.
* Since the version is fixed in the submodule on your side, you don't have to be afraid of your port suddenly stopping working when someone merges an incompatible change in the main repository. You will have to explicitly update to the latest master once in a while, of course, but then you can do it at your leisure, and can test for breaking changes as you do it.
* You don't have to ask the maintainers of the main repository to update the version of your port's submodule each time you merge a commit to your port.
* People can check out just the port they are interested in, and pull all the submodules it uses. No need to download the code of all the ports.
* The only change required in the main repository when a port is getting split off is deleting the port's directory from it. No breaking changes, renames, moving things around, conflicts, etc. For a new port you don't need any changes.
* It's easy to fork existing ports, or switch between official and unofficial versions of them.

tannewt, 2016-09-12:
@toolmacher That is an interesting idea. However, I agree with @deshipu that it would be too difficult to keep the port in sync with the apis of the core.

In fact, I think that moving ports into the main repo is the way to go. Why? Having all ports in one place will make the existence of each port very well known and make it easier to unify shared APIs.

Imagine a change that standardizes the machine.I2C api between all of the ports by sharing a lower level C API:

1. with ports as submodules the API of the core cannot change until all submodule ports support the new API.

2. with ports as repos and upy as a submodule its up to each port to adopt a later version of the core. Each will be slightly different and always at slightly different versions.

3. with ports as subfolders in the main repo one commit can change the APIs of all ports at once. One version of micropython exists.

So, I prefer 3 because its the easiest way to unify the machine APIs.

deshipu, 2016-09-12:
> 2. with ports as repos and upy as a submodule its up to each port to adopt a later version of the core. Each will be slightly different and always at slightly different versions.

I see no real problem with that. There will always be ports of lower priority, or even completely abandoned by their authors. I see no reason why they should block development of the core.

> 3. with ports as subfolders in the main repo one commit can change the APIs of all ports at once. One version of micropython exists.

I think you are optimizing for a very rare case here. In fact, so rare, that it didn't even happen so far in the history of this repo. The idea of an atomic commit that updates in one go the API, all the places where it is used, all the documentation that refers to it and all the tests is an alien idea for some of the developers here. In fact even at this moment there are some broken drivers and documentation that weren't updated after an API change. I don't think it makes sense to optimize for something that never happens. Even if a proper review process for everyone was introduced and this was enforced, this kind of change should be really rare in a normal project.

On the other hand, with ports each in their own repository, you can have different sets of people with merge rights, faster code reviews and thus potentially quicker development. A smaller team of people can also easier agree on what to do, and what the standards for the particular port should be. It's also easier to test ideas in one port, and then implement them everywhere if they prove to work well. You also don't need to check out a lot of unneeded code to work on a particular port -- you only download what you actually need, that is, the port's code and the core MicroPython.

The whole move only makes sense if it will allow the MicroPython project to scale better. Putting everything in one place doesn't work very well for this.

dhylands, 2016-09-13:
> I think you are optimizing for a very rare case here. In fact, so rare, that it didn't even happen so far in the history of this repo. The idea of an atomic commit that updates in one go the API, [...]

I don't think its as rare as you think. Several times when working on teensy or stmhal, I've made changes which broke the other and the only reason I've noticed is because we build them both with travis.

@pfalcon has also made changes which broke stuff other than what he was working on (I'm not picking on Paul - anybody that works on larger projects does this from time to time), and the reason I noticed was because I have my own travis builds which build a bunch of additional targets that the regular travis build doesn't.

dpgeorge, 2016-09-13:
I like the idea of having everything in one repo so that it's easy to change things and keep (relative) consistency.  I like the idea of separate repos so the core is lighter, it can be quicker to develop and allows fine grained access control.

Right now I'm thinking that the project is still too immature to split it up.  We need to stick together (as a team and as a code base) to hammer out deep issues that have long-term consequences like the machine API and consistent/portable drivers.  IMO those are the 2 main things that we should work on in the coming months to solidify uPy's foundations (the core is already very stable, will always be defined by upstream CPython, and will slowly evolve).  And to work out the details of these APIs we need a single repo.  Then, once the foundations are set down and the docs are written, the ports will be built around them and things will be pretty consistent.

So, although it would be nice to restructure the top-level dir (and git blame still works on moved files, still has past history, but not git log) I don't think it adds any real benefit right now.  Yes it's a bit confusing to a new comer, but once you get used to it it's really fine.  Once we have a more mature project we can reconsider what to do (either move or split to other repos).

@tannewt what concerns do you have leaving the structure as-is?  Regarding the samd21 port, one thing that would definitely be moved would be the CMSIS library, it would go from stmhal/cmsis to lib/cmsis so it could be reused by other ports.  Actually, in the process of moving we'd upgrade it to the latest version.

tannewt, 2016-09-13:
@dpgeorge If everyone wants to leave it as is thats fine with me. I think its more important to have one repo.

My concern is just that as ports get added the purpose of each top level folder gets less and less clear. Easing the ramp up time to newcomers is a good thing because the more hands working on micropython the faster it can get better.

Having a subdirectory for ports also lessens the overhead of additional ports because it doesn't clutter the top level. That makes it easier to centralize the ports.

dpgeorge, 2016-09-13:
@tannewt I agree, and I'd like to clean up the top-level dir eventually.  Right now though we are not adding any new ports, and soon if we add samd21 that'll only be 1 extra.  Beyond that I'd say we should revisit this issue.

tannewt, 2016-09-13:
@dpgeorge if we want to do it eventually why not now?

dpgeorge, 2016-09-13:
Because eventually we might want to do it a different way (eg separate repos).  Unless we can come to an agreement now on the structure that is relatively future proof.

kamikaze, 2016-10-12:
it's hard to figure out where to find PyBoard "port". And yes... it's better to make a proper project structure now rather than after there will be even more ports/forks/etc. Technical debt is growing every day

dpgeorge, 2017-09-06:
All ports are now moved to the ports/ directory.

closed 2017-09-06 by dpgeorge