micropython/micropython #2406: WiPy: interrupts not returning to main loop?

dmartauz, 2016-09-09:


dpgeorge, 2016-09-09:
The 2 main differences for WiPy between v1.8.1 and v1.8.3 are an upgraded FreeRTOS library, and addition of multithreading.  Both of those things could impact the behaviour of IRQs.

Can you please post here a minimal example which fails on 1.8.3 and passes on 1.8.1?

dmartauz, 2016-09-09:
Thanks for fast response. This is minimal main.py which I just tested with 1.8.1 and 1.8.3:
```
SW1 = machine.Pin('GP17', mode=machine.Pin.IN, pull=machine.Pin.PULL_UP)

def SW1callback(line):
	print("uSW USR1 pressed")
	return

SW1irq = SW1.irq(trigger=machine.Pin.IRQ_RISING, handler=SW1callback)

while True:
	print(time.time())
	time.sleep_ms(1000)
```

1.8.1 - output:
```
473385892
473385893
473385894
uSW USR1 pressed
uSW USR1 pressed
uSW USR1 pressed
473385895
473385896
473385897

```
1.8.3 - output:
473385624
473385625
473385626
uSW USR1 pressed
uSW USR1 pressed
uSW USR1 pressed

And it hangs here until I press the switch again.

Btw is multithreading documented somewhere?

dpgeorge, 2016-09-09:
Thanks, that definitely looks like a bug.

> Btw is multithreading documented somewhere?

The `_thread` module is implemented almost fully on WiPy, as documented by CPython: https://docs.python.org/3/library/_thread.html

dpgeorge, 2016-09-09:
@dmartauz I don't have a WiPy to test against right now but I'm pretty sure that disabling threading will fix the problem.  You can disable it by changing lines 105-106 of cc3200/mpconfigport.h to:
```c
#define MICROPY_PY_THREAD                           (0)
#define MICROPY_PY_THREAD_GIL                       (0)
```
If this works for you then I'll temporarily disable threading on cc3200 until a proper fix is worked out.

robert-hh, 2016-09-10:
I gave it a try. Using plain interrupt with version 1.8.1 just caused the REPL to disappear. Disabling threading made REPL stable, and the ISR was called, just the print statement in the ISR never showed anything. But a counter, which I used instead, was increased. Actual test code:
```
import machine
import time

count = 0

SW1 = machine.Pin('GP17', mode=machine.Pin.IN, pull=machine.Pin.PULL_UP)

def SW1callback(line):
    global count
    print("uSW USR1 pressed")
    count += 1
    return

SW1irq = SW1.irq(trigger=machine.Pin.IRQ_RISING, handler=SW1callback)

while True:
    print(time.time(), count)
    time.sleep_ms(1000)
```
Newly created MP version: MicroPython v1.8.4-30-g99b05f4-dirty on 2016-09-10; WiPy with CC3200

B.T.W.: A single switch press caused count increases between 3 and 400; so much about bouncing.

dmartauz, 2016-09-12:
Damien, Robert, I am currently on holiday and I don't have WiPy with me. I will do the build and testing with and without threading disabled on Monday 19.9.

jgmdavies, 2016-09-13:
All,
I still can't get simple timer irqs to work, even with 1.8.4 - please see http://forum.micropython.org/viewtopic.php?f=11&t=2127&p=13807#p13807
I'm wondering how we know which Timers 'should' work in this situation, i.e. perhaps Timer 1 is in use for the WiFi or the REPL or something else?
Thanks,
Jim

robert-hh, 2016-09-17:
@dpgeorge: changing mpconfigport.h according to your suggestion seems to cure the problem for the moment.

blueslow, 2016-10-18:
Interupts still not working MicroPython v1.8.5 on Wipy, it does return to main but hangs a short while after:

PYB: soft reboot
Platform WiPy
Phyton Version: 3.4.0,(3, 4, 0),
Fw version: 1.2.0
MicroPython v1.8.5-6-g5c93d0b on 2016-10-18; WiPy with CC3200
Type "help()" for more information.
>>> dir()
['micropython', 'os', 'machine', 'UART', 'main', 'counter', 'WLAN', 'sys', 'pinInterrupt', '__name__', 'uart']
>>> main()
Initital value 1
Counter 0
Counter 0
Counter 0
Counter 0
Counter 0
Counter 0
Counter 1
Platform WiPy
Phyton Version: 3.4.0,(3, 4, 0),
Fw version: 1.2.0
MicroPython v1.8.5-6-g5c93d0b on 2016-10-18; WiPy with CC3200
Type "help()" for more information.
>>> 

After main loop printed Counter 0 for the sixth time the GP23 signal is connected to gnd via a button. The interupt routine pinInterrupt is called at least once (No debouncing on button) and counter is increased. This fact is shown on the seventh Counter print out from main. Then the rst button is pressed and  it return to REPL. Ctrl-C will not abort the running code. Some times it works but on later press of GP23 button it hangs.

boot.py is empty apart from a comment.
This is the code in main.py:

`# main.py -- put your code here!
from machine import UART
from network import WLAN
import sys
import os
import machine
import micropython

uart = UART(0, baudrate=115200)
os.dupterm(uart)  

micropython.alloc_emergency_exception_buf(100)
counter = 0
     
def pinInterrupt(pin):
	global counter
	counter += 1

def main():
		
	m1UpButton=machine.Pin('GP30', mode=machine.Pin.IN, 
							pull=machine.Pin.PULL_UP)
	print("Initital value %d" % m1UpButton.value())
	m1UpButton.irq(trigger=machine.Pin.IRQ_FALLING, handler=pinInterrupt)

	while True:
		s=machine.disable_irq()
		cnt = counter
		machine.enable_irq(s)
		print("Counter %d" % cnt)
		for i in range(20000):
			pass
		
	m1UpButton.irq().disable()
	print("Exit") 
`

I have also  tested,   

> robert-hh commented on 10 Sep • edited

 code above it does not return to main loop, allthough interrupt continue to work, but pressing ctrl.c does not make it return to REPL. Here is a short print out:
473385652 0
473385653 0
473385654 0
uSW USR1 pressed
uSW USR1 pressed
uSW USR1 pressed
uSW USR1 pressed
uSW USR1 pressed
uSW USR1 pressed

 Consequently the error is not corrected in V1.8.5

/Blueslow

robert-hh, 2016-10-18:
Did you disable threading in mpconfigport.h as suggested by Damien?

`#define MICROPY_PY_THREAD                           (0)` 
`#define MICROPY_PY_THREAD_GIL                       (0)`

blueslow, 2016-10-18:
No I did not.  I saw the note about it but forgot to mention int. I used the pre built release as is from http://micropython.org/download. I have not the right tool chain installed and I have not looked in to the details of building the firmware for micropython running on the wipy.

dpgeorge, 2016-10-19:
This issue should now be fixed by 17ba6ef5fa7585d3e91dada3044e8e755cf99537.  Please test and report back if it's working for you, thanks!

blueslow, 2016-10-19:
Hi,

Conclusion it works, at least in the short term. Good work Damien, thanks.

This is how I tested:
I downloaded zip from 
http://micropython.org/resources/firmware/wipy-20161019-v1.8.5-18-g84679e0.zip
extracted it and copied mcuimg.bin to /flash/sys/ with ftp and made a 
hard reset.

I have executed my script stated in 2406 
<https://github.com/micropython/micropython/issues/2406> and also 
Robert-hh's scripts during 10 minutes each. Both scripts now seems to 
work. Both scripts can also be interrupted by ctrl-c.  FTP and telnet 
connections works even during executing Robert-hh script (have not  
tested it with my script). FTP and telnet connections also works after 
the scripts been interrupted by ctrl-c.I have also built and downloaded  
the mcuimg.bin from the git repository and the scripts works equally well.

Another problem:
This is regardless of any scripts running.
However the FTP stops working after password is sent by the client, 
after connection is established when i build it my self. FTP client 
eventually times out, but the WIPY hangs and the heartbeat led goes off 
and remains so. until reset switch is pressed or wipy power cycled. This 
has also been observed in a previous build (1.8.5-11) which I made and 
has noting to do with 2406. 
<https://github.com/micropython/micropython/issues/2406> More likely 
something is wrong with my tool chain or me doing the wrong things when 
building.

/Klas

Den 2016-10-19 kl. 05:33, skrev Damien George:

robert-hh, 2016-10-19:
I cannot confirm the build issue. I updated to the latest state (git pull), rebuilt the image, downloaded it to WIpy. FTP still runs. Which client & which OS are you using?

blueslow, 2016-10-19:
Robert-hh, my ftp issue is a  completely different problem of what is discussed in this thread so people may regard this as offtopic.
Robert-hh the build issue is in my setup not in the image that I guess dpgeorge built (http://micropython.org/resources/firmware/wipy-20161019-v1.8.5-18-g84679e0.zip) which I used to confirm the corrections.  I'm using Linux mint 18 with filezilla. and that works unless I'm building the cc3200 mcuimg.bin by myself with the command make BTARGET=application BTYPE=release BOARD=WIPY. That problem  was also in the  V1.8.5-11, one strange thing allthough with that build was  that it worked initally but stopped working after I downloaded a couple of files and executed som scripts but after that it does not work. So the problem is eithher in my toolchain, my command that build the mcuimg or in the Wipy itself. What i would like to have is a means to instrument the FTP code with printouts to try to determine were it goes astray.  The toolchian that I'm using is the downloaded from the following source: https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa

closed 2024-08-28 by projectgus