micropython/micropython #2410: More fully featured websockets

danni, 2016-09-11:
The websocket and websocket_helper included with the ESP8266 is a bit weird to use and doesn't work as a client with most servers it seems. Would it be useful to include a more versatile implementation of websockets? I have a pure-python implementation of the client that looks like the `websockets` module that could (a) expand to include the server handshake, (b) support the webrepl, and (c) be ported to C if needed (not sure it is, it should express quite well in Python).

dpgeorge, 2016-09-12:
> The websocket and websocket_helper included with the ESP8266 is a bit weird to use and doesn't work as a client with most servers it seems.

Its main purpose was to be used as a server, so that's what it's capable of at the moment.  What kind of errors do you get using it as a client?  Maybe they are easy to fix.  Or maybe the additional functionality can be added in websocket_helper.py.

> Would it be useful to include a more versatile implementation of websockets?

Yes, I'd say so.  Then it can be used for more than just the WebREPL.  The decision to use websockets (and not some custom TCP protocol) was partly so it could be reused for other things.

> I have a pure-python implementation of the client that looks like the websockets module that could (a) expand to include the server handshake, (b) support the webrepl, and (c) be ported to C if needed (not sure it is, it should express quite well in Python).

The current implementation was mostly ported to C for efficiency, since you want the WebREPL to run nicely in the background and still have enough resources to do other things.  So going back to pure Python doesn't seem right.

danni, 2016-09-12:
Yeah it is fixable. It makes some assumptions the RFC says aren't valid. Now that I've written a Python implementation I reckon I could port it into C, although I don't fancy doing the pythonic parts like context managers. Maybe just the framing could be written in C and the rest in Python and it would be sufficiently performant.

pfalcon, 2016-09-12:
> I have a pure-python implementation of the client that looks like the websockets module that could (a) expand to include the server handshake, (b) support the webrepl, and (c) be ported to C if needed (not sure it is, it should express quite well in Python).

Then you won't be surprised to hear that modwebrepl.c is a conversion of initial Python version we wrote when experimented with webrepl. So, it's already ported to C, and already supports webrepl. And yes, it can be expanded further, just nobody needed that so far.

rosskusler, 2024-03-18:
I would love to see a more general websocket implementation in micropython. It would be great for an  IOT use case I have in mind. Still looking for a solution

mattytrentini, 2024-03-19:
> I would love to see a more general websocket implementation in micropython. It would be great for an IOT use case I have in mind. Still looking for a solution

Would the [Microdot Webocket extension](https://microdot.readthedocs.io/en/latest/extensions.html#websocket-support) help with what you're looking for @rosskusler?

rosskusler, 2024-03-19:
@mattytrentini thank you for responding. It looks like Microdot is a server side implementation. My scenario involves  micropython consuming websocket events. Are you aware of any good micropython websocket client librariies?

mattytrentini, 2024-03-19:
> Are you aware of any good micropython websocket client librariies?

Afraid not; I haven't had the need to use one (yet!).

jonnor, 2024-07-02:
Solutions seem to exist both for Websocket server and client. So I believe this issue can be closed or converted to a discussion.

closed 2024-09-08 by jonnor