micropython/micropython #2412: ESP8266 Error when measuring time intervals

peterhinch, 2016-09-12:
I have observed this on a number of occasions when timing code over long periods, but it is difficult to replicate reliably. Most test cases run indefinitely without issue. The following does eventually seem to fail if allowed to run for long enough (< 1hr).
```python
from machine import Timer
import time
def rats(t):
    global start, deltamax
    t = time.ticks_ms()
    delta = time.ticks_diff(start, t)
    deltamax = max(deltamax, delta)
    print(deltamax, delta, t)
    start = time.ticks_ms()
tim = Timer(-1)
start = time.ticks_ms()
deltamax = 0
tim.init(period=2000, mode=Timer.PERIODIC, callback=rats)
while True:
    if deltamax > 3000:
        tim.deinit()
        break
    time.sleep(1)
```
Outcome:
```
...
2001 2000 4291654
2001 2000 4293654
1069448857 1069448857 687
>>> 
```
The value of ``t`` seems to have rolled over (next expected value 4295654 which isn't an obvious candidate for rollover). Other test cases have rolled over without issue.

dpgeorge, 2016-09-12:
ticks_ms() is implemented as `system_get_time()/1000`, and system_get_time returns a 32-bit unsigned int.  So rolling over at 2^32/1000 is expected, and I think that's exactly what you are seeing above.  Question is, why don't you see this more often, it should be really easy to reproduce?

dpgeorge, 2016-09-12:
Actually, this shows a problem with the whole ticks_XXX stuff: in this case ticks_ms rolls around much earlier than ticks_us or ticks_cpu.  For ticks_diff to work correctly it should do modulo arithmetic with a divisor that is the minimum of all ticks_ms/ticks_us/ticks_cpu.  Ie it should be 2^32/1000.  That in turn means restricting ticks_us to only count to 4.3 seconds...

peterhinch, 2016-09-12:
> ticks_ms() is implemented as system_get_time()/1000

Light dawns! My other test cases used ticks_us() because I (wrongly) assumed this was a quicker way to test rollover. However those attempts failed to provoke the fault and rollover worked. Evidently, looking at the numbers, this ticks_ms() test is failing on rollover. Perhaps ticks_diff needs a modulo argument?

dhylands, 2016-09-12:
I suspect that ticks_ms needs to be use a mask/modulo in order to produce results which are compatible with ticks_diff

dpgeorge, 2016-09-13:
> I suspect that ticks_ms needs to be use a mask/modulo in order to produce results which are compatible with ticks_diff

ticks_ms needs to either increase it's wrap-around value to match ticks_diff, or ticks_diff needs to use a modulo to match ticks_ms (and then ticks_us and ticks_cpu need to use the smaller modulo).

Some solutions:

1) Make ticks_diff use 2^32/1000 for the modulo.
2) Increase wrap-around of ticks_ms to a full 32-bit unsigned int (and hence increase its range from 1hr to 1000hrs).  This would be done by having an overflow counter that is constantly checked, eg in ets_loop_iter.
3) Make ticks_ms/ticks_us/ticks_cpu return a typed integer, to distinguish the different ranges and hence different modulo values.  The low bit of the integer could be used for the type, ticks_ms would return an odd number and the other two, an even number.  Then ticks_diff can select the correct modulo based on the odd/even-ness.

I'd prefer option 2.

peterhinch, 2016-09-13:
+1 for option 2.

pfalcon, 2016-09-13:
> Make ticks_diff use 2^32/1000 for the modulo.

Keep in mind that division is expensive on esp8266.

Option 2 is definitely the way to go, the question is how to implement it. It's hard to believe that there's no tick counter. Well, there's - that "RTC" thing.

dpgeorge, 2016-09-13:
A simple fix would be something like:
```
uint32_t system_time_high_word, system_time_prev;

    // within ets_loop_iter
    uint32_t cur = system_get_time();
    if (cur < system_time_prev) {
        // overflow
        system_time_high_word += 1;
    }
    system_time_prev = cur;

uint32_t mp_hal_ticks_ms(void) {
    return ((uint64_t)(system_time_high_word << 32) + system_get_time()) / 1000;
}
```

pfalcon, 2016-09-13:
I still don't understand why the "simple" solution is to burden event loop with more stuff and not using pyb_rtc_get_us_since_2000() or its underlying functions.

dpgeorge, 2016-09-13:
> I still don't understand why the "simple" solution is to burden event loop with more stuff and not using pyb_rtc_get_us_since_2000() or its underlying functions.

Because the "RTC" is terribly inaccurate: without calibration I get 1 second true time being reported as 900ms according to the RTC.  With calibration the firrst 60 seconds loses about 50ms (ie 60 real seconds is reported as 59.95s), but then the next 60 seconds loses about 250ms.

pfalcon, 2016-09-13:
Ok then.

pfalcon, 2016-09-19:
@dpgeorge : So, please apply the fix above when you have a chance.

dpgeorge, 2016-09-20:
Fixed.  I tested it with the following code:
```python
import time, machine
rtc = machine.RTC()
while 1:
    print(time.ticks_ms(), time.ticks_us(), rtc.datetime())
    time.sleep(10)
```
I let it run for 100 minutes and it seems to work.  Partial output showing start, roll-over to 33 bits, and end is:
```
89728 89728515 (2000, 1, 1, 5, 0, 1, 16, 582)
99729 99729698 (2000, 1, 1, 5, 0, 1, 25, 102)
109730 109730571 (2000, 1, 1, 5, 0, 1, 33, 620)
...
4280114 1058889054 (2000, 1, 1, 5, 1, 0, 27, 466)
4290115 1068889980 (2000, 1, 1, 5, 1, 0, 35, 936)
4300116 5149121 (2000, 1, 1, 5, 1, 0, 44, 407) <---- here it rolls over
4310117 15150084 (2000, 1, 1, 5, 1, 0, 52, 877)
4320118 25151045 (2000, 1, 1, 5, 1, 1, 1, 348)
...
6060280 691571943 (2000, 1, 1, 5, 1, 25, 31, 333)
6070281 701572824 (2000, 1, 1, 5, 1, 25, 39, 755)
6080282 711573701 (2000, 1, 1, 5, 1, 25, 48, 177)
```

closed 2016-09-20 by dpgeorge