micropython/micropython #2414: Establish a no-increase-of-code-size regression check for pull requests

pfalcon, 2016-09-12:
Recently, we're getting increasing number of pull request, and many of these pull requests add features without paying attention to maintain minimal code size of MicroPython. I'd like to propose to establish a check, similar e.g. to travis or coveralls which would check that size of selected ports does not increase with a patch. If size increased, it would mark check as "failed" for further investigation. The list of ports mentioned above is definitely minimal port and minimal unix port, though it would be nice to establish control of main unix port size growth too. It's bound to grow in size, as new features are added constantly. That's exactly why I say that there should be control of the growing size, and the only means to establish it is apparently scrutinize which features are added to unix port, and which don't.

I'm writing this as someone who works to make sure that the code size at least of the minimal ports decreases over time, and unix port size stays under control (arguably, it's too big, and some features may need to be removed from the default config (like VFS was removed lately)).

dpgeorge, 2016-09-13:
Features are always going to add size to the firmware :)  But keeping minimal, unix/minimal and bare-arm approx constant does make sense (but even those ports will grow slowly, eg by moving to Python 3.5, 3.6 etc.

I'm not sure though how to make a custom github hook that mirrors the functionality of coveralls.  Maybe someone with experience writing such hooks can contribute something?

stinos, 2016-09-13:
_it would mark check as "failed" for further investigation_

Really great would be if it could instead, or in addition, also write the diff in code size but I guess Github has no API for that? But in any case I'm +1 on this except when it would lead to nearly every single commit being marked failed in which case it's useless clutter.

pfalcon, 2016-09-13:
> Features are always going to add size to the firmware :)

Yes, so it's important to have a policy on that. Unless I'm mistaken, we have such a policy, and it looks like it's time to formalize it down to algorithm, and raise every contributor's awareness of it.

> I'm not sure though how to make a custom github hook that mirrors the functionality of coveralls. Maybe someone with experience writing such hooks can contribute something?

We can always add check to travis.

pfalcon, 2016-09-13:
I forgot to mention that other projects have such a check, I can give Zephyr RTOS as an example, though I believe I saw more lately (well, any projects which targets resource-constrained systems and works in competitive area would have one, or it would lose to competitors by simple criteria).

deshipu, 2016-09-13:
I'm a big fan of having all the policies (at least the ones that can be) implemented in form of gate checks that apply equally to everyone contributing. However, I'm afraid that in this instance that wouldn't be the case -- things commited directly to the repository without a pull request and code review would completely circumvent this mechanism, meaning that it only applies to the unwashed masses.

stinos, 2016-09-13:
That logic applies to any kind of check, including this one.

pfalcon, 2017-03-27:
This was done in #2549 above, though in https://github.com/micropython/micropython/issues/2908 I consider to add similar check for unix port.

closed 2017-03-27 by pfalcon