micropython/micropython #2415: Discussion of Python 3.6 support

dpgeorge, 2016-09-13:
Python 3.6 beta 1 was released on 12 Sep 2016, and a summary of the new features can be found here: https://docs.python.org/3.6/whatsnew/3.6.html .  Being in beta stage means we can provide feedback on any of the new features/changes if we like.  A list of the major PEPs are:

* PEP 468 - Preserving the order of **kwargs in a function
* PEP 487 - Simpler customization of class creation
* PEP 495 - Local Time Disambiguation
* PEP 498 - Literal String Formatting
* PEP 506 - Adding A Secrets Module To The Standard Library
* PEP 509 - Add a private version to dict
* PEP 515 - Underscores in Numeric Literals
* PEP 519 - Adding a file system path protocol
* PEP 520 - Preserving Class Attribute Definition Order
* PEP 523 - Adding a frame evaluation API to CPython
* PEP 524 - Make os.urandom() blocking on Linux (during system startup)
* PEP 525 - Asynchronous Generators (provisional)
* PEP 526 - Syntax for Variable Annotations (provisional)
* PEP 528 - Change Windows console encoding to UTF-8 (provisional)
* PEP 529 - Change Windows filesystem encoding to UTF-8 (provisional)
* PEP 530 - Asynchronous Comprehensions

See #1329 for Python 3.5 support, which is still pending.

dpgeorge, 2016-09-13:
PEP 468, preserving the order of kwargs, is of particular interest to us because it might be hard to implement efficiently.  CPython 3.6 will have dict == OrderedDict (ie all dicts are ordered), using the PyPy implementation to make it efficient.  Essentially all dicts have 2 tables, one which is the ordered array of (key, value) pairs (ordered by order of insertion), and the other is a (compact) hash table that indexes the ordered array.

pfalcon, 2016-09-13:
I'd love to see "PEP 515 - Underscores in Numeric Literals" implemented. Easy and helpful. The rest, I'm more and more getting disappointed where CPython goes.

dpgeorge, 2016-09-13:
> I'd love to see "PEP 515 - Underscores in Numeric Literals" implemented

+10.  It would be pretty easy to add, and I agree really nice to then write something like `machine.freq(42_000_000)`.

dhylands, 2016-09-13:
>> I'd love to see "PEP 515 - Underscores in Numeric Literals" implemented
> +10. It would be pretty easy to add, and I agree really nice to then write something like machine.freq(42_000_000).

+1 here too. I've been using that while coding in rust and while a bit odd at first, it makes reading the larger integers easier.

dbc, 2016-09-16:
> I'd love to see "PEP 515 - Underscores in Numeric Literals" implemented. Easy and helpful. 

Oh yes, oh yes.  This is one my favourite language features in any language.  I love it for hex and binary literals.  Interestingly, at PyCon 2016 in Portland a few months ago, when the BDFL was reading over the list of PEP's, I cheered and applauded loudly for this one.... and about 2 thousand other people in the same room with me were dead silent.  :/ So I'm thinking us embedded programmer types have different priorities.

As to the rest, I'm thinking that preserving order in kwargs is hardly worth the expense of memory in Micropython, for as often as it is used. OTOH, it may be a forced march if it becomes important for code portability.  OTOOH, since we have 'micro' versions of the important libraries anyway, perhaps porting is a given in any case.

PEP530 just tickles me. Not sure it's of great value to Micropython, but I love it.

(Yes, I'm back!  Been away from Micropython since about last December. Sometimes, life happens, and good stuff falls below the cut-off line on the to-do list.  Hopefully, Micropython can crawl back above the line for a while.)

peterhinch, 2016-09-16:
@dbc Welcome back! I don't follow your point about PEP530 though - improved async support is surely welcome. Many embedded applications rely on asynchronous code.

dbc, 2016-09-16:
@peterhinch re: PEP530 -- I should have been more clear about my fear of RAM bloat. Totally agree that asynch is great for embedded, I'm a big fan.  I'd use async comprehensions if they were available, certainly -- like a cat on catnip.  And of course we want Micropython to be as close to 100% source compatible with CPython as is practical.  My fear is the "practical" part in Micropython -- if asynch comprehensions cause the SRAM footprint to grow significantly for programs using non-async list comprehensions or non-async generator expressions, then I'd be wary of it.

Thinking about it...  the semantics of an async comprehension are satisfied by just doing a normal comprehension.  The completion order is arbitrary, after all. So maybe the syntax can be accepted but the functionality could be a normal comprehension.

peterhinch, 2016-09-16:
@dbc Will the availability of the async keyword affect the bytecode emitted when the keyword is absent? This surely depends on the cleverness of the compiler. But I'll leave this to people who actually understand compilers :)

pohmelie, 2016-09-20:
PEP 498 - Literal String Formatting. It's step forward and "replace both existing % and str.format" looks like best solution for me. Definitely I want it in µpy.
PEP 525 - Asynchronous Generators (provisional). This is just awesome, since this can replace pretty ugly `async for` context class. It is just same as generator and `__enter__`/`__iter__` methods of object. But, this requires too much changes, since this can't be done just as sugar for `async`/`await` in µpy right now. And probably no one need this in µpy.

dbc, 2016-09-20:
PEP 498: I read that in detail now.  So.. PEP 498 looks like the future. I never use %-formatting in new code.  When PEP 498 strings happen in CPython, I will probably go exclusively to PEP498 format strings for new code and never look back, they are much more readable.

@dpgeorge says " likely too much bloat " 

Ignoring PEP 498 means sacrificing CPython compatibility in a very visible way.  Removing .format() in favor of literal string formatting is a very visible breaking change.  Both of those options hit every user every day.

My naive assumption would be that much of the low-level formatting code could be factored out of .format() and literal string formatting.  Pre-processing the PEP 498 strings would add to the compiler, certainly.  Or is it RAM bloat and not code bloat that is the concern?  An f-string becomes a byte-code expression in RAM instead of a simple string.  Is this the concern?

I guess if you twisted my arm, I'd choose dropping %-formatting and .format() in favor of f-strings if I had to choose between f-strings and .format().

pohmelie, 2016-09-20:
@dbc, one thing is pretty annoying. If you want have only f-strings and got template from somewhere as string you should do something like
``` python
>>> s = "{x}"
>>> x = 1
>>> eval(f"f'{s}'")
'1'
```

deshipu, 2016-09-20:
You can still do:
```python
>>> s = "{x}"
>>> x = 1
>>> s.format(x=x)
```

pohmelie, 2016-09-20:
@deshipu, @dbc and I talking about «replace both existing % and str.format», so there is not str.format/format/%

deshipu, 2016-09-20:
I see, I didn't catch that. In any case, they are not getting removed, so more power to you.

pohmelie, 2016-09-20:
@deshipu, as @dpgeorge said in first post this is possible in theory.

dbc, 2016-09-20:
>  If you have only f-strings and got template from somewhere as string you should do something like ...

Yes, that is somewhat klunky.

peterhinch, 2016-09-21:
@pohmelie @dbc said
>Removing .format() in favor of literal string formatting is a very visible breaking change. Both of those options hit every user every day.

I agree that removing .format() would be very bad news. Likewise the idea of removing the string modulus operator. In my view such changes would break far too much code and would be a worse departure from cPython compatibility than opting not to support f-strings.

ARF1, 2017-03-02:
+1 For [PEP 487](https://www.python.org/dev/peps/pep-0487/) - Simpler customization of class creation

This is deceptively named. It is MUCH saner and light-weight replacement for metaclasses in many cases. Also, it should be fairly simple to implement given that the machinery will be fairly similar to `__new__`.

scls19fr, 2018-09-20:
[f-string](https://realpython.com/python-f-strings/) is something I'd love to see in MicroPython
[PEP 498 -- Literal String Interpolation](https://www.python.org/dev/peps/pep-0498/)

kamikaze, 2018-12-08:
Kill the .format() , voting for f-strings ! broken code is an abandoned code in 99% cases. I don't see any reason to not update the code for this change.

mattytrentini, 2019-04-01:
I believe PEP-15 is implemented in at least 1.10. I think we can strike it through!

dpgeorge, 2019-04-02:
PEP 515 (underscores in numeric literals) was implemented in 6a445b60fad87c94a1d00ce3a043b881a1f7a5a4

asottile, 2019-05-02:
for those looking for another way to get f-strings, I've added better support to `future-fstrings` for use as a preprocessor: https://github.com/asottile/future-fstrings#transform-source-for-micropython

It has a PEP-498 compliant source transformer that produces equivalent `'...'.format(...)` calls

pmp-p, 2019-09-20:
thank you @asottile but sadly it seems, we won't benefit of bpo-36817 since https://github.com/asottile/future-fstrings/issues/40 was closed.

Where is the "Discussion of Python 3 support" status ? i may have missed it. I hardly see *any* 3.x future support with notimpl on major language features ( including some that exist since 2.x )

i only found 3.5 https://github.com/micropython/micropython/issues/1329

asottile, 2019-09-20:
that issue is still doable, I just don't really feel motivated to work on it -- if someone wants to pick it up then by all means do so :)

jimmo, 2019-09-20:
For f-strings, did you see #4998 referenced above?

pmp-p, 2019-09-22:
@jimmo *no* i'm not searching for 3.6/3.7/3.8 totallly optionnal features from #4998

Apart from the warm thanks addressed to the person deserving them for adaptation ( after all he did not have to take the time to add that ) of his usefull tool for use with micropython in replacement of clumsy https://github.com/pmp-p/fstrings_helper my question was :

> Where is the "Discussion of Python 3 support" status ? i may have missed it. I hardly see any 3.x future support with notimpl on major language features ( including some that exist since 2.x )

i only found 3.5 #1329, i'm more interested for completion of 3.4 and reaching stability.

kevinkk525, 2019-10-30:
Any chance to implement PEP 526 (Syntax for Variable Annotations (provisional))?

I understand that it doesn't have any impact beyond the parser but would be hugely beneficial in development (at least to me it would be). I guess if it is too much work then using [strip-hints](https://pypi.org/project/strip-hints/) is an option. Running it on each script before uploading/freezing would be an acceptable workaround if it works reliable.

dpgeorge, 2019-10-31:
> Any chance to implement PEP 526 (Syntax for Variable Annotations (provisional))?

Yes I would like to see this implemented soon.  It would then allow more sophisticated type hinting, and can also get use in viper annotations.