micropython/micropython #2417: Consider merging alternative parser implementation from microbit port

pfalcon, 2016-09-13:
MicroPython "starts" to "lose" to competition when it comes to amount of memory required to get source code to run (i.e. parse/compile it). For example, JerryScript has hand-written parser which doesn't have AST phase. Well, MicroPython has it to AFAIK, in microbit port, so it would be nice to merge it to mainline unless there'e compelling reasons not to.

@dpgeorge, Can you please explain what were the requirements when writing microbit parser, and what drawbacks it has comparing to the main parser?

dpgeorge, 2016-09-13:
The microbit parser can be found on the parse-bytecode branch of this repo.  The reason for it was to be able to parse/compile scripts in less memory.  I don't remember the details of the metrics, but it uses something like 2/3 of the RAM (heap) than the one on master, and has slighly larger firmware size (performance of it is unknown).

Here is how it works.  The current parser in master builds an AST from parse nodes, with tradtional pointers creating the tree.  Eg a node might be the struct (type0, node1, node2), where node1 and node2 are word-sized pointers that point to the sub-nodes.  Eg node1=(type1, arg1) and node2=(type2, arg2).

In the microbit version the parse tree is stored as "parse bytecode" with the nodes nested in each other, and all the pointers eliminated.  So the example above is: (type0, type1, arg1, type2, arg2).  In this case the types and args can be compressed to use just a single byte in some cases.  The number of args for a given node is usually known implicitly, otherwise it's explicitly stored.

The reason not to merge this code would be 1) code size increases; 2) probably it's slower to parse scripts.  I'd need to rebase the parse-bytecode branch and do some proper measurements to be sure.

There are other ways to reduce RAM usage, eg by incrementally compiling the script (each top-level entity like a class or module-level-function can be compiled as soon as it's finished being parsed, regardless of what comes next in the script).

pfalcon, 2016-09-13:
I see, so parse-bytecode still retains all other features of AST-based implementation, like being able to post-process and optimize AST? JerryScript claims to use handwritten recursive-descent parser. It then allows to parse short input in the very small memory. At the expense of no AST-level optimizations and complicated parser maintenance in case new rules need to be added (not issue for JS, an issue for Python (well, CPython)).

dpgeorge, 2016-09-13:
> I see, so parse-bytecode still retains all other features of AST-based implementation, like being able to post-process and optimize AST?

Yes.  I don't see anyway around it, since you need to look ahead for the scope of variables before they can be compiled to bytecode (or native code).  Eg if you define a nested function that closes over variables in the outer function.

But is it really needed to be able to compile/execute toy examples like "print(1+2)" in, say, 50 bytes of RAM?  The microbit already shows that uPy can work well with a 9.5k heap and 2k stack.  Anything less than that is not very realistic.

pfalcon, 2016-09-13:
> But is it really needed to be able to compile/execute toy examples like "print(1+2)" in, say, 50 bytes of RAM? The microbit already shows that uPy can work well with a 9.5k heap and 2k stack. Anything less than that is not very realistic.

It's useful (and easy to interpret/compare) metric. Also, titles of posts from esp8266 forum: "MemoryError when importing a custom module", "Potential urequests out of memory?", "Memory Error Strikes Back", etc., etc.

dpgeorge, 2016-09-13:
> It's useful (and easy to interpret/compare) metric.

Perhaps.  But it's not necessarily reflecting the true cost of a real program.  So I don't think it's worth spending time on.

> Also, titles of posts from esp8266 forum: "MemoryError when importing a custom module"

Parser improvements can only help a little here, because ultimately the bytecode all needs to be stored in RAM.  Frozen bytecode (perhaps dynamically frozen) is really the only way forward in these cases.

To quantify these problems one needs to profile the amount of heap used for the parse/compile stage, and then the amount of heap taken by the final bytecode.

pfalcon, 2016-09-13:
> To quantify these problems one needs to profile the amount of heap used for the parse/compile stage, and then the amount of heap taken by the final bytecode.

Yes, but I'm sure 50% of complications is due to AST storage.

Anyway, as parse-bytecode offers some improvement, but not dramatic changes to parsing behavior, I guess this can be closed for now.

dpgeorge, 2016-09-14:
Here is some rough data to compare the parse-bytecode algo with the traditional algo (at v1.7 of uPy).

[image]

I got it to parse and compile urequests.py, unix version on 32-bit i386 arch.  The horizontal axis is the sequence of malloc/realloc/free calls that are being traced (there is no GC running).  The lines are made of the following pieces (for both of the lines):

- The initial up-slope is the parser allocating RAM for the parse tree.
- The first substantial dip is the parser freeing its temporary data structures.
- The next up-slope is the compiler allocating RAM for the bytecode and other data structures (eg scope of variables).
- The final down-slope is the freeing of the parse tree.
- The final up-slope at the end is the execution of the script (which doesn't do anything, just loads the urequests module).

The parse-bytecode algo looks like it doesn't fully free all its RAM because the final memory used at the end should match for both lines (but it doesn't).

Very roughly you could say that the parse-bytecode algo has peak RAM (heap) usage of 50% of the traditional algo.

pfalcon, 2016-09-15:
> Very roughly you could say that the parse-bytecode algo has peak RAM (heap) usage of 50% of the traditional algo.

I'd say it's pretty worthy improvement. Also, parsing/compiling by a top-level object would be great too. I don't know what to do about this ticket though, because what we should do is working on esp8266 stretch goals. But this ticket is at least related to reports of multiple esp8266 users.

dpgeorge, 2018-12-04:
I forgot one very important limitation of the micro:bit alternative parser: it requires a *continuous* region of heap memory to store the *entire* parse bytecode.  On small heap systems like the micro:bit, with small scripts, this works ok because usually there is only main.py to compile and then the program runs (so there is enough continuous memory at the start to compile it).  But on systems that import scripts during execution it won't work well because there is unlikely to be a large continuous chunk of heap available.  This is not easy to change, it requires adding a level of abstraction when iterating through the parse bytecode so the bytecode can be split into different memory chunks.

So I'll close this issue because it's unlikely there will be further progress integrating this parser.

closed 2018-12-04 by dpgeorge