micropython/micropython #2419: I2C bus recovery

deshipu, 2016-09-14:
I wonder how useful would it be to add a bus recovery mechanism to the software I2C implementation. It would be something similar to what is discussed here: https://github.com/esp8266/Arduino/issues/1025

The particular use case is when either the slave or the master get reset in the middle of a transmission, and leave the bus in inconsistent state, usually with the other party pulling one of the lines down and preventing communication. This can easily happen when we are doing aggressive power management, entering deep sleep or cutting power to the sensors.

The proposed solution is to strobe the clock line several times and send a NACK, to make sure all the devices on the bus get to finish the last byte they were sending.

rolandvs, 2016-11-25:
IMHO a graceful exit is **preferred** so this would be a good addition to the robustness of the I2C functionality.

MaartenJB, 2017-02-07:
@rolandvs, I agree, this would be really nice, particularly when working in a noisy environment.