micropython/micropython #2421: machine.irq_enable() reboots esp port

thetreerat, 2016-09-15:
when I try to issue a machine.irq_disable(),  machine.irq_enable(True) or machine.irq_enable(False) the board reboots. 

it takes about 15-25 seconds after you hit return in the repl. you can't type or get a >> prompt in the 15-25 seconds then it posts a boot sequence. 

 I see it documented  here http://docs.micropython.org/en/latest/esp8266/library/machine.html

am I using it wrong?  some help please

thetreerat, 2016-09-15:
sorry I for got I using the 1.8.4 stable release

dpgeorge, 2016-09-15:
Sorry, the docs are not correct.  In implementing the esp version of disable_irq/enable_irq it was decided that the return value of disable_irq, and the argument to enable_irq, should be an *opaque* value.  It may be an integer, a bool, or something else.  To use it correctly you should pass the return value from disable_irq to enable_irq to restore the irqs to the state they were in before calling disable_irq.  Ie:
```python
# at this point P in the code the irqs are in some state

state = machine.disable_irq()
# irqs are disabled
# do some critical stuff!
machine.enable_irq(state)

# now, irqs are back in the same state as they were at point P above
```

thetreerat, 2016-09-15:
Thanks, I try this.

dpgeorge, 2016-09-23:
Docs are fixed in eeb9d99333fceaf924f7e367325c584d0a07cb48.

closed 2016-09-23 by dpgeorge

skimj, 2017-02-23:
I'm experiencing the same behavior. (with esp8266-20170108-v1.8.7.bin)
`
import machine
irqstate = machine.disable_irq()
`
(FWIW disable_irq() returns 32 if not assigned to a variable)

At this point the REPL prompt is frozen and the board reboots after a delay (approximately 15s like OP). Am I missing an earlier config step or is this a bug?

skimj, 2017-02-23:
Now I can't recreate my initial bug and everything is working for me. My previous comments are still valid but only in a slightly weird use case: If you do have not initialized an IRQ, then trying `disable_irq()` causes the reboot. After first defining a pin IRQ, `disable_irq()` seems to be working OK (FYI it still returns 32 in this case).

I wish I knew what my original issue was that I've unknowingly resolved...

dpgeorge, 2017-02-24:
> At this point the REPL prompt is frozen and the board reboots after a delay

Yes the board will reset if you have irqs disabled for too long.  And the REPL won't work either because the UART can process chars.  You should only disable and enable irqs in a function.