micropython/micropython #2422: Subclassing issues

CRImier, 2016-09-15:
Hello! 

I've written a small test for subclassing - I'm porting my HD44780 driver. However, I'm having problems with an example as simple as this:

```
#contents of class_test.py
def BaseClass():
    var = 0
    def __init__(self):
        print(self.var)

def TestClass(BaseClass):
    def __init__(self):
        print(self.var)

cl = TestClass()
#EOF

>>> import class_test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "class_test.py", line 10, in <module>
TypeError: function takes 1 positional arguments but 0 were given
```
Could you help?

That's a test case I've decided to make when I've gotten this error:

```
>>> import uhd44780_gpio
>>> from machine import Pin
>>> screen = uhd44780_gpio.Screen(pins=[Pin(2), Pin(0), Pin(4), Pin(5)], en_pin=Pin(13), rs_pin=Pin(15))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "uhd44780_gpio.py", line 25, in __init__
  File "uhd44780.py", line 59, in __init__
  File "uhd44780.py", line 69, in init_display 
  File "uhd44780.py", line 83, in home
AttributeError: 'Screen' object has no attribute ''
```
A class from uhd44780_gpio (Screen) is inherited from uhd44780 (class is HD44780) but doesn't seem to inherit properties from HD44780. I can attach those files if necessary.

deshipu, 2016-09-15:
I think I see the problem. You defined them as functions, with the `def` keyword, not as classes, with the `class` keyword.

dpgeorge, 2016-09-15:
You need to use `class` to define classes, not `def` (which defines a function).

CRImier, 2016-09-15:
You're goddamn right, gentlemen. Sorry. I think I'll re-open a function once I can reproduce the real issue with ```AttributeError: 'Screen' object has no attribute ''``` .

closed 2016-09-15 by CRImier

deshipu, 2016-09-15:
Do you have your code published somewhere?

CRImier, 2016-09-15:
@deshipu Unfortunately, not. It's the same idea that I have [here](https://github.com/CRImier/pyLCI/tree/master/output/drivers) (hd44780.py contains the base class and all the others have lower-level interfaces for actually writing bytes through different interfaces). Once I port it, I plan to post it to GitHub and possibly make a MicroPython package.

dhylands, 2016-09-16:
@CRImier I have one over here, setup similarly to how you describe:
https://github.com/dhylands/python_lcd

lcd.py has the common code, and the other files implement various HALs or tests for the HALs.

CRImier, 2016-09-16:
@dhylands Yes, thank you very much for writing this, I've looked through it to see how to implement various bit manipulation techniques as well =) However, I like my interface much more - it's simpler and more concise IMO, though I've not implemented any 8-bit interfaces - but it'd be very simple to if a need arises. Also, it is ESP-ready, that is, not depending on pyb module.  Overall, the idea is pretty much the same, indeed =)

dhylands, 2016-09-16:
Yeah I wrote it long before the machine module existed (and actually before MicroPython even). I think it should be as simple as replacing pyb with machine. I'll be curious to compare the implementations when you're done.

CRImier, 2016-09-16:
@dhylands @deshipu Check [the code](https://github.com/CRImier/MicroPython-modules/tree/master/uhd44780) out =) Now tested with basic functions. Easily expandable, low-level interfaces take the bare minimum.