micropython/micropython #2423: "MyObject" object has no attribute ''

CRImier, 2016-09-15:
Hello!

I'm preparing a MicroPython workshop and I have some code for HD44780 ported from CPython and I can't see what's wrong with it, and the exception message doesn't help either.

```
>>> import gpio_screen
>>> from machine import Pin
>>> screen = gpio_screen.Screen(pins=[Pin(2), Pin(0), Pin(4), Pin(5)], en_pin=Pin(13), rs_pin=Pin(15))
#prints dir(self)
['noDisplay', 'setCursor', 'noCursor', 'LCD_ENTRYSHIFTDECREMENT', 'LCD_SETDDRAMADDR', 'LCD_MOVERIGHT', 'LCD_DISPLAYOFF', 'displaycontrol', 'LCD_CURSORSHIFT', 'LCD_DISPLAYON', 'println', 'cursor', 'row_offsets', 'LCD_CURSOROFF', 'displaymode', 'LCD_BLINKOFF', 'clear', 'display', 'LCD_5x10DOTS', 'LCD_FUNCTIONSET', 'createChar', 'blink', 'noAutoscroll', 'LCD_ENTRYMODESET', 'LCD_1LINE', 'noBlink', 'LCD_ENTRYSHIFTINCREMENT', 'LCD_8BITMODE', 'LCD_CLEARDISPLAY', 'LCD_DISPLAYMOVE', 'LCD_CURSORON', 'LCD_5x8DOTS', 'LCD_BLINKON', 'display_function', '__init__', 'LCD_DISPLAYCONTROL', '__module__', 'LCD_ENTRYLEFT', 'LCD_4BITMODE', 'LCD_ENTRYRIGHT', 'LCD_2LINE', 'home', 'LCD_MOVELEFT', 'scrollDisplayLeft', '__qualname__', 'LCD_SETCGRAMADDR', 'leftToRight', 'rightToLeft', 'init_display', 'autoscroll', 'LCD_CURSORMOVE', 'scrollDisplayRight', 'LCD_RETURNHOME', 'write_byte', 'write4bits', 'rs_pin', 'pins', 'en_pin', 'cols', 'rows']
#executes self.write_byte(self.LCD_RETURNHOME)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "gpio_screen.py", line 78, in __init__
  File "gpio_screen.py", line 88, in init_display
  File "gpio_screen.py", line 103, in home
AttributeError: 'Screen' object has no attribute ''
```
It's pretty cryptic for me, and almost looks like some invalid character present in file. If I replace ```self.LCD_RETURNHOME``` with an integer, it works until the next place where another constant from the class is called where it fails with the same error. Any ideas?
[gpio_screen.py renamed as .txt attached](https://github.com/micropython/micropython/files/474681/gpio_screen.txt)

danicampora, 2016-09-15:
```python
const()
```
It's a MicroPython specific optimization that doesn't exist in CPython. In order to make your code work, simply remove the `self.` from all the class constants that you defined as `const()`. @dpgeorge should be able to explain in detail why it works like this.

CRImier, 2016-09-15:
Ohh. Understood. So they'll be basically defined as module-wide globals?

This is counter-intuitive. Is it mentioned anywhere in references? I can write it if necessary.

blazewicz, 2016-09-15:
It's mentioned here: https://docs.micropython.org/en/latest/pyboard/reference/speed_python.html#the-const-declaration

peterhinch, 2016-09-15:
The compiler replaces the name of the constant with the numeric value, rather like a #define in C. I think the error is thrown because the compiler sees self.42 in place of self.FORTY_TWO.

The syntax is more consistent if the constants are declared as globals rather than class variables. Another twist (documentation in progress) is as follows. If a constant name is prepended with an underscore the constant is unavailable to other modules so the compiler avoids allocating RAM for it:
```python
_FORTY_TWO = const(42)
```

CRImier, 2016-09-15:
@peterhinch Ahha. Got it.

One more question about that underscore thing - does it mean that if I have a base class with underscored constants in one file (and all the functions using them in that file as well) and have a subclass that the base class uses for a lower-level interface, passing constants by values to it (like write_byte which is implemented in subclass), it will work and consume less RAM?

CRImier, 2016-09-15:
Oh, and just to mention  - I got it working. The display is not responsive, unfortunately, but my logical analyzer shows everything's sent correctly, so the code itself should work.

dpgeorge, 2016-09-16:
> I think the error is thrown because the compiler sees self.42 in place of self.FORTY_TWO.

Yes, this is what it does at the moment and I'd class this as a bug in the parser.

peterhinch, 2016-09-16:
@CRImier The leading underscore syntax uses a feature of Python (not just MicroPython) whereby globals with a leading underscore are module private: they cannot be imported to another module. MicroPython maintains Python compatibility while avoiding allocating RAM when such variables are declared with const().

The special significance of the leading underscore applies only to globals. In classes a leading underscore signifies private by convention but is not enforced by the language, hence RAM must be allocated. The compiler still performs the literal substitution where any variable declared as const() is used.

dpgeorge, 2016-09-27:
Fixed by 6d310a5552f43b3f984a5d8020b41301ab7937a4.  Tests in 38b54b65d4b504339cd16c88175c31f257a68f3a.

closed 2016-09-27 by dpgeorge