micropython/micropython #2430: Can not initialize WDT on esp8266

muhlpachr, 2016-09-17:
```
>>> from machine import WDT
>>> wdt = WDT(timeout=2000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: function does not take keyword arguments
```

muhlpachr, 2016-09-17:
```
>>> wdt = WDT(2000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError:
```

dpgeorge, 2016-09-18:
Please use `WDT()`  to create the default WDT.  At the moment it doesn't support setting the timeout.

muhlpachr, 2016-09-19:
What is timeout of default WDT pelase ?
Will wdt.feed() work as expected  ?
Documentation describe planned implementation or is it copy&paste mistake from other port please ?
http://docs.micropython.org/en/latest/esp8266/library/machine.WDT.html
Thank you.

pfalcon, 2016-09-19:
@muhlpachr : Please see https://github.com/micropython/micropython/issues/2154

pfalcon, 2016-09-19:
> Documentation describe planned implementation or is it copy&paste mistake from other port please ?

Please see first section of the standard library description:
http://docs.micropython.org/en/latest/esp8266/library/index.html

closed 2016-09-19 by pfalcon