micropython/micropython #2431: alloc_emergency_exception_buf on esp8266 do not exist

muhlpachr, 2016-09-17:
```
>>> import micropython
>>> micropython.alloc_emergency_exception_buf(100)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'alloc_emergency_exception_buf'
>>> dir(micropython)
['__name__', 'mem_info', 'qstr_info', 'stack_use', 'heap_lock', 'heap_unlock']
```

dpgeorge, 2016-09-23:
I guess it can be added, it's a useful function to have.

Would be nice to have a shorter name though, `alloc_emg_exc_buf` anyone?

deshipu, 2016-09-23:
How about just `exc_buffer`? It's implied that we are setting its size, and 'emergency' doesn't add much information.

hosaka, 2016-09-27:
I would agree that "emergency" is not as helpful, but would vote for expanding the "exc" to `exception_buf` or `exception_buffer`. I'm happy to submit a patch for this.

dpgeorge, 2016-09-28:
It needs the word "emergency" because otherwise you'd think it was allocating a general buffer for use by all exceptions.

deshipu, 2016-09-28:
Normally I would agree that spelling out "exception" is a good idea, however, with the limited memory and with "exc" being used to mean "exception" in Python's standard library already (like here https://docs.python.org/2/library/traceback.html#traceback.print_exc), this is not so painful.

How about using a more meaningful word than "emergency" to indicate when that buffer is used? Perhaps `callback_exc_buf` or `interrupt_exc_buf`?

dpgeorge, 2016-09-28:
> How about using a more meaningful word than "emergency" to indicate when that buffer is used?

What about `alloc_fallback_exc_buf`, or even `alloc_fallback_exc`?

deshipu, 2016-09-28:
I think that `fallback` hits the meaning well, not sure if we need `alloc` or any other verb there, but I'm not a native speaker.

pfalcon, 2016-09-28:
Given that I have (and going to have) bunch of proposals of what to change in uPy API(s), I'd compensate that by being conservative with changing something which could be not changed (IMHO of course), like this case, and give my -0.01 on it.

However, I'd say that if there's pressure to cut whole words out, then "exception buffer" already gives enough hints that it's a special helper for exceptions (we don't buffer exception normally, so if we talk about exception buffer, that's already something for exceptional conditions, d'oh). That naturally contracts to exc_buf. And if there's future desire to cut chars from alloc_exc_buf(), can be made set_exc_buf().

pfalcon, 2016-10-11:
If there's no consensus on renaming the function, I guess it still should be enabled for 1.8.5.

dpgeorge, 2016-10-13:
Ok, let's defer discussion about changing the name.

Enabled in ed878275b020c006ec197bb5f5bdd8179fb36a4c.

closed 2016-10-13 by dpgeorge