micropython/micropython #2432: WebREPL is turned off by default

iBobik, 2016-09-18:
[In docs](http://docs.micropython.org/en/latest/esp8266/esp8266/tutorial/repl.html#webrepl-a-prompt-over-wifi) there is assumed WebREPL on ESP8266 is turned on by default, just connect to it.

> Once you are on the same network as the ESP8266 you click the “Connect” button (if you are connecting via a router then you may need to change the IP address, by default the IP address is correct when connected to the ESP8266’s access point). If the connection succeeds then you should see a welcome message.

But in boot.py it is commented out by default:

```
# This file is executed on every boot (including wake-boot from deepsleep)
#import esp
#esp.osdebug(None)
import gc
#import webrepl
#web
repl.start()
gc.collect()
```

So in reality we have to first turn on WebREPL by serial REPL, then we con connect to it by network:

```
import webrepl
webrepl.start()
```

dpgeorge, 2016-09-19:
In the branch esp-extra-scripts a patch is applied to uncomment these lines, and hence start webrepl on boot.  This branch is the one that is built to create the official, stable releases at https://micropython.org/download/#esp8266.  Daily builds don't have webrepl enabled, and I added a comment on that download page to explain this.

closed 2016-09-19 by dpgeorge

pfalcon, 2016-09-19:
Also http://docs.micropython.org/en/latest/esp8266/esp8266/quickref.html?#webrepl-web-browser-interactive-prompt . Also http://forum.micropython.org/viewtopic.php?f=16&t=2140 .

deshipu, 2016-09-19:
Perhaps it would be prudent to also update the documentation for which the issue was reported in the first place? Because you know, that is where people are going to look.