micropython/micropython #2437: same methods different names

kamikaze, 2016-09-20:
Is it possible to make consistent method names for same classes like I2C for different ports? Because now it is not easy to write portable code:

        if self.pyboard:
            res = self._i2c.mem_read(3, FRAM_SLAVE_ID >>1, device_addr <<1)
        else:
            res = self._i2c.readfrom_mem(FRAM_SLAVE_ID >>1, device_addr <<1, 3)

deshipu, 2016-09-20:
The work on this is continuing. The idea is that if you import stuff from `pyb`, it has the old API (so as to not break people's programs), and if you import it from `machine`, it will have the new API.

dpgeorge, 2016-09-27:
This is kind of a duplicate of #2449.  To write portable code please use the machine module and the methods it defines for the peripherals.  pyboard does have machine.I2C but it's software only at the moment.

pfalcon, 2017-05-14:
Hopefully above comments answer it.

closed 2017-05-14 by pfalcon

dpgeorge, 2017-05-14:
stmhal/pyboard's machine.I2C class is now fully function, uses HW acceleration and it's recommended to use it as a complete replacement for pyb.I2C.