micropython/micropython #2445: Add callbacks to network API

iBobik, 2016-09-21:
Currently when I want to write code reacting to network status (do something when connected/disconnected), it is pretty hard to do more things in one time - I have to make a loop with periodical network checks and other things.

Will be much cleaner to be able to write in callbacks and control blocking.

```
import network
sta_if = network.WLAN(network.STA_IF)
sta_if.active(True)
sta_if.connect('SSID', 'pass', blocking=True)

def sta_reconnect(status):
  if not sta.isconnected():
    # Reconnect
sta_if.status(callback=sta_reconnect)

# Do usual things in loop or in other callbacks
```

Without callback it is not much better than write it in the C :-)

peterhinch, 2016-09-22:
Have you considered using uasyncio?

iBobik, 2016-09-22:
Yes, it just makes periodic checks more readable, but callback is something much better.

Jan Pobořil
http://honza.poboril.cz/

22. 9. 2016 v 7:46, Peter Hinch <notifications@github.com>:

> uasyncio

pfalcon, 2016-09-22:
What makes you think this is possible in a general case?

iBobik, 2016-09-23:
I assume it is possible until relevant people tell it is too hard or impossible and why. :-)

pfalcon, 2017-05-14:
No detailed RFC, patches, etc. Closing. Feel free to reopen with suitable backing.

closed 2017-05-14 by pfalcon