micropython/micropython #2446: frozen dictionary comprehension - NotImplementedError: byte code not implemented

ryannathans, 2016-09-22:
Created a frozen module called test.py with the following contents

`a = {b: b for b in range(6) if b != None}`

compiled stmhal as usual with `make -j12 BOARD=STM32F439 FROZEN_MPY_DIR=scripts`

Test:
```
>>> import test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 1, in <module>
NotImplementedError: byte code not implemented
```

The code works fine if test.py is placed on the filesystem instead of freezing it in.

ryannathans, 2016-09-22:
Update: the same problem is also occurring with list comprehension

ryannathans, 2016-09-22:
forgot to recompile mpy-cross..... fuck

closed 2016-09-22 by ryannathans

gratefulfrog, 2016-10-06:
I got this problem also.

Is there any way of knowing when the mpy-cross compiler needs to be rebuilt?

I mean after a fetch/pull, how can I know what needs rebuilding? Is there a general top level makefile that could be used to automatically rebuild whatever is needed?

Sorry for my ignorance.

Thanks,
Bob

dpgeorge, 2016-10-07:
> Is there any way of knowing when the mpy-cross compiler needs to be rebuilt?

Not really.

The change I made that induced this error should have changed the version of the .mpy files, but I forgot to do that.  In future I'll try to remember, and then you'll get a "version mismatch" error when using an out-of-date version of mpy-cross.