micropython/micropython #2449: stmhal: Should machine.I2C be using SW I2C or HW I2C?

dhylands, 2016-09-22:
In commit e4f963a3 @dpgeorge changed machine.I2C to use the generic I2C interface. My reading of this is that its doing bit banged I2C.

Is this the intention?

Shouldn't machine.I2C be using HW I2C whenever possible?

This was brought up over in the forum when somebody noticed that machine.I2C.scan is producing different results from pyb.I2C.scan on the same hardware.

https://forum.micropython.org/viewtopic.php?f=15&t=2426

dpgeorge, 2016-09-23:
> Shouldn't machine.I2C be using HW I2C whenever possible?

Yes it should, but I didn't get to that part yet :)

> somebody noticed that machine.I2C.scan is producing different results from pyb.I2C.scan on the same hardware.

That's strange.  The differences between pyb.I2C.scan and machine.I2C.scan are:

1) pyb.I2C.scan does a write request with no data, while machine.I2C.scan does a read request for no data.

2) pyb.I2C.scan has clock stretching, while machine.I2C.scan does not (it does now, but not for the version of the firmware in the forum report).

Would be good to work out how to improve machine.I2C.scan.

torwag, 2016-09-25:
Just to understand this better,  what is the idea to differentiate between hardware and software? 
I think we had several discussions in this regards within the forum and feel people might be confused. 
Should pyb.* point always to software solutions,  whereas machine.* always refer to hardware? 
Would it make sense to have a machine.soft.* to clearly indicate that there is a differentiation between hard- and software implementation? 
Or would decorates help to make more clear what kind of implementation one is going to use?

dhylands, 2016-09-25:
If the set of pins passed in match up with the HW pins, then I'd sasy that a HW implementation should be used. If they don't then you can fallback to a SW implementation.

deshipu, 2016-09-25:
@dhylands I think this could be a very surprising behavior, especially when the software implementation is so different. While it would be very convenient to automatically select the hardware peripheral by specifying its pins, instead of having to look up its number in the board's reference, silent automatic fallback to software I2C when there is no hardware peripheral available on the specified pins smells of PHP's design philosophy to me, and could lead to hard to debug errors if the two implementations are different enough (and if there is no difference, then why have hardware i2c at all). I also think that you should be able to use software I2C (or SPI, or UART, or anything else) on pins that have hardware I2C available, if you choose for whatever reason (debugging, working around hardware bugs, etc.).

Besides, with boards such as the upcoming ESP32, you have to select both the number of the hardware interface, and the pins, as all of the pins can be switched between the peripherals at runtime. I think the current hardware API definition, where you specify the peripheral number (with negative numbers for software versions), and optionally pins, works very well for that.

dhylands, 2016-09-25:
Well I was making the assumption that both implementations would have the same API, doesn't make sense otherwise (pyb.I2C needs to be upgraded). They could also just be 2 totally separate implementations (machine.HW_I2C and machine.SW_I2C). You could also have a constructor arg which constrained it (i.e. only instantiate one or the other and not auto).

Different approaches seem to be suitable in different situations.

The real advantage of HW I2C comes when threading becomes involved as it uses much less CPU resources, and your timing constraints are dramatically minimized.

torwag, 2016-09-26:
I feel that users should get a more visible indicator what kind of implementation they are going to use.  Especially when it comes to read/understand third party code resp. code examples  or port a piece of code. People might need to identify clearly whether functions rely on certain board / chip features or whether they rely on software implementations and thus,  being somewhat board agnostic. Another example is the usage of interrupt routines, whereas software and hardware implementations might work completely different. Furthermore,  power routines (sleep modes)  would have different effects on both implementations.  Both might be good examples of tricky debugging (as mentioned by deshipu). 
Thus,  in my opinion the indicator should be even more visible as it is right now (negative indices). Not sure whether namespace (e. g.  machine.soft.i2c or machine.sw_i2c) or decorators (e. g.  @hw and @soft) are better

capaulson, 2016-09-26:
Hey, I originally posted about differences between hardware and software in the Micropython forums. I want to elaborate a bit more on what I've noticed. I have a board with a number of sensors, ADCs and DACs. I'd prefer not to post specific chips publicly, but I can email or PM them if interested. 

Basically, the hardware I2C will return this result: 

```
>>> i2c = I2C(1, I2C.MASTER)                                                    
>>> i2c.scan()                                                                  
[39, 44, 72, 73, 90] 
```

while the software I2C will return this result: 
```
>>> i2c = I2C(machine.Pin('B6'), machine.Pin('B7'))                             
>>> i2c.scan()                                                                  
[39, 72, 90] 
```

Occasionally, I'll see this result from the software I2C:
```
>>> i2c.scan()                                                                  
[39, 72, 90, 101]
```

I do have logic probe results for both hardware and software. I can confirm that software runs read commands while hardware runs write commands. I can find a way to share these data if useful.

dpgeorge, 2016-09-27:
> . I also think that you should be able to use software I2C (or SPI, or UART, or anything else) on pins that have hardware I2C available, if you choose for whatever reason

Yes, you can do this.  You just specify -1 as the peripheral, eg: `machine.I2C(-1, scl=pin1, sda=pin2)`.  Even if pin1 and pin2 can do HW I2C this constructor will use a SW implementation.  On pyboard if you do something like `machine.I2C('X')` then it uses the default HW block on the default pins for the X position (at least, that's what it should do when fully implemented).

> Especially when it comes to read/understand third party code resp. code examples or port a piece of code.

Drivers and 3rd party code should -- as much as is possible -- not construct peripheral objects but rather take them as arguments and act on them with standard methods/function.  Most drivers can be written this way.  Then construction of I2C busses, pins, etc is just up to the top-level code and is anyway board specific.  The way you construct peripherals is the same across boards, but the specific arguments to the constructor (whether it's SW or HW for example) is board specific and you can't really escape that fact.

dpgeorge, 2016-09-27:
@capaulson thanks for that particular bug report.  What output from i2c.scan() do you consider correct, the one with the 5 values?

4tmorris, 2016-09-27:
I get a similar result on my board.
```
MicroPython v1.8.4-65-g67d52d8 on 2016-09-26; CC3D Mini Revolution with STM32F405RG
Type "help()" for more information.
>>> from pyb import I2C
>>> i2c=I2C(1, I2C.MASTER)
>>> i2c.scan()
[30, 119]
>>> from machine import I2C
>>> i2c=I2C(machine.Pin('B8'), machine.Pin('B9'))
>>> i2c.scan()
[30]
```
The chips on the bus are HMC5883=30 and MS5611=119.

kamikaze, 2016-09-27:
voting for a HW I2C as a must )

danicampora, 2016-09-27:
I have seen that behaviour as well. Some devices do not ACK the read command, but do ACK on the empty write.

Cheers,
Daniel

peterhinch, 2016-09-27:
I've seen it with two identical devices (FRAM modules). Their different addresses are set by a hardware link. On fairly rare occasions the scan shows only the first (lowest address) device.

dpgeorge, 2016-09-28:
> On fairly rare occasions the scan shows only the first (lowest address) device.

@peterhinch is that a scan using pyb.I2C (hardware implementation, sending dummy writes) or machine.I2C (software implementation, sending dummy reads)?

It looks like the scanning should use empty writes to be more reliable.

torwag, 2016-10-03:
@dpgeorge You are right, drivers should not rely on peripheral objects but rather receive them during initialisation. My point was more, that I believe many users get confused about the hw vs. sw implementations. As easy as a solution with negative indices seems to be from a developers and heavy users perspective, I frighten that many (new or occasionally) users do not get this nuance difference in the code.
If then both implementations even behave differently, or have different limitations, users might have a hard time to understand why. A dedicated (self-explained by code) way to call either hardware or software implementations might make users more aware of the pros, cons and differences.
BTW. Just from a curious point of view, even if you disagree or say you do not want to change it now or forever, what would be the better way to do such a thing, extend the namespace (as most are familiar with this) or decorators (maybe the more elegant way)?

dpgeorge, 2016-10-04:
> I believe many users get confused about the hw vs. sw implementations. As easy as a solution with negative indices seems to be from a developers and heavy users perspective, I frighten that many (new or occasionally) users do not get this nuance difference in the code.
> ...
> what would be the better way to do such a thing,

Simplest and most straightforward would be separate classes: `machine.SoftI2C` and `machine.HardI2C`.  But I think it's already pretty "simple" in that if you don't specify any peripheral id then it defaults to software, eg `machine.I2C(scl=pin1, sda=pin2)` uses software.  The user needs to go out of their way to select a hardware peripheral by explicitly naming it, eg `machine.SPI(1)`.

dpgeorge, 2016-12-02:
machine.I2C in stmhal now uses hardware I2C blocks when possible (for F4 MCUs), see f8b71aabb4c53595d449e9b6f85f9ed35592ab12.

closed 2016-12-02 by dpgeorge