micropython/micropython #2455: Stack corruption in finally block

jpochyla, 2016-09-24:
`main.py`
```python
import test_module

def test():
    try:
        raise Exception()
    finally:
        val = 'flowers'
        test_module.foo('garbage', val)
        print(val)

x = test()
```

`test_module.py`
```python
def foo(a, b):
    pass
```

If the finally handler is not inside a function, or if the called function is in the same module, issue disappears. I'm not proficient enough in uPy internals to fix this, unfortunately.

prusnak, 2016-09-24:
Confirmed with Unix port on Linux 32-bit/64-bit

dpgeorge, 2016-09-24:
Thanks for the bug report, this one is a good find!  The problem is that the stack is crashing into the local variables storage (stack grows from the bottom up, local storage lives on the stack starting from the top).  It doesn't happen with @micropython.native on the test() function, so it's a problem specific to the bytecode emitter.  Basically, the emitter is not allocating/calculating enough local state for the stack.

dpgeorge, 2016-09-27:
Should be fixed as by-product of this optimisation: f040685b0cbec18feb981fdad96389a3b28b676d.  Test in 3f0c1c24528d8ce818434d8b553136d315b0d548.

closed 2016-09-27 by dpgeorge

prusnak, 2016-09-27:
I confirm that fix works on Unix. Awesome, thanks!

jpochyla, 2016-09-27:
Amazing, thanks a lot!