micropython/micropython #2457: stmhal: RTC stops ticking and cannot be updated after soft reboot

ryannathans, 2016-09-26:
After ctrl-D, the RTC does not tick and any calls to setting the datetime of the RTC take a long time and seem to fail and time out.

Tested on STM32F439 and PYBLITEV1.0 (STM32F411)

```
>>> r = pyb.RTC()
>>> r.datetime((2016, 9, 26, 0, 4, 19, 24, 135))
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 25, 85)
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 26, 112)
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 27, 133)
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 28, 174)
PYB: sync filesystems
PYB: soft reboot
>>> r = pyb.RTC()
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 37, 197)
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 37, 197)
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 37, 197)
>>> pyb.RTC().datetime((2016, 9, 26, 0, 4, 19, 37, 198)) #long wait, maybe 3 seconds before return
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 37, 197)
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 37, 197)
>>> r.datetime()
(2016, 9, 26, 0, 4, 19, 37, 197)
```

peterhinch, 2016-09-26:
I can't replicate this. A Pyboard V1.0 flashed with firmware built from source this morning (26th Sept) works as expected.

ryannathans, 2016-09-26:
@peterhinch is your original statement still correct, that it works on the Pyboard V1.0?

peterhinch, 2016-09-26:
Yes.

dpgeorge, 2016-09-27:
Thanks for the report, should now be fixed.

commit 88ca7ff5 (2016-09-27): stmhal/modmachine: Fix clearing of reset-cause flags.

To reset the flags we should write to the single bit only, not the entire
register (otherwise all other settings in the register are cleared).

Fixes #2457.
closed 2016-09-27 by dpgeorge