micropython/micropython #2460: stmhal common.ld refactor broke STM32L476DISC build

ravotus, 2016-09-28:
Hello,

Recently, commit 9310dad15d2572de89d853c71118eed5f8309028 introduced a change to refactor most of the common functionality of the STM32 linker scripts into a common.ld file. Unfortunately, the STM32L476DISC build no longer works after this change:

```
LINK build-STM32L476DISC/firmware.elf
arm-none-eabi-ld: build-STM32L476DISC/firmware.elf section `.isr_vector' will not fit in region `FLASH_ISR'
arm-none-eabi-ld: region `FLASH_ISR' overflowed by 8332 bytes
```

It looks like the L4 series has 2k flash sectors instead of 16k, so adding a couple of object files to the first flash sector doesn't work as nicely on that family.

As a side note, I noticed that the travis CI build does not build the STM32L476DISC board. Is that port stable enough to be included in the CI build file?

dpgeorge, 2016-09-28:
Please see #2454 for a fix for this.

> I noticed that the travis CI build does not build the STM32L476DISC board. Is that port stable enough to be included in the CI build file?

It is probably stable enough, but the travis CI build & testing is already slow enough that we don't want to bog it down further.

closed 2016-09-28 by dpgeorge

dhylands, 2016-10-04:
@ravotus - I run my own travis builds for a bunch of targets that aren't included in the regular MicroPython one.

I have a script that checks once per hour, and if any new commits exist, then it rebases to that and this causes a travis run under my account: https://travis-ci.org/dhylands/micropython (in particular on the travis-master branch)

I get notified by email if this fails, and will either fix the issue, comment on the commit which broke the build, or open a new issue.