micropython/micropython #2464: Binding AF_UNIX sockets fails

Snigelson, 2016-09-29:
Using the Unix port of Micropython at version v1.8.4-87-g53bfcc9-dirty, trying to bind a Unix domain socket fails.

The following code:

    import socket
    s = socket.socket(socket.AF_UNIX, socket.SOCK_STREAM)
    s.bind('/tmp/socket')

works under CPython, but fails (without much description) under Micropython:

    Traceback (most recent call last):
      File "unix_socket.py", line 3, in <module>
      File "/.../socket.py", line 49, in bind
      File "/.../socket.py", line 22, in _resolve_addr
    OSError: [addrinfo error -8]

On line 13-14 in ```socket.py``` (from latest micropython-lib) in function ```_resolve_addr``` there are the following lines:

    if isinstance(addr, (bytes, bytearray)):
        return addr

By passing the socket name as bytes, or changing the tuple passed to ```isinstance``` to ```(bytes, bytearray, str)```, I get past ```_resolve_addr``` and instead get the following error message:

    Traceback (most recent call last):
      File "unix_socket.py", line 3, in <module>
      File "/.../socket.py", line 49, in bind
      File "/.../socket.py", line 49, in bind
    OSError: [Errno 22] EINVAL

EINVAL is a bit more descriptive, but doesn't help much. According to the man page of ```bind```, EINVAL can mean one of the following:

* "The socket is already bound to an address." - Nope.
* "addrlen is wrong, or addr is not a valid address for this socket's domain." - More likely.

Both passing name as string and bytes works in CPython.

dpgeorge, 2016-09-29:
The `_resolve_addr` helper function in micropython-lib does not handle/understand AF_UNIX addresses.  But you can format them yourself.  You'll need to pass a sockaddr_un-compatible struct to bind, as a bytes/bytearray.  Try something like:
```python
addr = struct.pack('H108s', socket.AF_UNIX, "/tmp/socket")
s.bind(addr)
```

Snigelson, 2016-09-29:
I see. Thank you!

This is ugly, though :) I suppose this will be fixed in the future?

dpgeorge, 2016-09-29:
> I suppose this will be fixed in the future?

If the above works then this is an issue for micropython-lib rather than here.  And it's more of an enhancement than a fix, to make it work more like CPython.

Snigelson, 2016-09-29:
Ah, yeah. I don't really know why I didn't open this issue in micropython-lib in the first place.

closed 2016-10-04 by dpgeorge