micropython/micropython #2465: NeoPixels on dimly after np.fill((0,0,0)) -- should all be off

ricblue, 2016-09-29:
I think I may have found a bug in the NeoPixel driver 

Board = Adafruit ESP8266 Feather, reflashed with 1.8.4 

LED board is Adafruit NeoPixel FeatherWing -- 4x8 matrix of NeoPixels on a Feather layout 
https://www.adafruit.com/product/2945
Only hardware mod vs stock is to cut the track 16 jumper and solder track 15 jumper instead on the NeoPixel FeatherWing

Following the MicroPython docs for NeoPixel function ...
http://docs.micropython.org/en/latest/e ... xel-driver
This seems to work just as expected...
>>> from machine import Pin
>>> from neopixel import NeoPixel
>>> pin = Pin(15, Pin.OUT)
>>> np = NeoPixel(pin, 32)

Basic color functions work fine....
>>> np.fill((255,255,255))
>>> np.write()

and pixels are addressable correctly...
>>> np[0]=(100,0,0)
>>> np[1]=(0,100,0)
>>> np[2]=(0,0,100)
>>> np.write()

BUT while I try to set all to off (black) - ALL the pixels are still DIMLY LIT
>>> np.fill((0,0,0))
>>> np.write()

The LEDs appear to be the same brightness as np.fill(1,1,1)

One check: >>>print(np[0]) correctly displays (0,0,0) 

Now, really weird...
>>> np.fill((255,255,255))
>>> np.write()
All LEDs go very bright, as expected
But now follow that with the same np.fill(0,0,0) 
>>> np.fill((0,0,0))
>>> np.write()
AND NOW THE LEDS ARE ALL OFF!

But now just repeating 
>>> np.fill((0,0,0))
>>> np.write()
and they go back to dim !!!

AND EVEN RANDOM COLORS...
>>> np.fill((50,50,50))
>>> np.write()
>>> np.fill((0,0,0))
>>> np.write()

This consistently gives some LEDs off, some dim white, some COLORS - green or purple or yellow

I've erased my ESP8266 and reflashed with 1.8.4.  Didn't help.

Really feels like a driver bug of some sort

dpgeorge, 2016-09-30:
I can't reproduce this with 8 pixels (and I only have 8 of them...).  I suspect it might be a power consumption issue for your case.  If you have 32 pixels all fully on then that will draw close to 2 amps (60mA*32).

I would try the following:
- instantiate the NeoPixel object with just 8 pixels, np = NeoPixel(pin, 8) and see if you can reproduce the problem
- try using a different programming language/environment (eg Arduino) and see if you get the same problem with 32 pixels

ricblue, 2016-09-30:
Damian -- thank you for the quick response.  (BTW I'm a very satisfied Kickstarter backer -- amazing work!)

-- It doesn't seem to be power-related as np.fill(255,255,255) would be max power draw and is no issue.  only np.fill(0,0,0) is a problem

-- Tried your suggestion of 8 pixels -- IDENTICAL symptoms.  np.fill(0,0,0) = eight dim pixels, rest blank

-- I thought it might be electrical (e.g. short circuit) BUT the fact that np.fill(255,255,255) np.write() then np.fill(0,0,0) np.write() WORKS CORRECTLY seems pretty strong evidence it isn't electrical

In case this makes any difference, I've been doing all of this from the REPL prompt, not a download .py file

dpgeorge, 2016-09-30:
@ricblue please try the following:
- run `machine.freq(160000000)` first to set the CPU freq to 160MHz, then try neopixel tests again
- instead of doing np.write() try this code: `esp.neopixel_write(np.pin, np.buf, False)` (make sure to run `import esp` first!)
- please try using Arduino to make the same test; this would really help to narrow down the issue

ricblue, 2016-09-30:
@dpgeorge  --- raising the CPU freq fixed the problem!

machine.freq(160000000)  and the issue disappears!  np.fill((0,0,0)) now works correctly and the screen is blank not dim

This is reproducable and repeatable.  If i switch back to machine.freq(80000000) the problem immediately reappears

esp.neopixel_write:    at 80MHz (is80MHz = True) -- fails exactly the same way np.fill((0,0,0)) give dim not blank

BTW I think this is a typo...  
_import esp
esp.neopixel_write(pin, grb_buf, is800khz)_ 
rb_buf, is800khz)_ 
I think the freq choices are 80khz and 160khz so the "is800khz" should be "is80khz"

SEPARATELY... I think there might be a bug in the esp.neopixel_write(pin, grb_buf, is800khz)   This function appears to always work if is80khz is set to "True" and NEVER works if it is "False", which is what that flag seems it should be at 160MHz (after machine.freq(160000000).  

To be clear 
>>> machine.freq(160000000)
>>> np.fill((0,0,0))
>>> esp.neopixel_write(np.pin, np.buf, False)
... gives all bright white LEDS -- likely a clocking/timing problem

but 
>>> machine.freq(160000000)
>>> np.fill((50,50,0))
>>> esp.neopixel_write(np.pin, np.buf, True)
... works correctly - nice yellow LEDS

torwag, 2016-10-01:
It seems that the (bitbanging)  neopixels driver is close to its limit running at 80 MHz.  I had all kind of funny effects driving 5m (150) LEDs.
Setting the CPU to 160MHz and introducing the interrupt patch solved all problems for me. 
I will submit a patch towards the documentation to cover this.

dpgeorge, 2016-10-03:
> BTW I think this is a typo...

It's not a typo: there are different NeoPixel models that use different timing (800kHz vs 400kHz, apparently).

> it seems that the (bitbanging) neopixels driver is close to its limit running at 80 MHz.

That sounds like the problem, but I find it strange because even at 80MHz there should be enough cycles to get proper timing, namely 100 cycles per bit.

torwag, 2016-10-03:
Initally I thought that we might get tricked by interrupts for the wifi-stack on the esp8266. However, switching of IRQ should have solved the problem, but it didn't. The misbehaviour get less but is still there. I do not have a scope close to my esp-toyground at the moment. Maybe it is a good idea to record a few cycles of the bit-patterns to see if there is some sort of problem notable. The neopixel protocol is unfortunately, rather error-prone. A single wrong bit can create all kind of funny effects.

dpgeorge, 2016-10-03:
> I do not have a scope close to my esp-toyground at the moment. Maybe it is a good idea to record a few cycles of the bit-patterns to see if there is some sort of problem notable.

If someone has the time, please do measure with a scope the neopixel output at 80MHz and post results, thanks!

deshipu, 2016-10-05:
I made two captures with a logic analyzer at 80Mhz for my 8-led strip. One sets all leds to 0xff, 0xff, 0xff, the other sets them to 0x00, 0x00, 0x00. Attached below:

[neopixel.zip](https://github.com/micropython/micropython/files/512306/neopixel.zip)

I'm using the logic analyzer program from https://www.saleae.com/, you can use that to open the files.

mcauser, 2016-10-05:
@deshipu have you tried this analyser? https://github.com/monkbroc/neopixel-saleae
The last byte is not identified correctly (bug), but it can give you a decimal value for each of the proceeding bytes. Works with both 800 and 400kHz.

[image]

I'll upload my captures shortly, including uPy and Arduino versions for comparison.

In testing, I noticed a different issue. Set all 8x pixels on my NeoPixel stick to red, then green, then blue and very rarely 1 pixel wont change. I'll try to capture it on the logic analyser.

mcauser, 2016-10-05:
NeoPixel 8x stick and 32x strip tests with Saleae logic analyser captures:
[neopixel-tests.zip](https://github.com/micropython/micropython/files/512511/neopixel-tests.zip)

dpgeorge, 2016-10-06:
Thanks @deshipu for the data.  I filed an issue upstream at https://github.com/adafruit/Adafruit_NeoPixel/issues/112

deshipu, 2016-10-06:
By the way, I'm using a 8x neopixel stick from China, and I couldn't reproduce the issue.

ricblue, 2016-10-06:
@dpgeorge -- thanks for digging into this! 

One thought.  I haven't looked but I assume from the symptom we're seeing that the LSB of the desired NeoPixel value is clocked in last -- The error from the expected timing is greatest on the eighth bit and that is being mis-read ("mis-decoded") by the NeoPixel.  As I mentioned earlier in this thread, the false display you get from np.fill((0,0,0)) looks to me to be visually identical to np.fill((1,1,1)) -- it certainly isn't np.fill(128,128,128))

So couple of things strike me...
1) as it is only the Least Significant Bit that appears to be wrong, maybe that means the timing can't be worse that +/-12% off, otherwise the "2"s bit would be wrong sometimes as well (the "dim" appearance looks to me to always look like (1,1,1)

2) From your github/adafruit report, I see that there are some timing parameters 
"ESP8266:     0.4,    0.8,    1.25"   in play here.    It might be interesting to change there relevant parameters up or down by about 5% and see if that worsens or fixes the issue.  (The correct calculated value would be better for a permanent fix but this test might help confirm we're on the right track)

3) I had assumed this was a problem only when trying to display np.fill((0,0,0)) BUT maybe the Least Significant Bit of the brightness value is ALWAYS or OFTEN being displayed wrong.   We can see the difference between all off (0,0,0) and dim (looks like (1,1,1,)) BUT we'd never know if (32,32,32) was being displayed as (33,33,33) etc.  If we can't discriminate between those colors maybe it doesn't matter but makes the glitch more pervasive and more worth fixing

Richard

deshipu, 2016-10-06:
One more thought. How are your neopixels connected, and on which pin? Because pin 15 has a pulldown resistor on many boards (needed to make the esp8266 boot in the correct mode), which might make the voltage rise a bit slower and thus make the pulse longer? A similar situation is possible with long cables, etc.

torwag, 2016-10-06:
I would like to add that there is not only a problem with dim and not dim led units. If you drive e.g. 5m of LEDs (150 in total). I observed at 80MHz all kind of funny effects. LEDs turn green or red whereas they should be white or off. What I did was to run a "LED-star-trail" up the LED-stripe. You can find the code here 
#2211. Ignore the time measurements, they where just to prove the effect if IRQ settings.
At 160MHz I had no issues.

EddieParis, 2016-10-15:
My 2 cents about 160MHz vs 80MHz issue. As mentioned by dpgeorge in https://github.com/adafruit/Adafruit_NeoPixel/issues/112 the problems comes from the loop overhead and PIO set operation. So the timings tend to be too long. If the CPU is set to 160MHz the overhead is less visible in terms of timing, and the waveforms tend to be back in the tolerance.
The timings are so short that the overhead must really be taken in account for time0 and time1 calculations.

About long strips erratic glitches, I think the issue is linked to CCOUNT looping. It is a 32 bits counter and it loops every 50 secs@80 and 25 secs@160MHz. So the longer the strip is, the longer the transmission is and the more probable having CCOUNT looping is. When CCOUNT loops back to 0 the timing for a specific bit is totally wrong, and can have some unpredictable effects.

EddieParis, 2016-10-16:
I made some measurements on my board with an open bench logic sniffer. With original firmware @80MHz, I got T0H around 480ns which is normally inside leds tolerance, but this does not work !
Original firmware @160MHz gives T0H around 420ns and works perfectly.

So I decided to rebuild firmware and changing line 36 of espneopixel.c to
    time0  = fcpu / 2857143; // 0.35us
This to have a value closer to the spec. This gives me a T0H value of 400ns@80MHz which works perfectly.

So to solve this issue I would definitely recommend to change time0 value. Here is a modified firmware available for those who want to test without rebuilding the project : 
[micropython-neopixel-modif.zip](https://github.com/micropython/micropython/files/531667/micropython-neopixel-modif.zip)

I can make the correction and issue a pull request if corrections works fine for you too folks.

dpgeorge, 2016-11-07:
Fixed by above patch.

closed 2016-11-07 by dpgeorge