micropython/micropython #2466: stmhal: os directory operations throw cryptic errors on paths with trailing '/'

ryannathans, 2016-09-30:
It seems as though directory operations (listdir, mkdir, chdir, etc.) with the 'os' module throw cryptic errors when the path contains a trailing '/'. For example: '/sd/cache/' as opposed to the working '/sd/cache'.

```
>>> os.listdir('/flash/cache/')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 22] EINVAL

>>> os.listdir('/flash/cache')
[]
```

```
>>> os.mkdir('/sd/cache/test2/')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: Error creating directory '/sd/cache/test2/'
>>> os.mkdir('/sd/cache/test2')

```

```
>>> os.chdir('/flash/cache/')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: No such file or directory: '/flash/cache/'
>>> os.chdir('/flash/cache')

```

hosaka, 2016-10-04:
Why "cryptic" errors? 
`man errno` shows EINVAL as "Invalid Argument", and there's of course pages on the web that describe errors in detail.

I don't know if the FatFS supports stripping trailing slashes, I can only see trailing spaces trim in directory related functions. Perhaps we just shouldn't use trailing spaces and add a note to the documentation or strip them in the OS/VFS module before passing them to FatFS. But knowing the maintainers vision of keeping the code tight, it will probably be the former.

jonnor, 2024-09-29:
There seems to be an error here, with the same degree of information as other errors in MicroPython. So I think we can close this.

Generally we cannot promise to have very descriptive/detailed/pedagogical errors because error strings take up a lot of program space, which is highly constrained in microcontroller environments.

closed 2024-09-29 by jonnor