micropython/micropython #2468: zlib: Segfault on zlib.DecompIO() on unix

hosaka, 2016-09-30:
I was playing with the zlib on unix port today, trying to improve the test coverage and came across a problem that I don't really understand. The code is a slight cut down version of tests/extmod/uzlib_decompio_gz.py
```python
import uzlib as zlib
import uio as io

```

dpgeorge, 2016-10-01:
The test above gives `OSError: EINVAL` for me (not a segfault).  Removing the "-8" arg to DecompIO constructor makes it work correctly.

hosaka, 2016-10-01:
I've just tried to reproduce on my home pc to no success. Please ignore this for the time being, if I manage to figure out what I did that caused the segfault, I will update you.

pfalcon, 2016-10-12:
@hosaka : First of all, thanks much for your work on improving test coverage - that's both important on its own, and just the right thing to do at the current stage of the development, so double-hit, kudos. 

Regarding this bug - I recently pushed https://github.com/micropython/micropython/commit/39968aaaff575800218b47488f85eed7081cd0cb which fixes an issue of compression/decompression dictionary size mismatch. Surprisingly, I didn't see it leading to crash, instead to data corruption. Well, now it's properly caught.

The testcase you give above unlikely to be affected by that issue though, as data set (and thus dictionary) is small anyway.

hosaka, 2016-10-12:
@pfalcon Thanks Paul. I am hoping to progress a little more on the test coverage when I get some free time.

I didn't try to reproduce this again, but this could well be related to the commit you've linked. I'll stick it on my todo list and will test again with the new change. I will close this for the time being as I think we've exhausted the topic for now. Thanks again for the praise!

closed 2016-10-12 by hosaka