micropython/micropython #2474: Strange beavior with async kwargs

dmazzella, 2016-10-04:
Hi all,
i have tried this code:

```
$ ./micropython.exe
MicroPython v1.8.4-120-g99d62c4-dirty on 2016-10-04; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> class A(object):
...     def b(self, a=False):
...         pass
...
...
...
>>> class A(object):
...     def b(self, async=False):
...         pass
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 2
SyntaxError: invalid syntax
>>>
```

why async can't be used as kwargs?

Thanks,
D.

deshipu, 2016-10-04:
Same reason as why you can't use "class" or "def" or "except" -- it's a keyword.

dmazzella, 2016-10-04:
```
Python 3.4.3 (v3.4.3:9b73f1c3e601, Feb 24 2015, 22:43:06) [MSC v.1600 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> class A(object):
...     def b(self, async=False):
...         pass
...
>>>
```

deshipu, 2016-10-04:
It was added in 3.5: https://www.python.org/dev/peps/pep-0492/

dmazzella, 2016-10-04:
this work on 3.5

```
Python 3.5.2 |Continuum Analytics, Inc.| (default, Jul  5 2016, 11:45:57) [MSC v.1900 32 bit (Intel)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> class A(object):
...     def b(self, async=False):
...         pass
...
>>>
```

deshipu, 2016-10-04:
Apparently cpython's parser is smarter than MicroPython's about keywords.

dpgeorge, 2016-10-04:
The `async` and `await` names are not actually keywords in CPython at the moment.  From the PEP:

> async and await names will be softly deprecated in CPython 3.5 and 3.6. In 3.7 we will transform them to proper keywords. Making async and await proper keywords before 3.7 might make it harder for people to port their code to Python 3. 

Since it takes code space to support "soft keywords" it was decided to make them full keywords from the very beginning in uPy.  After all, any new code written in Python 3 should not use async/await as they will become keywords in the future, and then the code will not work.

closed 2016-10-04 by dmazzella