micropython/micropython #2475: esp8266: can't build with MICROPY_PY_USSL = 0 or MICROPY_SSL_AXTLS = 0

deshipu, 2016-10-04:
Setting either of those flags to 0 results in:

```
LINK build/firmware.elf
build/py/../extmod/moduhashlib.o:(.text.sha1_update+0x0): undefined reference to `SHA1_Update'
build/py/../extmod/moduhashlib.o: In function `sha1_update':
moduhashlib.c:(.text.sha1_update+0x1f): undefined reference to `SHA1_Update'
build/py/../extmod/moduhashlib.o:(.text.sha1_digest+0x0): undefined reference to `SHA1_Final'
build/py/../extmod/moduhashlib.o: In function `sha1_digest':
moduhashlib.c:(.text.sha1_digest+0x1a): undefined reference to `SHA1_Final'
build/py/../extmod/moduhashlib.o:(.text.sha1_make_new+0x0): undefined reference to `SHA1_Init'
build/py/../extmod/moduhashlib.o: In function `sha1_make_new':
moduhashlib.c:(.text.sha1_make_new+0x32): undefined reference to `SHA1_Init'
Makefile:204: recipe for target 'build/firmware.elf' failed
make: *** [build/firmware.elf] Error 1
```

hosaka, 2016-10-04:
moduhashlib depends on the axtls SHA1 crypt, disabling MICROPY_PY_UHASHLIB_SHA1 fixes the compilation. In unix mpconfigport.h we have 
```C
#if MICROPY_PY_USSL && MICROPY_SSL_AXTLS
#define MICROPY_PY_UHASHLIB_SHA1    (1)
#endif
```
Perhaps a similar #if is needed in esp8266/mpconfigport.h

hosaka, 2016-10-05:
Should I add a patch for this?

dpgeorge, 2016-10-06:
Fixed in 056da75a8a034e3b91c9b52ca812394914342059.  ussl and axtls take about 21k of code space.

closed 2016-10-06 by dpgeorge