micropython/micropython #2481: Zephyr RTOS port

pfalcon, 2016-10-05:
I've been working on Zephyr RTOS port for some time now, as part of prototyping/proof of concept work at day job: https://github.com/pfalcon/micropython/tree/zephyr. Over the last week, I met/was contacted by 5-6 people who expressed interest in hacking on it, and actually, there's already a parallel fork (initially based on my branch) which implements more functionality.

My initial concern with Zephyr port was whether I will be able to to support/maintain it going forward, and I got confirmation that I will be able to spend at least some basic time on it as part of my day job. Based on this, I'd like to start gradually spooling my changes to the master, for it to serve as a common base for everyone interested in this port.

pfalcon, 2016-10-05:
@daniel-thompson : FYI

dpgeorge, 2016-10-06:
> I'd like to start gradually spooling my changes to the master, for it to serve as a common base for everyone interested in this port.

Sounds good!

Would you consider putting it in ports/zephyr, as per ideas in #2401?

pfalcon, 2016-10-06:
> Sounds good!

Thanks!

> Would you consider putting it in ports/zephyr, as per ideas in #2401?

I don't know, it may be a somewhat complicated to land it under ports/ directly, especially that there're 2 diverged trees for Zephyr support. Perhaps, move it after everything is landed?

dpgeorge, 2016-10-06:
> Perhaps, move it after everything is landed?

I guess that's ok.  It's just, now that we are going to have a few more ports landing (zephyr, and hopefully samd21 soon) it would be good to decide (soon, but not immediately) and commit to a decision on how to structure the repo going forward.  For example, if we want to clean up the top level dir we can move logo/ to a separate repo (it doesn't really belong here), and put some parts from drivers/ into lib/, and the rest from drivers/ could go in micropython-lib, or a new drivers repo.

Anyway that's a bit off topic.  Would be good to see Zephyr support and I can try it out!

pfalcon, 2016-10-06:
> It's just, now that we are going to have a few more ports landing (zephyr, and hopefully samd21 soon) it would be good to decide (soon, but not immediately)

I guess the most productive way to go forward with this is to create a ticket "Prepare for 2.0 release" and list/discuss all the pending changes of that sort, like: moving ports under ports/, changing "mp_" prefix, changing "Micro Python" to "MicroPython" consistently, etc., etc. Well, I personally don't find anything of that critical at this time, so would leave people interested in that to drive that (creation and maintenance of the ticket above), or approach that closer to the end of the year, with the outlook to release 2.0 some time in the new year (Q1 tentative).

dpgeorge, 2016-10-07:
> I guess the most productive way to go forward with this is to create a ticket "Prepare for 2.0 release" and list/discuss all the pending changes of that sort

I agree with you.  So this issue is not hijacked, and we give everyone a chance to prepare for breaking changes, I've created #2486.

pfalcon, 2017-03-12:
From https://github.com/micropython/micropython/issues/2946#issuecomment-285597081

> I tried but couldn't get zephyr port running (it built but qemu_x86 just sat there doing nothing; using the minimal build).

Thanks for the report, confirmed. The Zephyr port build system does have few issues. And it's not exactly easy to solve some, because the idea is that it should follow in interface of Zephyr's build system whenever possible, but then I try to make the other things work like in other MicroPython ports, and that doesn't work well together. I'm going to look into ways to fix that.

pfalcon, 2017-03-12:
> I tried but couldn't get zephyr port running (it built but qemu_x86 just sat there doing nothing; using the minimal build).

Fixed in https://github.com/micropython/micropython/commit/40e541063faf15bd8965f2d65d5a5be08fdef851 by introducing a make warpper script to build/run "minimal" zephyr port.

Another issue, that any make invocation caused a full rebuild, was fixed in https://github.com/micropython/micropython/commit/736a8a8ac7abfeab5edb8dfd73ecf0b2c0282e61.

There's still a known issue that switching to build for second BOARD= after you built first one requires explicit "make clean" inbetween.

dpgeorge, 2017-03-14:
> Fixed in 40e5410 by introducing a make warpper script to build/run "minimal" zephyr port.

Thanks, that works for me.

> Another issue, that any make invocation caused a full rebuild, was fixed in 736a8a8.

That's also a good fix, although it's still a bit annoying that it needs to relink (which is slow) each time you just want to run the emulator (eg ./make-minimal qemu).

Also, I get "UART buffer overflow - char dropped" when in paste mode if the paste has moderate size (roughly 300 bytes).

pfalcon, 2017-03-14:
> That's also a good fix, although it's still a bit annoying that it needs to relink (which is slow) each time you just want to run the emulator (eg ./make-minimal qemu).

Slow as "not instantaneous"? Well, that's how Zephyr build system works I believe, I'm not even sure if it relinks, or just checks whether some stuff needs to be updated. I'll double-check, but I believe other apps work in the same way, so little can be done on uPy side.

And well, you didn't check "make clean" - after a git pull, that'll start with recompiling Kconfig before proceeding to clean something, but not all (that's what "clean" means in Z land). That's what I call "slow". That behavior was contributed and I'm losing patience with it (I literally do rm -rf outdir manually instead), so will submit a patch to revert to normal uPy behavior soon. @daniel-thompson: ping ;-).

> Also, I get "UART buffer overflow - char dropped" when in paste mode if the paste has moderate size (roughly 300 bytes).

Well, that's why there a UART buffer at all. Does esp8266 work better for such pastes? No idea how to fix that, that would involve digging in Zephyr task priorities and scheduling to make sure uPy thread gets a chance to run inbetween data being pumped into circular buffer.

daniel-thompson, 2017-03-14:
On 14/03/17 07:55, Paul Sokolovsky wrote:
> Slow as "not instantaneous"? Well, that's how Zephyr build system works
> I believe, I'm not even sure if it relinks, or just checks whether some
> stuff needs to be updated. I'll double-check, but I believe other apps
> work in the same way, so little can be done on uPy side.

Currently the micropython side forces a relink by unconditionally 
bydeleting zephyr.elf because zephyr build system adds the micropython 
libraries as custom linker flags and therefore cannot run the proper 
dependancy chain.

Agree that relinking to run an emulator is somewhat annoying. We could 
perhaps improve this if the command to remove zephyr.elf only ran when 
we relink that micropython library.

> And well, you didn't check "make clean" - after a git pull, that'll
> start with recompiling Kconfig before proceeding to clean something, but
> not all (that's what "clean" means in Z land). That's [...]

For a normal Zephyr application "clean" removes the object files (i.e. 
things that can be automatically recreated) whilst "mrproper" removes 
everything (including files that may have been customized by the user 
such as the config). It would really suck if micropython definition of 
"clean" removes files that it cannot reconstruct. That will annoy people 
much more!

If you *insist* on tolerating build system bugs[1] then consider copying 
zephyr approach and using mrproper for this!

[1] The "make clean" habit is usually a consequence of insufficient
     dependency tracking ;-)... although I'll happily concede that the
     unconditional link above is an example of over-zealous dependency
     tracking.

dpgeorge, 2017-03-15:
> Well, that's why there a UART buffer at all. Does esp8266 work better for such pastes?

I can reliably paste up to 1k of data into esp8266 paste-mode (it has a 256 byte buffer).

pfalcon, 2017-03-27:
> Currently the micropython side forces a relink by unconditionally by deleting zephyr.elf

@daniel-thompson: Thanks for the quick reply, sorry, I went on vacation on missed to reply. Well, it seems that you know/remember how uPy/Z build system works better than me ;-) And I guess the best reply is switching to proper use of "make exports", long overdue: https://github.com/micropython/micropython/pull/2980

> For a normal Zephyr application "clean" removes the object files (i.e. things that can be automatically recreated) whilst "mrproper" removes everything (including files that may have been customized [...]

I see, no response again re: "make clean" trying to build Kconfig ;-). But ok, consider me convinced for next few months still, and I didn't include any "make clean" changes in https://github.com/micropython/micropython/pull/2980 (my idea was to make Z's version of clean to be available via "make z_clean", and have old good unconditional clean where up front).

> If you *insist* on tolerating build system bugs[1] then consider copying zephyr approach and using mrproper for this!

Why not the other way around - make clean cleans, and make-something sub-cleans, and Zephyr adopting it? :-P

daniel-thompson, 2017-03-27:
> Why not the other way around - make clean cleans, and make-something sub-cleans,
> and Zephyr adopting it? :-P

If you want a whole of uPy solution you don't have to copy the mrproper... you could use distclean or pristine instead ;-).

That said, IIUC, most of uPy doesn't care about this distinction because the configuration settings are mostly hard coded for each device and therefore zephyr makes things harder because it is a platform rather than a single device (and the unix port doesn't get into these tangles because very few people try to compile features out of the C library to save space).

Anyhow, I think the prior art of a variety build systems (autotools, cmake, etc) is on my side in suggesting that "good old clean" is *not* the name of a rule that nukes things that cannot be reconstructed automatically. Unfortunately what there isn't is total consensus on what the name of a super-clean rule should be :-( .

jibonaronno, 2018-03-31:
Hi,
I am looking for some help regards latest building of micropython over zephyr. I build Zephyr for stm32f4_disco and also build micropython for STM32F4DISC separately. But cannot build could not build micropython for zephyr by the given instruction as https://github.com/micropython/micropython/tree/master/ports/zephyr page. I am using windows 10, MinGw and MSYS. Shown error:

aronno@DESKTOP-FHE7SAF ~/micropython/ports/zephyr
$ make V=1 BOARD=stm32f4_disco
Makefile:29: outdir/stm32f4_disco/Makefile.export: No such file or directory
python makeprj.py prj_base.conf prj_stm32f4_disco.conf prj_stm32f4_disco_merged.conf
Replacing
mkdir -p outdir/stm32f4_disco && cmake -DBOARD=stm32f4_disco -DCONF_FILE=prj_stm32f4_disco_merged.conf -Boutdir/stm32f4_disco -H.
-- Building for: NMake Makefiles
CMake Deprecation Warning at C:/Users/aronno/zephyr/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:1 (include)

-- Found PythonInterp: C:/Users/aronno/AppData/Local/Programs/Python/Python36/python.exe (found suitable version "3.6.5", minimum required is "3.4")
-- Selected BOARD stm32f4_disco
Zephyr version: 1.11.99
prj_stm32f4_disco_merged.conf:25: warning: attempt to assign the value "5" to the undefined symbol NET_NBUF_RX_COUNT
Parsing Kconfig tree in C:/Users/aronno/zephyr/Kconfig
Using C:/Users/aronno/zephyr/boards/arm/stm32f4_disco/stm32f4_disco_defconfig as base
Merging prj_stm32f4_disco_merged.conf
warning: TEST_RANDOM_GENERATOR (defined at C:/Users/aronno/zephyr\subsys/random/Kconfig:8) was assigned the value "y" but got the value "n" -- check dependencies
-- Generating zephyr/include/generated/generated_dts_board.h
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@0 has a unit name, but no reg property
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@1 has a unit name, but no reg property
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart6@0 has a unit name, but no reg property
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart6@1 has a unit name, but no reg property
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@2 has a unit name, but no reg property
stm32f4_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
-- The C compiler identification is GNU 7.2.1
-- The CXX compiler identification is GNU 7.2.1
-- The ASM compiler identification is GNU
-- Found assembler: C:/gccarmemb/bin/arm-none-eabi-gcc.exe
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Failed
CMake Error at C:/Users/aronno/zephyr/cmake/extensions.cmake:962 (message):
  Assertion failed: The toolchain is unable to build a dummy C file.  See
  CMakeError.log.
Call Stack (most recent call first):
  C:/Users/aronno/zephyr/CMakeLists.txt:18 (assert)

-- Configuring incomplete, errors occurred!
See also "C:/MinGW/msys/1.0/home/aronno/micropython/ports/zephyr/outdir/stm32f4_disco/CMakeFiles/CMakeOutput.log".
See also "C:/MinGW/msys/1.0/home/aronno/micropython/ports/zephyr/outdir/stm32f4_disco/CMakeFiles/CMakeError.log".
make: *** [outdir/stm32f4_disco/Makefile] Error 1

dpgeorge, 2019-06-24:
The Zephyr port was added some time ago.  Further issues with it can be discussed in independent issues.

closed 2019-06-24 by dpgeorge