micropython/micropython #2483: sys.exit() causes softreset, bug or feature(maybe better documentation)?

mika-koivusaari, 2016-10-06:
>>> import sys
>>> sys.exit()
PYB: soft reboot
#6 ets_task(40100390, 3, 3fff6300, 4)
MicroPython v1.8.4-65-g67d52d8 on 2016-10-04; ESP module with ESP8266

Documentation doesn't really say what this function does, and I was wondering if it should do a soft reboot as it does?

For reference I'm making a script that does things, goes to deep sleep and repeats. And I wanted to check if one pin is pulled down and then stop executing, for making software update, debugging etc. I can do it differently, but I was left wondering whats the purpose of this function?

dhylands, 2016-10-06:
On an MCU, the only way to stop is to remove power. The sys.exit function raises a SystemExit exception (which is the same thing that CPython does). You can intercept that using a try/except and perform your update or whatever other thing you want to do.

pfalcon, 2016-10-12:
Fixed by https://github.com/micropython/micropython/pull/2495

closed 2016-10-12 by pfalcon